xbianonpi/xbian #742: Rpi 2 Network settings.

coderpop, 2015-06-21:
Within Kodi>System>settings>xbian>network, the settings are not held when entered vai kodi menu and you have to ssh in to set them from xbian-config. Also the notice "are you sure to apply modification" keeps popping up, even if no changes have been made!

CurlyMoo, 2015-06-21:
Until a new python developers comes along, the xbian config inside xbmc isn't maintained anymore.

The-Exnor, 2015-06-22:
@CurlyMoo  what happened to the old python dev?

CurlyMoo, 2015-06-22:
Not sure.

mk01, 2015-07-08:
@The-Exnor 

was travelling to Brazil, so most likely eaten by huge anaconda.

belese, 2015-07-13:
No i'm still alive, Anaconda wasn't so big ;-)
But i've just no time for the moment, and don't use xbian or any HTPC for now.

Maybe one day, but not sure. 
But if someone want to take development of python addon, i can give som advice.

Belese

CurlyMoo, 2015-07-13:
@hackedbellini was planning to done some development.

hackedbellini, 2015-07-17:
I'm :)

Just got my old rpi setup to use as a development device and I'm getting started right now.

@belese 

> But if someone want to take development of python addon, i can give som advice.

As a matter of fact, there is one thing that I'm struggling to understand there. I wanted to start by fixing the translations (that seems to be broken), converting them to the `.po` format for use with gettext, but I'm having a hard time with the current one.

I saw that there are some `strings.xml` files defining strings for a numeric id (e.g. `<string id="31010">Resize SD</string>"`). But the calls to ugettext uses strings like `xbian-config.main.reboot_question`.

I thought that there may have a mapping between them, to link those string identifier to the id in the xml. I grepped for both the numeric id and some of those strings inside the ugettext calls and didn't find any place defining that mapping or anything like that. Actually, the only matches for the strings and the ids were themselves.

I tried searching for a kodi documentation about that to give me a clue, but I found nothing helpful. Would be useful if you could explain to me how that translation system works. More specifically, how the the strings in ugettext are linked to the one in the xml file.

Maybe it would be better if we can talk on irc to avoid discussing it here, since the topic is unrelated to the issue =P

Btw, I'm assigning this issue to me

CurlyMoo, 2015-07-17:
Use this to write down everything in the xbian-package-config-xbmc wiki so future developers understand as well.

hackedbellini, 2015-07-20:
Just discovered it. Those xml files are not used anymore. Just found the `xbian-config.main.reboot_question` (and other alikes) on the transifex' .po file.

I don't like that approach very much. Since with po files there's no need to use string ids, I think it is much better to use the string directly. If you take a look at [stoq](https://github.com/stoq/stoq) for example, a project which I'm a developer, we do translations like this:

* Here is an example of a string marked for translation: https://github.com/stoq/stoq/blob/master/stoq/gui/services.py#L584
* Here is how it looks on the translation template: https://github.com/stoq/stoq/blob/master/po/stoq.pot#L8429
* Here is how it looks on a translated po file (in this case, pt_BR): https://github.com/stoq/stoq/blob/master/po/pt_BR.po#L8436

The advantages of using the strings directly is that it is much easier to look at the code and understand what is going on, instead of having to guess of look at both the source and the po file for that.

So, I'm going to make use of the strings directly. I'll also write a script to migrate the existing translations so that none of them are lost. Do anyone have anything against this?

CurlyMoo, 2015-07-20:
Nope, as long as it's documented.

CurlyMoo, 2015-07-20:
And are you going to keep transifex as the translations source?

hackedbellini, 2015-07-20:
@CurlyMoo of course! Transifex will remain the same... the only changes will be on the source code.

belese, 2015-07-22:
@hackedbellini
Hi, 
if i remember, 
this file need to be update in git, 
https://github.com/xbianonpi/xbian-package-config-shell/blob/master/content/usr/local/include/xbian-config/gettext/xbian.en.po
the transifex need to be updated with this file (the transifex admin password is in a thread on xbian private forum, don't remember).

Then run this script on local giot copy package : 
https://github.com/xbianonpi/xbian-package-config-shell/blob/master/update.translation
(you need to configure user and password in a hidden file, see transifex help).

and push on xbian git after, and translation should be updated.

Belese

@CurlyMoo 
I've reinstalled XBian yesterday, and tried the xbian-config-python, but it was working (except the translation 'click to load' miss), are there another problem?

CurlyMoo, 2015-07-22:
A few, network settings i though.

belese, 2015-07-22:
ok because i've connected my wifi with XBian-config, and it was working as it should, 
but not try the Fixed IP.

hackedbellini, 2015-07-22:
@belese 

> ANd don't forget if you change that translation, string, id, are shared with xbian-config-shell,
and was done before, that's why i've continue this way, and only add translation i need for xbmc/kodi.

I see... hrm, I think the best thing to do here is to split those translations instead of having them being shared. This reduces complexity and make everything less error-prone.

belese, 2015-07-22:
Maybe, don't know, it's a lot of work (more or less), 
and translators will have to translate two times the same content.

hackedbellini, 2015-07-22:
@belese 

Yes, the "translate two times the same content" is kind of an issue, but I don't see it as a major one, specially with [transifex' translation memory](http://docs.transifex.com/features/tm/).

Also, I'll port all the existing translations so only the one we add/modify from now on will need to be translated (and probably they will not be on shell too, unless they have a reason to be).

And to list some advantages here:

1) Having them separately means that each project can adapt the strings to better fit its context. Of course, some of the strings will still be very alike (if not the same), but sometimes we need to contextualize them depending if they are being displayed on the shell or on kodi.

2) Also, it avoids breaking stuff on the other part. For example, a string with a `%s` inside it, if it is removed by someone working on shell, it would break the python's string formatting.

3) By not using the inherited identifiers from shell and using the strings directly, it makes it easier to understand what is going on and spot errors. For example, `_("Foo %s bar %s") % (foo, bar)` is easier to read, more obvious and less error-prone than `_("foo.bar.foo_bar_string") % (foo, bar)`