xbianonpi/xbian #743: OpenVPN/Tun0 interface causes "System Info" to crash

farthinder, 2015-06-25:
Kodi 14.2
Xbian 1.0 (Knockout) 
Kernel 3.18.8
HW: Raspberry Pi 2
Fully updated (apt-get update/upgrade)

I´ve installed these additional packets:
build-essential, libssl-dev,  libpam0g-dev, git, liblzo2-dev, dns-utils
I did this to make and install OpenVPN 2.3.7

If i start up Kodi as normal there is no problem opening "System Info" (found under the System menu in the home menu), but if I open a OpenVPN-connection to my VPN-server Kodi crashes and relaunches when i open the "System Info" menu. If i kill the VPN-connection the System Info opens again.

farthinder, 2015-06-25:
Log first opening the System info without OpenVPN running, and than the second time with OpenVPN running:

18:08:42 T:1961074688  NOTICE: special://profile/ is mapped to: special://masterprofile/
18:08:42 T:1961074688  NOTICE: -----------------------------------------------------------------------
18:08:42 T:1961074688  NOTICE: Starting Kodi (14.2 Git:Unknown). Platform: Linux ARM 32-bit
18:08:42 T:1961074688  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
18:08:42 T:1961074688  NOTICE: Kodi compiled May  7 2015 by GCC 4.6.3 for Linux ARM 32-bit version 3.2.51 (197171)
18:08:42 T:1961074688  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 3.18.8+
18:08:42 T:1961074688  NOTICE: FFmpeg statically linked, version: 2.4.6-xbmc-2.4.6-Helix
18:08:42 T:1961074688  NOTICE: Host CPU: ARMv7 Processor rev 5 (v7l), 4 cores available
18:08:42 T:1961074688  NOTICE: ARM Features: Neon enabled
18:08:42 T:1961074688  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
18:08:42 T:1961074688  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
18:08:42 T:1961074688  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
18:08:42 T:1961074688  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
18:08:42 T:1961074688  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
18:08:42 T:1961074688  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
18:08:42 T:1961074688  NOTICE: Local hostname: xbian
18:08:42 T:1961074688  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
18:08:42 T:1961074688  NOTICE: -----------------------------------------------------------------------
18:08:43 T:1961074688  NOTICE: load settings...
18:08:43 T:1961074688  NOTICE: Found 1 Lists of Devices
18:08:43 T:1961074688  NOTICE: Enumerated PI devices:
18:08:43 T:1961074688  NOTICE:     Device 1
18:08:43 T:1961074688  NOTICE:         m_deviceName      : HDMI
18:08:43 T:1961074688  NOTICE:         m_displayName     : HDMI
18:08:43 T:1961074688  NOTICE:         m_displayNameExtra:
18:08:43 T:1961074688  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
18:08:43 T:1961074688  NOTICE:         m_channels        : FL,FR
18:08:43 T:1961074688  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
18:08:43 T:1961074688  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
18:08:43 T:1961074688  NOTICE:     Device 2
18:08:43 T:1961074688  NOTICE:         m_deviceName      : Analogue
18:08:43 T:1961074688  NOTICE:         m_displayName     : Analogue
18:08:43 T:1961074688  NOTICE:         m_displayNameExtra:
18:08:43 T:1961074688  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
18:08:43 T:1961074688  NOTICE:         m_channels        : FL,FR
18:08:43 T:1961074688  NOTICE:         m_sampleRates     : 48000
18:08:43 T:1961074688  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
18:08:43 T:1961074688  NOTICE:     Device 3
18:08:43 T:1961074688  NOTICE:         m_deviceName      : Both
18:08:43 T:1961074688  NOTICE:         m_displayName     : HDMI and Analogue
18:08:43 T:1961074688  NOTICE:         m_displayNameExtra:
18:08:43 T:1961074688  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
18:08:43 T:1961074688  NOTICE:         m_channels        : FL,FR
18:08:43 T:1961074688  NOTICE:         m_sampleRates     : 48000
18:08:43 T:1961074688  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
18:08:43 T:1961074688  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
18:08:43 T:1961074688  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <?xml version="1.0" encoding="UTF-8" ?>
                                            <advancedsettings>
                                              <video>
                                                <defaultplayer>dvdplayer</defaultplayer>
                                                <defaultdvdplayer>dvdplayer</defaultdvdplayer>
                                              </video>
                                            </advancedsettings>
18:08:43 T:1961074688  NOTICE: Loaded settings file from special://profile/advancedsettings.xml
18:08:43 T:1961074688  NOTICE: Contents of special://profile/advancedsettings.xml are...
                                            <advancedsettings>
                                              <network>
                                                <cachemembuffersize>20971520</cachemembuffersize>
                                                <readbufferfactor>4.0</readbufferfactor>
                                              </network>
                                              <samba>
                                                <clienttimeout>10</clienttimeout>
                                              </samba>
                                              <fanartres>720</fanartres>
                                              <imageres>540</imageres>
                                              <fullscreen>true</fullscreen>
                                              <libass>
[... 682 lines cut ...]
[... cut ...]
18:10:29 T:1961074688   DEBUG: OnKey: return (0xf00d) pressed, action is Select
18:10:29 T:1961074688   DEBUG: WakeUpScreenSaverAndDPMS: WakeUpScreenSaverAndDPMS - bPowerOffKeyPressed 0
18:10:29 T:1961074688   DEBUG: WakeUpScreenSaverAndDPMS: WakeUpScreenSaverAndDPMS - result of WakeUpScreenSaver 0
18:10:29 T:1961074688   DEBUG: Activating window ID: 10007
18:10:29 T:1961074688   DEBUG: ------ Window Deinit (Home.xml) ------
18:10:29 T:1961074688   DEBUG: ------ Window Init (SettingsSystemInfo.xml) ------

farthinder, 2015-06-25:
Log from after the crash
18:10:32 T:1960656896  NOTICE: special://profile/ is mapped to: special://masterprofile/
18:10:32 T:1960656896  NOTICE: -----------------------------------------------------------------------
18:10:32 T:1960656896  NOTICE: Starting Kodi (14.2 Git:Unknown). Platform: Linux ARM 32-bit
18:10:32 T:1960656896  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
18:10:32 T:1960656896  NOTICE: Kodi compiled May  7 2015 by GCC 4.6.3 for Linux ARM 32-bit version 3.2.51 (197171)
18:10:32 T:1960656896  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 3.18.8+
18:10:32 T:1960656896  NOTICE: FFmpeg statically linked, version: 2.4.6-xbmc-2.4.6-Helix
18:10:32 T:1960656896  NOTICE: Host CPU: ARMv7 Processor rev 5 (v7l), 4 cores available
18:10:32 T:1960656896  NOTICE: ARM Features: Neon enabled
18:10:32 T:1960656896  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
18:10:32 T:1960656896  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
18:10:32 T:1960656896  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
18:10:32 T:1960656896  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
18:10:32 T:1960656896  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
18:10:32 T:1960656896  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
18:10:32 T:1960656896  NOTICE: Local hostname: xbian
18:10:32 T:1960656896  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
18:10:32 T:1960656896  NOTICE: -----------------------------------------------------------------------
18:10:32 T:1960656896  NOTICE: load settings...
18:10:32 T:1960656896  NOTICE: Found 1 Lists of Devices
18:10:32 T:1960656896  NOTICE: Enumerated PI devices:
18:10:32 T:1960656896  NOTICE:     Device 1
18:10:32 T:1960656896  NOTICE:         m_deviceName      : HDMI
18:10:32 T:1960656896  NOTICE:         m_displayName     : HDMI
18:10:32 T:1960656896  NOTICE:         m_displayNameExtra:
18:10:32 T:1960656896  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
18:10:32 T:1960656896  NOTICE:         m_channels        : FL,FR
18:10:32 T:1960656896  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
18:10:32 T:1960656896  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
18:10:32 T:1960656896  NOTICE:     Device 2
18:10:32 T:1960656896  NOTICE:         m_deviceName      : Analogue
18:10:32 T:1960656896  NOTICE:         m_displayName     : Analogue
18:10:32 T:1960656896  NOTICE:         m_displayNameExtra:
18:10:32 T:1960656896  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
18:10:32 T:1960656896  NOTICE:         m_channels        : FL,FR
18:10:32 T:1960656896  NOTICE:         m_sampleRates     : 48000
18:10:32 T:1960656896  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
18:10:32 T:1960656896  NOTICE:     Device 3
18:10:32 T:1960656896  NOTICE:         m_deviceName      : Both
18:10:32 T:1960656896  NOTICE:         m_displayName     : HDMI and Analogue
18:10:32 T:1960656896  NOTICE:         m_displayNameExtra:
18:10:32 T:1960656896  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
18:10:32 T:1960656896  NOTICE:         m_channels        : FL,FR
18:10:32 T:1960656896  NOTICE:         m_sampleRates     : 48000
18:10:32 T:1960656896  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
18:10:32 T:1960656896  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
18:10:32 T:1960656896  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <?xml version="1.0" encoding="UTF-8" ?>
                                            <advancedsettings>
                                              <video>
                                                <defaultplayer>dvdplayer</defaultplayer>
                                                <defaultdvdplayer>dvdplayer</defaultdvdplayer>
                                              </video>
                                            </advancedsettings>
18:10:32 T:1960656896  NOTICE: Loaded settings file from special://profile/advancedsettings.xml
18:10:32 T:1960656896  NOTICE: Contents of special://profile/advancedsettings.xml are...
                                            <advancedsettings>
                                              <network>
                                                <cachemembuffersize>20971520</cachemembuffersize>
                                                <readbufferfactor>4.0</readbufferfactor>
                                              </network>
                                              <samba>
                                                <clienttimeout>10</clienttimeout>
                                              </samba>
                                              <fanartres>720</fanartres>
                                              <imageres>540</imageres>
                                              <fullscreen>true</fullscreen>
                                              <libass>
                                                <styleoverrides>
[... 613 lines cut ...]
[... cut ...]
18:10:40 T:1818227728   DEBUG: Version Check: Version available  {u'major': u'14', u'extrainfo': u'final', u'tagversion': u'', u'tag': u'stable', u'addon_support': u'yes', u'minor': u'2', u'revision': u'20150326-7cc53a9'}
18:10:40 T:1818227728   DEBUG: Version Check: There is no newer stable available
18:10:40 T:1818227728    INFO: CPythonInvoker(0, /home/xbian/.kodi/addons/service.xbmc.versioncheck/service.py): script successfully run
18:10:40 T:1818227728    INFO: Python script stopped
18:10:40 T:1818227728   DEBUG: Thread LanguageInvoker 1818227728 terminating
18:10:44 T:1960656896   DEBUG: ------ Window Deinit (DialogKaiToast.xml) ------

CurlyMoo, 2015-06-25:
As you might have seen, you're post a absolutely unreadible. Use `github markdown` to fix it please.

farthinder, 2015-06-25:
There we go, a little better :)

CurlyMoo, 2015-06-25:
https://github.com/xbmc/xbmc/issues

Here you go :wink:

farthinder, 2015-06-25:
I have a simular setup on OS X with no problem, thats why i considered this a Xbian problem.

CurlyMoo, 2015-06-25:
Then try OSMC to be sure.

farthinder, 2015-06-25:
oki

farthinder, 2015-06-25:
Is that really a fair comparison when OSMC runs jessie?

CurlyMoo, 2015-06-25:
Yes

farthinder, 2015-06-25:
OSMSC does not boot after apt-get update/upgrade and OpenVPN installation, I will not sink more time in to this but rather live with not using the System Info. I have a problem when using a mysql db with Xbian that i might submit a new ticket about after testing it on two platsfforms. 

Do you know who i could pandor to inorder to get a Kodi xbian-packet for 15.0 Beta2?

CurlyMoo, 2015-06-25:
This is what's available:
http://apt.xbian.org/pool/devel/rpi-wheezy/x/xbian-package-xbmc-nightly/

farthinder, 2015-06-25:
I don't quite follow you?
The official beta2 on kodi.tv was released 14-Jun-2015 13:39  and the latest file i find via the link you sent is from 25-Feb-2015

mkreisl, 2015-06-25:
@farthinder 
You can test my builds from here https://www.mediafire.com/folder/2ngwumkbzw36z/XBian_Packages
libcec3 is also needed, libass4 if you have SSA/ASS subtitles

CurlyMoo, 2015-06-25:
@mkreisl, why don't you upload to devel of our official repo?

farthinder, 2015-06-25:
@mkreisl Thanks, i´ll have a look!

mkreisl, 2015-06-25:
@CurlyMoo 
Still don't know how to do that. I asked @mk01 but got now answer. Maybe you can tell me how to do this per eMail or irc

farthinder, 2015-06-25:
For anyone who discoves this later on, I have done some further testing:
Changing the name of the virtual interface to eth1 does not fix it.
As soon as the OpenVPN-tunne is up Kodi crashes when trying to access any FTP-shares (the only type of share i have tested), regardless if the FTP share is access over VPN or over the internet.

farthinder, 2015-06-25:
Im not sure what Gits policy on curse words are but suffice it to say that this day has been spent verifying a bug in Kodi 14 that has apparently been fixed in 15 beta1

CurlyMoo, 2015-06-25:
At least you learned how to use github markdown :+1:

closed 2015-06-25 by CurlyMoo

mkreisl, 2015-06-25:
@farthinder 
i have put the actual nightly version into the devel repo. You can download it from there

farthinder, 2015-06-25:
Many, many thanks!
I will have a look at it tomorrow.

If your interested we could package my openvpn solution and add that as as
a separate package?
It's nothing fancy, it establishes a vpn tunnel before kodi starts in order
for kodi to access MySQL and shares of the vpn. I haven't made a gui or
anything.

mk01, 2015-07-08:
@farthinder 

would it be possible for you to start kodi.bin via gdb (debugger) and send back the backtrace ? 
the cause of this is 99% in the new networking code with IPv6 support. the fact that v15 works ok is that this change is not applied there. 

the networking rewrite is currently being upstreamed (so will become part Kodi's code). once this happens without this bug worked out, v15 (and v16) will be broken too. 

most probably the crash happens while new networking info is being re-read from system (after you establish the tunnel) - but a backtrace would be great to have.

farthinder, 2015-07-08:
Can you find me a toturioal or somthing on how to generate the backtrace and i´ll have a look. 
If you´d rather just test it your self all you need to do is make and install OpenVPN 2.3.7 and have it connect, I have a guide for it if your interested.

mk01, 2015-07-08:
basically let's try just this while logged as xbian user:

```
sudo -s
apt-get install gdb libc6-dbg
exit
gdb -args /usr/local/lib/kodi/kodi.bin --standalone -fs -n --debug
start
cont
(now ignore displayed errror)
cont
(not xbmc should be loaded)
```

start the vpn. as soon as xbmc crashes, the gdb will return to cmd line notifying you about process crash. type
```
where
```

send the output. this could be enough for me to identify the problem. 

(if not (standard packages have no debug symbols included) I will send you link to xbian-package-xbmc including debugging symbols. then you just do the same.)

mk01, 2015-10-01:
@farthinder 

this should be Fixed now