xbianonpi/xbian #744: Kodi missing libass dependency

hackedbellini, 2015-06-28:
Was going to watch a mkv with embedded subtitles on kodi, but those subtitles didn't show up.

After searching a while, I found [this old issue](https://github.com/xbianonpi/xbian/issues/91). Based on that, I installed the `libass5` package, which solved the problem for me.

Probably would be useful to have it as a dependency for kodi package.

CurlyMoo, 2015-06-28:
I think you are one of the best issue detector for XBian that is not a team member :+1:  So i was wondering if you had any programming skills in general that would make it usefull for us if you could apply these changes directly?

mkreisl, 2015-06-28:
@hackedbellini 
I have no problems showing ssa/ass subtitles with libass4. Only Kodi 15 requires a slightly newer version of libass4 (0.10.1 instead of 0.10.0)
kodi is still linked against libass4 and has already a dependency for this lib.
Which version of Kodi do you use? I assume you're using Jessie, not Wheezy. On Jessie there doesn't exist libass4, only libass5. So this could be a explanation

hackedbellini, 2015-06-29:
@CurlyMoo as a matter of fact, I do have programming skills =P. Programming is what I do for a living.

That is actually one of the reasons I always try to report an issue every time I find some problem, even if it turns out not to be one. I know how hard it is to get feedback (specially good ones) from general users.

And actually, contributing to xbian is something I want to do, but sometimes I lack the time to do so.

I would easily open a PR for this one, but after reading what @mkreisl  said, it seems that it is not as simple as adding `libass5` to `Depends` on the debian control file.

And @mkreisl , yes, I'm running xbian on Jessie. I'm updated on the latest stable+staging packages (for kodi it would be `14.3~IsengardBackports-1434302336`).

Knowing that jessie doesn't have `libass4` on its repositories, what do you think would be the best solution here? Maybe porting it to xbian repository?

CurlyMoo, 2015-06-29:
Tell me how you would like to contribute? What are your programming skills?

mkreisl, 2015-06-29:
@hackedbellini 
> Knowing that jessie doesn't have libass4 on its repositories, what do you think would be the best solution here? Maybe porting it to xbian repository?

I thought about this yesterday for a while, founding no good solution. I think we should keep this in mind. I looked to the config for IMX6, which also uses Jessie. And there is no dependency for libass[4|5]. I will add it there. Backporting libass4 is no good idea, because libass5 already exists and works well.

The best would be making tabula rasa, killing Wheezy and switching completely to Jessie :innocent:

CurlyMoo, 2015-06-29:
> The best would be making tabula rasa, killing Wheezy and switching completely to Jessie :innocent:

Already taking quite long...

hackedbellini, 2015-06-30:
@mkreisl it make sense. Even the [xbmc-bin package in debian jessie](https://packages.debian.org/jessie/xbmc-bin) depends on `libass5`.

@CurlyMoo 

> Tell me how you would like to contribute? What are your programming skills?

I haven't though to much about "how I want to contribute" to it yet. Probably in any area that I can be useful =P

Regarding my programming skills, specially regarding things that would be useful for xbian, I'm mostly experienced in python. More than 5 years working with python projects, as you can see in [my openhub page](https://www.openhub.net/accounts/hackedbellini) and on the projects I contributed on github. I also have good scripting knowledge (sh/bash scripting in general) and debian packaging (e.g. I was the maintainer of [this ppa](https://launchpad.net/~stoq-dev/+archive/ubuntu/lancamentos) the company I work for uses to release the software we developthere). I also have good linux skills since I'm a linux desktop user since 2006 (my laptop runs Debian Testing since squeeze =P).

Other than that, I have a good experience with gtk+, but that doesn't add too much for xbian as far as I can tell.

Because I'm still finishing my graduation while I work, sometimes I lack spare time to contribute (the reason to why I usually only open an issue for the problems I found). But having some, I'll be more than happy to contribute to xbian :). If you have any suggestions on what/where I can be useful, I'll do my best to help!

CurlyMoo, 2015-06-30:
> I'm mostly experienced in python.

WTF! We have an `job application` open for some time now to maintain the `xbian-config-xbmc` package because the old developer couldn't do it anymore. And all that time we had that developer amongst us...

> If you have any suggestions on what/where I can be useful, I'll do my best to help!

You'll will see soon enough where we want your help :innocent:

hackedbellini, 2015-06-30:
> WTF! We have an job application open for some time now to maintain the xbian-config-xbmc package because the old developer couldn't do it anymore. And all that time we had that developer amongst us...

hahaha sorry for not stepping in earlier =P

I can surely help maintaining that package :)

> You'll will see soon enough where we want your help :innocent: (always good to learn new smiley's)

Looking forward it! :)

Next week is my last week of class on the university for this semester. After that, I'll start having some time to contribute here!

> It's also usefull if we can meet soon on IRC. Tell me when you have time.

I'm almost always online on irc because of ircclould. I'm already in `#xbian@freenode` there, is that the right channel? My nick there is the same as my nick here on github, `hackedbellini`. I don't know yours so can you ping me there? =P

mk01, 2015-07-08:
@hackedbellini 

kodi is linked against the libass which is available at its compilation. rpi2 is compiled agains Jessie and therefore will work with libass5. rpi2 is compiled against wheezy and therefor expects libass4.

mk01, 2015-07-08:
so yes, feel free to open a PR. best would be if you check the other branches too (imx6/jessie so libass5, etc) and update the dependencies accordingly (where ever is libassX missing)

closed 2015-09-14 by mkreisl