xbianonpi/xbian #745: something wrong when kodi is idle

fangdingjun, 2015-06-29:
I use  the lastest xbian disk image(2015-06-27)  from sourceforge
on my raspberrypi model B+

there is some thing wrong when kodi is idle,  this is my step

1. start video playback, ssh to xbian and use top to monitor system status

     now, cpu usage is 20%~30% and the load average < 1.5
2. after 2 hours video playback, the load average is still < 1.5
now, stop the video playback, I found that  the load averate increases slowly,  but the cpu usage is still 2%~5%

     after 2 hours idle, the cpu load average up to 30, but the cpu usage is still 2%~5%

3. kill the kodi.bin, after kodi restarted,  load average back to 2.0 and increase slowly, but the cpu usage is still 2%~5%

[image]

fangdingjun, 2015-06-30:
I think the problem is caused by screensaver.
I change the screensaver from dim to none, the problem not happened again. the cpu load average stay at 0.3.

mk01, 2015-07-08:
@fangdingjun 

the reason for this is a miss-calculation of load during the system being really idling. I know it might sound strange, but ... 

what is happening is that when screensaver kicks in, tv is turned off or switched to another source, XBian shutdowns Kodi's renderer completely. When this happens kernels switches to different mode of waking itself up (like you would make your heart to beat at minimum rate possible). also the governor driving CPUs frequency is changed from performance to a conservative - trying to keep freq at min.

you can see that also in your top output where kodi.bin load goes under 2%. with all this being at relaxed state and kernel not even "ticking" regularly, the load calculation of moving average probably thinks that system is TOO busy not able to catch events and regular ticks etc. 

beside the nonsense load average calculation it has no other effect (so the system is really relaxed, having minimum power consumption - or enough cpu time free for other processes - if you would have any started).

it was a common issue with older kernels (like before 3.12), then it was resolved and now back again. what kernel you are running ? 3.18, 19 or 4.1 ?

(oh just to finish why turning off screensaver provides "solution" - it's because Kodi never enters idle state, Xbian doesn't shutdown gfx rendering and system never gets relaxed).

fangdingjun, 2015-07-09:
Thanks for you professional explain. I got it.

The xbian is stable release,
xbian-package-kernel              3.18.8+-1425523930
version is 3.18.8+, compiled at  Thu Mar 5 03:43:34 CET 2015.

The Raspmbc I used before, the kernel  3.12, has no this issue.

mk01, 2015-07-09:
@fangdingjun 

yes, 3.12 was fixed, that's why I said before.
try 4.1...

this is from RPI with 4.1 - kodi is sleeping since Tue maybe.
```
root@rpi ~ # uptime
 09:21:01 up 19 days,  8:14,  2 users,  load average: 1.03, 1.81, 1.81
```

this is from RPI2 with same kernel
```
root@rpi2 ~ # uptime
 09:22:01 up 22 days,  1:24,  1 user,  load average: 0.04, 0.03, 0.05
```

RPI2 has 4 cores where one core is at all times providing constant tick. that's why the moving average of 1m, 5m and 15m loads looks better.
Point being - although it is making one asking question what is wrong - all is fine.

Real load above 5-7 would mean on RPI unresponsive system, heating etc. If at any time you would doubt, check the low level hw info with 
```
sudo -s
xbian-stats
```

btw: I remember those 20+ nonsense loads from numbers of 3.18 too.

fangdingjun, 2015-07-09:
@mk01 
Could you give me a guide to upgrade the kernel to 4.1?

I try to change the source from stable to  stagging and devel, but there is no newer kernel.
```
deb mirror://apt.xbian.org/mirror.txt      stable  main rpi-wheezy
```

fangdingjun, 2015-07-13:
I have upgaded the kernel to 4.1 to use the follow commands
```bash
apt-get install xbian-update
apt-get upgrade
```

Set the screensaver to dim and turn off the TV,  the system load average is still high, but the cpu usage is less than 10%.

The follow is my environment.
top:
```
top - 16:26:02 up  7:28,  1 user,  load average: 8.31, 8.94, 8.13
Tasks:  95 total,   1 running,  94 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.7 us,  0.3 sy,  6.3 ni, 91.0 id,  0.0 wa,  1.7 hi,  0.0 si,  0.0 st
KiB Mem:    365784 total,   236560 used,   129224 free,      748 buffers
KiB Swap:   250876 total,        0 used,   250876 free,   147072 cached

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND
 1172 xbian     44   0  632m  90m  51m S   6.6 25.3  47:28.65 kodi.bin
17335 xbian      6   0  4632 2400 2016 R   1.3  0.7   0:00.26 top
 1579 ntp        1   0  5480 3304 2932 S   0.3  0.9   0:12.54 ntpd
13026 xbian      1   0  9308 3504 2924 S   0.3  1.0   0:00.65 sshd
```

xbian-statis:
```
Governor: performance
  Memory: 512MB (368MB ARM, 144MB GPU)
Min Freq:  700Mhz |  250Mhz |  400Mhz
Max Freq:  840Mhz |  320Mhz |  400Mhz
Voltages:       0, 1.200    |  0, 1.200
   Other: TEMP_LIMIT=85C
 Version: Jun 10 2015 20:40:12, version a0f9db571a2bb980469502bd4de2c9793b3223c4 (clean) (release)
vcg path: /opt/vc/bin/vcgencmd
  Codecs: H264 VP8 VORBIS MJPG
  Booted: Mon Jul 13 08:57:25 CST 2015

Time          ARM     Core     h264  Core Temp (Max)   IRQ/s      RX B/s      TX B/s
========  =======  =======  =======  ===============  ======  ==========  ==========
16:28:19   840Mhz   320Mhz     0Mhz  50.84C (50.84C)     290         628       4,230
16:28:21   840Mhz   320Mhz     0Mhz  49.77C (50.84C)     261          74         419
16:28:23   839Mhz   320Mhz     0Mhz  50.31C (50.84C)     263         240         591
16:28:26   840Mhz   320Mhz     0Mhz  49.77C (50.84C)     261          74         448
16:28:28   840Mhz   320Mhz     0Mhz  49.77C (50.84C)     260          74         261
16:28:30   840Mhz   320Mhz     0Mhz  49.77C (50.84C)     262         101         405
16:28:32   840Mhz   320Mhz     0Mhz  49.23C (50.84C)     260          74         340
16:28:34   840Mhz   320Mhz     0Mhz  49.77C (50.84C)     263         234         550
```

kernel version:
```Linux xbian 4.1.0-rc8+ #10 PREEMPT Sat Jun 20 00:44:00 CEST 2015 armv6l GNU/Linux```

Kodi version:
```xbian-package-xbmc                14.2-1431026245```

/boot/config.txt:
```
#initramfs initramfs.gz  0x1000000
gpu_mem_1024=256
gpu_mem_512=144
gpu_mem_256=100
initial_turbo=3
hdmi_ignore_cec_init=1
disable_overscan=1
disable_splash=1
arm_freq=840
core_freq=320
sdram_freq=400
```

/boot/cmdline.txt:
```
telnet zswap.enabled=1 zswap.compressor=lz4 console=tty1 root=/dev/mmcblk0p2 rootflags=subvol=root/@,autodefrag,compress=lz4 rootfstype=btrfs rootwait logo.nologo quiet noswap loglevel=0 partswap startevent=mountall selinux=0 splash nohdparm --startup-event mountall
```

closed 2016-06-17 by mkreisl