xbianonpi/xbian #746: Menu only in 60Hz refresh rate. Cannot set 50Hz refresh for kodi menu....

coderpop, 2015-07-14:
Hi,

With Pi 2 I have a problem. It keeps defaulting the kodi menu display to 60Hz and I need 50Hz for PAL playback.

Kodi/Xbian switches fine to 50hz when a video is played, but the projector has to re-sync as the video starts causing a blanking of the display for a time while the pi goes from 60Hz menu resolution to 50Hz playback resolution. It also happens when the video stops and the display goes back to 60hz for the menu to display.

I have tried setting 50Hz from within Kodi from system>settings>system>video preferences, but when I select 50Hz it changes the refresh rate from 60Hz to 50Hz, then flashes the menu page briefly and then reverts back to 60Hz! - It just won't stick on 50Hz no matter what I do.

I have tried hdmi_mode=31 in config.txt with no success.

I then added the following to the advanced user xml file in /home/xbian/.kodi/userdata, again no luck! 

<override>
<fpsmin>23.96</fpsmin>
<fpsmax>50.01</fpsmax>
<refreshmin>49.0</refreshmin>
<refreshmax>51.0</refreshmax>
</override>

<fallback>
<refresh>50.0</refresh>
</fallback>

<fallback>
<refreshmin>49.0</refreshmin>
<refreshmax>51.0</refreshmax>
</fallback>

If I go to system>settings>video>playback> and make sure that adjust display rate to match video is set to OFF the blanking period is no longer an issue, but it is not the fix as now the 50Hz videos are being played at 60Hz.

So how does one get 50Hz Kodi menu screen refresh?

CP.

mkreisl, 2015-07-14:
> I have tried setting 50Hz from within Kodi from system>settings>system>video preferences, but when I select 50Hz it changes the refresh rate from 60Hz to 50Hz, then flashes the menu page briefly and then reverts back to 60Hz! - It just won't stick on 50Hz no matter what I do.

Maybe the resync of your display takes too long after a refresh switch, so you don't see the dialog box which asked you if this setting is ok or not. And if you don't click `yes` Kodi will return to the old refresh rate immediately. 

> Kodi/Xbian switches fine to 50hz when a video is played, but the projector has to re-sync as the video starts causing a blanking of the display for a time while the pi goes from 60Hz menu resolution to 50Hz playback

You can set a delay Kodi paused playback after a refresh switch. You can set this time higher than your display needs for resync.

coderpop, 2015-07-14:
Hello,

Thanks for putting me on the right track. I *did* see the dialog box, but for less than a second and could not do anything with it.
The delay time has a maximum value of 20S, it will not go any higher, so that didn't work either.
I then tried another display (not projector but monitor this time) however that would not allow a 50Hz refresh at any resolution.
In the end I got it by going back to the projector and plugging in a USB keyboard and while the screen was blank (re-syncing) I hit left arrow and enter.
Now it is how I need it!

Many thanks

CP.

closed 2015-07-14 by coderpop

mkreisl, 2015-07-14:
> In the end I got it by going back to the projector and plugging in a USB keyboard and while the screen was blank (re-syncing) I hit left arrow and enter.

That's what I always have to do, only difference is using the CEC remotecontrol instead of a keyboard.