xbianonpi/xbian #747: Curl error every 10 minutes

Smultie, 2015-07-18:
This shows up in my kodi.log every 10 minutes:

17:29:02 T:1524626352   ERROR: CCurlFile::FillBuffer - Failed: Timeout was reached(28)
17:29:02 T:1524626352   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
17:29:02 T:1524626352   ERROR: CCurlFile::Open failed with code -1 for http://addons.serveftp.net/addons.xml.md5
17:29:02 T:1524626352   ERROR: Open - failed to open source <http://addons.serveftp.net/addons.xml.md5>
17:29:02 T:1524626352   ERROR: Failed to fetch checksum for directory listing http://addons.serveftp.net/addons.xml for repository repo.xbian.xbmc.
17:39:02 T:1524626352   ERROR: CCurlFile::FillBuffer - Failed: Timeout was reached(28)
17:39:02 T:1524626352   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
17:39:02 T:1524626352   ERROR: CCurlFile::Open failed with code -1 for http://addons.serveftp.net/addons.xml.md5
17:39:02 T:1524626352   ERROR: Open - failed to open source <http://addons.serveftp.net/addons.xml.md5>
17:39:02 T:1524626352   ERROR: Failed to fetch checksum for directory listing http://addons.serveftp.net/addons.xml for repository repo.xbian.xbmc.
17:49:02 T:1524626352   ERROR: CCurlFile::FillBuffer - Failed: Timeout was reached(28)
17:49:02 T:1524626352   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
17:49:02 T:1524626352   ERROR: CCurlFile::Open failed with code -1 for http://addons.serveftp.net/addons.xml.md5
17:49:02 T:1524626352   ERROR: Open - failed to open source <http://addons.serveftp.net/addons.xml.md5>
17:49:02 T:1524626352   ERROR: Failed to fetch checksum for directory listing http://addons.serveftp.net/addons.xml for repository repo.xbian.xbmc.
17:59:03 T:1524626352   ERROR: CCurlFile::FillBuffer - Failed: Timeout was reached(28)
17:59:03 T:1524626352   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
17:59:03 T:1524626352   ERROR: CCurlFile::Open failed with code -1 for http://addons.serveftp.net/addons.xml.md5
17:59:03 T:1524626352   ERROR: Open - failed to open source <http://addons.serveftp.net/addons.xml.md5>
17:59:03 T:1524626352   ERROR: Failed to fetch checksum for directory listing http://addons.serveftp.net/addons.xml for repository repo.xbian.xbmc.
18:09:03 T:1524626352   ERROR: CCurlFile::FillBuffer - Failed: Timeout was reached(28)
18:09:03 T:1524626352   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
18:09:03 T:1524626352   ERROR: CCurlFile::Open failed with code -1 for http://addons.serveftp.net/addons.xml.md5
18:09:03 T:1524626352   ERROR: Open - failed to open source <http://addons.serveftp.net/addons.xml.md5>
18:09:03 T:1524626352   ERROR: Failed to fetch checksum for directory listing http://addons.serveftp.net/addons.xml for repository repo.xbian.xbmc.
18:19:03 T:1524626352   ERROR: CCurlFile::FillBuffer - Failed: Timeout was reached(28)
18:19:03 T:1524626352   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
18:19:03 T:1524626352   ERROR: CCurlFile::Open failed with code -1 for http://addons.serveftp.net/addons.xml.md5
18:19:03 T:1524626352   ERROR: Open - failed to open source <http://addons.serveftp.net/addons.xml.md5>
18:19:03 T:1524626352   ERROR: Failed to fetch checksum for directory listing http://addons.serveftp.net/addons.xml for repository repo.xbian.xbmc.

Seems to be related to xbian's repo. Server offline it looks like. Any way to fix this?

CurlyMoo, 2015-07-18:
Can you please format the issue using `github markdown`?

Smultie, 2015-07-18:
Like that? If not; just tell me what you want to see....

CurlyMoo, 2015-07-18:
Makes all the difference :+1:

er0l, 2015-08-27:
I can confirm the same issue using XBian 20150817-0 on rpi2. it seems this host is down?

closed 2015-11-09 by mkreisl