xbianonpi/xbian #748: Patch to support netconsole

vondrt4, 2015-07-21:
I'm having some hangs on my Pi with a DVB-T tuner attached. It also may be due to a bad sdcard. The result is that the Pi responds to pings, but doesn't start a SSH server successfully. It also stops logging over NFS and the console is dead.

In this case, the netconsole would surely show me a kernel panic or some other useful message.
But the netconsole doesn't work with USB network cards.
The people here have done a patch to allow it:
http://forum.odroid.com/viewtopic.php?f=82&t=5077
Could you please apply it on the Xbian kernel?

mkreisl, 2015-07-21:
I'm not sure if this is a good idea.

I looked a little bit around in the kernel sources, none of the usb network driver (wired and wireless) does have enabled this function. There must be a serious reason for that.

CurlyMoo, 2015-07-21:
We can always try compiling it as a module to verify.

mkreisl, 2015-07-22:
Ok, you convinced me :wink:
I'll put this patch in
@vondrt4 
Which kind of hardware do you have, RPI1 or RPi2?

vondrt4, 2015-07-22:
I have an original RPI1B.

mkreisl, 2015-07-24:
I built some special kernels with enabled netconsole over smsc95xx. They can download it from here http://apt.xbian.org/pool/devel/rpi-wheezy/x/xbian-package-kernel/ for RPi1 and from here http://apt.xbian.org/pool/devel/rpi2-jessie/l/linux-image-bcm2836/ for RPi2.
I already tested it for RPi1, it seems to work well.

vondrt4, 2015-08-03:
Thanks a lot, it works. I hope it doesn't break anything, but judging by the description of the Netpoll API: "The netpoll API allows use of network devices in critical contexts via hardware polling. It provides a common interface for facilities such as netconsole, kgdb-over-ethernet, and network crash dumps," it won't be used by anything, if the netconsole is not enabled.

I'm now trying if it helps debug my problem.
Meanwhile, I tried to use the hardware watchdog and the watchdog daemon, as per https://www.domoticz.com/wiki/Setting_up_the_raspberry_pi_watchdog, and, interestingly, the RPi gets restarted after a few days and the DVB-T tuner doesn't register again on the USB bus. It does come back if I soft reset it before it fails, or if I power cycle it. So I think I have my cuprit already. It is an AF9015 stick, Not Only TV brand.
If I was a badly implemented USB slave controller on an USB host which sometimes loses USB isochronous frames, I think I'd do exactly this ;->. It also demonstrates by artifacts in the MPEG stream, particularly if the SDcard is in use (->higher interrupt latency).

CurlyMoo, 2015-08-03:
Do please change the title if we are going to debug not the patch but the system crashes.

vondrt4, 2015-08-03:
Not necessary. The patch should be a priority here. I'd like to keep it in the kernel for posterity :-).

closed 2017-01-27 by mkreisl