xbianonpi/xbian #749: No network since kernel update

RolfTimmerman, 2015-07-22:
I've updated my RPi1 model B today and the kernel was updated (to 4.1.2+) but now I don't have a network connection anymore. I've tried static as well as DHCP.

er0l, 2015-07-30:
Something similar on RPi2, network is up and running while booting and in about one minute it stops working. 

In the time network is up I can SSH and do something, but as soon as Kodi is fully operational network goes down. 

I have connected via serial port and restarted network by hand and it works, but on next reboot the same happens.

In the logs except this message, nothing else is worth mentioning.

[Wed Jul 29 23:56:22 2015] init: network-interface (eth0) pre-start process (361) terminated with status 124
[Wed Jul 29 23:56:22 2015] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup

Any ideas?

mk01, 2015-09-13:
@RolfTimmerman @er0l 

I assume this is no longer actual ?

er0l, 2015-09-13:
For me it's no longer actual since I moved to Jessie. No network issues so far. Thx

closed 2015-09-13 by CurlyMoo