xbianonpi/xbian #750: Latest updates break I2S support?

coderpop, 2015-07-23:
Update and upgrade (from SSH > xbian-config) on 22.7.15 has broken I2S dac audio.

There is no longer an option for alsa output and Kodi has reverted to HDMI out. It's not clear how this happened or why.

It happened on a Rpi and Rpi2, both now with no sound. Will investigate further if I get time.

coderpop, 2015-07-24:
And the culprit is......

xbian_package_kernel 4.1.2+

As soon as this is installed, I2S audio stops working.....

CurlyMoo, 2015-07-24:
Thanks for reporting.

mkreisl, 2015-07-24:
@coderpop 
Probably you have to enable first this kind of hardware in the Device Tree system
https://www.raspberrypi.org/documentation/configuration/device-tree.md
I don't have such hardware so unfortunately I can not test it

coderpop, 2015-07-24:
Device tree was working with kernel 3.18.8 but as soon as it was upgraded to 4.1.2 then it no longer works. I can't go back to the 3.18.8 as it is no longer available in the apt-get cache.

Why would this kernel upgrade remove device tree settings in config.txt anyway?

coderpop, 2015-07-24:
Update, the DT settings were all preserved in config.txt, so it's not that.

mkreisl, 2015-07-24:
IMHO DT wasn't enabled in 3.18.8 kernel per default, did you enable it manually?
Maybe the .dtb files are broken in 4.1, they come directly from the kernel package and not longer from the firmware package. That's all what I can say for the moment

coderpop, 2015-07-24:
It was enabled, see this link....

https://github.com/raspberrypi/firmware/commit/6efbc2e7811189ee497dc5d676819661031e11e3

All I had to do was add the dtoverlay for I2S and hifiberry in config.txt and make sure the modules were all present and loaded and it worked. This with pi and pi2 with different dacs.

Since I could not downgrade back to 3.18.8 and a btrfs rollback didn't solve the issue either, I have now re-flashed the card to it's previous working state.

mkreisl, 2015-07-24:
Yes,  that's right. But the rpi bootloader only recognizes the .dtb files if there is a specific trailer in the kernel image. And this trailer was not appended at the 3.18 kernel. It's not documented, but its necessary

coderpop, 2015-07-24:
You are correct. (and I have learned something new!)...

I just tried removing the dtoverlay info in config.txt and it all still works in kernel 3.18.8 - so it's not using DT parameters.

What I don't understand is that the DT info is in configtxt, but kernel 4.1.2 doesn't work with this.

mkreisl, 2015-07-24:
> What I don't understand is that the DT info is in configtxt, but kernel 4.1.2 doesn't work with this.

This is exactly the issue. I built some kernels which based (defconfig and source) from here https://github.com/raspberrypi/linux for another reason.
They can download it from here http://apt.xbian.org/pool/devel/rpi-wheezy/x/xbian-package-kernel/ for RPi1 and from here http://apt.xbian.org/pool/devel/rpi2-jessie/l/linux-image-bcm2836/ for RPi2.

You can see, a lot of commits there are DT patches, it seems that DT is a heavy WIP dev

It would be great if you could test these kernels to look if there is DT also broken or not

coderpop, 2015-07-24:
I would love to help out, really, but I don't know how to handle those files.

If someone can give clear instructions of what to do with the downloaded files, I will try my best to test and help out.

CP

mkreisl, 2015-07-24:
1 Download the .deb file you want to install (via browser, wget, curl or whatever), 
for example download `xbian-package-kernel_4.1.2+-1437702061_armhf.deb`
2 Install it with dpkg: `dpkg -i xbian-package-kernel_4.1.2+-1437702061_armhf.deb` 
After testing, if you want to reinstall the original kernel, you can do
3 the command `apt-get download xbian-package-kernel` and then
4 again `dpkg -i xbian-package-kernel......deb`

mkreisl, 2015-07-24:
I forgot as usual, installing requires rot privileges, so run `sudo dpkg -i ...` if you are no logged in as root

coderpop, 2015-07-25:
OK, here is the result....

Rpi = Success 
Rpi2 = Fail

Hope this helps. What was the issue on the Rpi? What did you do to fix it? 

CP.

mkreisl, 2015-07-25:
Thank you for testing
> Hope this helps. What was the issue on the Rpi? What did you do to fix it?

Not really. I did not fix anything. As I told, I only used the the source and kernel configuration from  here https://github.com/raspberrypi/linux instead of using the xbian source and config. I've really no explanation why it works on RPi1 and it fails on RPi2. The kernel source of RPi2 was 8 days older than RPi1, maybe this is the issue. I told you, DT is WIP

Did you add dtoverlay for I2S and hifiberry in config.txt on both RPi's? 

> Also, there is an option to switch between kernels in xbian-config. Is there a way to use this in testing? It would make things much quicker!

Yes, I know this feature, but I have no idea how to handle it.

coderpop, 2015-07-25:
Dtoverlay command is the same on both pi and pi2, so should work.
Added to both......

dtoverlay=hifiberry-dac
dtparam=i2s=on

I've no idea on the kernel switcher in xbian-config either. Maybe someone who does know will read this and help us out (re-flashing cards takes time!)...

mkreisl, 2015-07-25:
Ok .... 4.1.3 is out, I will generate new packages for Rpi1 and RPi2. This commit https://github.com/raspberrypi/linux/commit/7906fa935ee088391c6a3092374e477f51e799cd makes me optimstic that it could work with the new kernel. 
The new kernel packages should be available next day in the devel repo

coderpop, 2015-07-27:
I just noticed that 4.1.2 had been replaced with 4.1.0 RC8 on the devel repo, so I tried it.

It works with pi 2 !!!

Thanks for that :)

Hope this helps.

mkreisl, 2015-07-27:
Thanks for response, I assumed that you already used this kernel on RPi2 during your post this message:

> OK, here is the result....
> 
> Rpi = Success
> Rpi2 = Fail

Anyway, it's good hear this :smile:

Btw, I build a new 4.1.3 kernel for RPi2, but unfortunately this version doesn't work anymore - Kodi won't come up - now error message, no logfile - nothing :rage:

So I suspended working and testing on kernel 4.1.x for the moment

coderpop, 2015-07-27:
In the failed RPi2 test I used 4.1.2+ .... That was the one that didn't work at all with I2S dac. 

Today 4.1.0 RC8 does work, so that is something to help others who report the same problem.

Sad about 4.1.3 though, very frustrating. Still, a little time and who knows? ;)

mkreisl, 2015-07-28:
First good news: I have kernel 4.1.3 working with Kodi on a RPi1 :smile:
 4.1.3 requires the latest firmware from today, the version from sunday was too old :rage:

coderpop, 2015-07-29:
Heheheh. That sounds like great progress to me :)

When you have it for Rpi2 I can test for you if you like.

How did you update the firmware? With raspbian it's sudo rpi-update, but that doesn't work under xbian. Also, sudo apt-get install rpi-update also fails. ???

mkreisl, 2015-07-29:
> When you have it for Rpi2 I can test for you if you like.

Already done and uploaded to the devel repo
You can download it from here
RPi2 http://apt.xbian.org/pool/devel/rpi2-jessie/l/linux-image-bcm2836/
RPi1 http://apt.xbian.org/pool/devel/rpi-wheezy/x/xbian-package-kernel/
and the firmware package from here http://apt.xbian.org/pool/main/x/xbian-package-firmware/ [1]

> Rpi has alsa-utils and alsa-base installed. Pi2 does not have these packages installed (according to dpkg -l '*utils') yet it still works. Is there anything in this?

Don't know. Probably Kodi does not need this

[1] you have to wait until 20 clock until it appears

coderpop, 2015-07-29:
Installed firmware and then kernel and tested.....

Lovely job! ;)   Rpi2 now working with I2S dac on 4.1.3......

Will test on rpi later today or tomorrow.

Thank you.  :)

mkreisl, 2015-07-29:
Good to hear :smile:: thank you for response

Can you explain which lines vanished in /boot/config.txt

I'm sure that installing kernel or firmware does not touch config.txt

coderpop, 2015-07-30:
/boot/config.txt contains the lines....

dtoverlay=hifiberry-dac
dtparam=i2s=on

These were no longer present in the file after upgrading the kernel and firmware. I am guessing that somehow a default file is being installed/created. There was a similar problem with Xbian earlier this year when every upgrade removed lines added to /etc/fstab. It was eventually cures but I never found out what exactly was going on, but this seems very similar.

Note however that this is only happening on Pi2. So maybe it handles something differently.

Hope this helps.

mkreisl, 2015-07-30:
Hmmm, it is very strange. 

I looked into the kernel and firmware package, there is no code which could do this. There is only one line in firmware package which removes lines containing `emmc_pll_core` in config.txt (`sed -i '/emmc_pll_core/d' /boot/config.txt`). 

> removed lines added to /etc/fstab. It was eventually cures but I never found out what exactly was going on, but this seems very similar

This is done by `xbian-update`. /etc/fstab is more or less hardcoded. It also took me some time to figure out what's going on there ...

coderpop, 2015-07-30:
Strange indeed and I can't offer any other explanation.

The Rpi2 was working with 4.1.0 RC8 and playing music (this would have mandated DT entries in config .txt to be present and they were). I stopped the music, upgraded the firmware and then to 4.1.3 and rebooted. Nothing else. 

I'll see if it happens again or if anyone else posts a similar experience.

But thank you once again for getting this back working, it's great when it's playing nicely! ;)

mkreisl, 2015-07-30:
>  .txt to be present and they were). I stopped the music, upgraded the firmware and then to 4.1.3 and rebooted. Nothing else.

I have no doubt that you only did this. I will keep this in mind and put your dt lines into my config.txt before I install a new kernel

coderpop, 2015-07-30:
Hmm.... Now this is odd!!!

I just checked the Rpi2 from within kodi>system>system info>hardware and it shows CPU speed as 600MHz. But looking into overclocking in xbian-config it shows 900NHz as CPU freq.

Can you have a look and see if you are getting the same info please ?

mkreisl, 2015-07-30:
I remember you have disabled DPRIORITY - yes?

Try setting `CPUFREQ_GOVERNOR="ondemand"` or `CPUFREQ_GOVERNOR="performance"` in file `/etc/default/xbian-module-init-tools`

This kernel has `powersave` as default

coderpop, 2015-07-31:
Commenting and adding "ondemand" seems to have worked. It was not an issue with performance, just something I noticed and wondered about... 

Thanks again. I have learned some new things doing this and now have the best Xbian experience ever!

CurlyMoo, 2015-07-31:
We are still hoping to bundle all this information into a user manual just like you can find on http://manual.pilight.org, but sadly no XBian user had volunteered to start writing something.

Also see http://forum.xbian.org/thread-3008.html

coderpop, 2015-08-15:
There is some sort of bug that I have found.

When playing video the sound is perfect via the I2S dac. But pause the video for a long period and then select 'play' again and you get blasted with white noise from the speakers.

Stopping the video and re-starting clears the issue.

I have repeated this a few times now since it first happened. If it helps I can get logs from when it happens, just tell me what logs you want and I will attempt to post them soon.

mkreisl, 2015-08-16:
I think it is better to open a new issue.

What is a "long period"

Did you already  played with this option: `System-Settings->Audio output->Keep audio device alive`. Maybe this helps

closed 2016-06-17 by mkreisl