xbianonpi/xbian #751: SFTP not working (SFTPSession: Failed to connect 'kex error : ...)

knutshub, 2015-07-24:
Hi,
I'm trying to connect inside Kodi (Videos > Files > Add Videos > Browse > Add network location > Protocol: Secure shell (SSH/SFTP)) to a server using SFTP. But his doesn't work (HTTPS is working).

On my raspberry pi running xbian (Kodi 14.2 Git:Unknown (Compiled: May 7 2015))

    ssh -V : OpenSSH_6.0p1 Debian-4+deb7u2, OpenSSL 1.0.1e 11 Feb 2013
    openssl version : OpenSSL 1.0.1e 11 Feb 2013

On the server:
    
    ssh -V : OpenSSH_6.8p1, OpenSSL 1.0.1p 9 Jul 2015
    openssl version : OpenSSL 1.0.1p 9 Jul 2015

Here are the corresponding lines from kodi.log:

    08:19:53 T:2884342800   ERROR: SFTPSession: Failed to connect 'kex error : did not find one of algos diffie-hellman-group1-sha1 in list curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group14-sha1 for kex algos'
    08:19:53 T:2884342800   ERROR: SFTPSession: Not connected, can't list directory '/home/xxxxxxx/files/'
    08:19:53 T:3034685440   ERROR: GetDirectory - Error getting sftp://USERNAME:PASSWORD@xxxxxxx.whatbox.ca:22//home/xxxxxxx/files/
    08:19:53 T:3034685440   ERROR: CGUIMediaWindow::GetDirectory(sftp://USERNAME:PASSWORD@xxxxxxx.whatbox.ca:22//home/xxxxxxx/files/) failed

In IRC of whatbox, they said it's a problem with the older version of SSH in Kodi/XBMC. But I'm not sure how to proceed and upgrade that one.

Can someone clarify this, please? And does there is a workaround to solve this problem? Is it a good idea to upgrade OpenSSH manually?

Thanks for any help. 

Here is the corresponding topic in the xbian forum: http://forum.xbian.org/thread-3063.html

mkreisl, 2015-07-24:
I tested it by connecting to my homeserver (running debian wheezy). Had no problem, can play videos without any issues.

knutshub, 2015-07-24:
I did some more research on this topic. 

I found a post in a forum that seems related to this problem. Here someone suggested to add: 

    KexAlgorithms curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group14-sha1,diffie-hellman-group-exchange-sha1,diffie-hellman-group1-sha1

to 

    /etc/ssh/sshd_config

(Source: https://bbs.archlinux.org/viewtopic.php?id=189535 )

I couldn't test it yet. But I will reply here, if I checked it. But the error message sounds like the used algorithm isn't available/activated in this SSH version in xbian. 

But I don't know enough about key exchange algorithms to know if this addition will make it is insecure.

knutshub, 2015-07-24:
And this issue from OpenELEC seems also related and it was solved by upgrading libssh:
https://github.com/OpenELEC/OpenELEC.tv/issues/3587

knutshub, 2015-07-26:
No solution so far. Editing /etc/ssh/sshd_config doesn't help. No SSH login possible afterwords - had to restore a snapshot. I edited the post above.

How are the changes to use a more recent version of OpenSSH?

mkreisl, 2015-07-26:
> How are the chances to use a more recent version of OpenSSH?

< 0
Unfortunately you can't use openssh from wheezy backports for an RPi1, so I only see 2 solutions
1. Upgrade your RPi manually to Jessie (I already did this for testing, no problem) ...
2. Wait for an automatically upgrade to Jessie ...
and test it again
AFAIK  an account is needed on whatbox, so it is hard to test anything

knutshub, 2015-07-26:
>  1. Upgrade your RPi manually to Jessie (I already did this for testing, no problem) ...

OK, I will try this in the next days.

I just tested to use sftp on the command line and it works. How is this possible? Why it's not working inside Kodi?

mk01, 2015-10-01:
@knutshub 

is this an issue ?

tarasis, 2015-10-01:
I have just set this up and I am experiencing this issue (SFTP to a Arch server)

mk01, 2015-10-01:
@tarasis 

first of all check, that your ssh is something of actual version.
```
root@rpi2 ~ # ssh -V
OpenSSH_6.7p1 Debian-5, OpenSSL 1.0.1k 8 Jan 2015
```

JanPetterMG, 2015-12-07:
Any news on this issue at all?
I've had the exact same problem for months now. The server is an Debian 8.2 jessie server-edition, with multiple clients running on Windows, Debian, OpenELEC and Android.
Just tested ````Kodi 16 beta 3```` and the issue still exists.

My setup works perfectly with ````Kodi 14.2````, but any newer version won't work at all.
I just don't see any real alternatives at all. FTP unsecure, SMB / NFS local network only.

As far as I can see, this is a Kodi problem, not SSH server problem. ````diffie-hellman-group1-sha1 is weak and within theoretical range of the so-called Logjam attack````, so why has Kodi started using it then? Unsupported or disabled on most up-to-date servers...

SSH server, up to date (no newer version available for Debian jessie at least)
````
OpenSSH_6.7p1 Debian-5, OpenSSL 1.0.1k 8 Jan 2015
````
Kodi 16 beta 3
````
16:24:28 T:27960    INFO: SFTPSession: Creating new session on host 'HOST:51822' with user 'Kodi'
16:24:28 T:27960   ERROR: SFTPSession: Failed to connect 'kex error : did not find one of algos diffie-hellman-group1-sha1 in list curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group14-sha1 for kex algos'
16:24:28 T:27960   ERROR: SFTPSession: Not connected, can't list directory 'media/TV-shows/'
16:24:28 T:27960   ERROR: XFILE::CDirectory::GetDirectory - Error getting sftp://USERNAME:PASSWORD@HOST:51822/media/TV-shows/
````

mkreisl, 2015-12-07:
@JanPetterMG 
I checked it again in my environment with 16 b3, server Debian Jessie now, works perfectly.

Got this in my Kodi logs:
```
Dec  7 17:23:14 kmxbimx T:1953591296    INFO: SFTPSession: Creating new session on host 'kmcubie:22' with user 'manfred'
Dec  7 17:23:15 kmxbimx T:1953591296    INFO: SFTPSession: Server unkown, we trust it for now
```
I noticed the second line, this is missing in your logs. 
So my question: do you try to login via password or key - I'm using password, it is enabled here in sshd because I need this for using X2GO

JanPetterMG, 2015-12-07:
@mkreisl I'm using password.
I've tested 16 b3 in Windows 10 only, 15 has been tested on most devices, but didn't work...
I'm going to test 16 b3 on other devices too, because this is strange...

mkreisl, 2015-12-07:
So, it seems to be a general Kodi issue, not XBian.
Please open a Ticket there http://trac.kodi.tv/

mk01, 2015-12-11:
this even is not kodi, that is certificates / configuration at the server side. ....
I remember that from past, unfortunately do not remember more.

mkreisl, 2015-12-11:
@mk01 Yes, this could be. But _unfortunately you do not remember more_. My server configuration is default, never changed anything (as far as I remember)

puggan, 2015-12-25:
The error
"Failed to connect 'kex error : did not find one of algos diffie-hellman-group1-sha1 in list ..."
are related to an outdated version of libssh, acording to:
https://github.com/OpenELEC/OpenELEC.tv/issues/3587

mk01, 2016-01-12:
@mkreisl 

I don't remember more in the sense of specific Ciphers which has been disabled by default (in what ssh version). After little browsing:

```
For those using ssh over rsync or just scp to move files around on a LAN, be aware that a number of version 2 ciphers have been disabled in the 6.7p1-1 release of openssh (see release notes) including the following:

3des-cbc
blowfish-cbc
cast128-cbc
arcfour
arcfour128
arcfour256
aes128-cbc
aes192-cbc
aes256-cbc
rijndael-cbc@lysator.liu.se

That leaves the following available: 
aes128-ctr
aes192-ctr
aes256-ctr
aes128-gcm@openssh.com
aes256-gcm@openssh.com
chacha20-poly1305@openssh.com
```

easiest way is to put back those disabled by default now (by editing /etc/ssh/sshd_config) and putting
```
Ciphers 3des-cbc,blowfish-cbc,cast128-cbc,arcfour,arcfour128,arcfour256,aes128-cbc,aes192-cbc,aes256-cbc,rijndael-cbc@lysator.liu.se
```

if there is no control over sshd, then perhaps by putting similar line - but with the second group of cipher names into /etc/ssh/ssh_config. or creating ~/.ssh/config with specific host and cipher config like this:
```
Host ANYNAME
     Hostname ssh.server.net
     Cipher aes256-ctr
```

this should take effect for sftp/ssh sessions opened from within xbmc too.

mk01, 2016-01-12:
anyhow, the whole problem can be the other way around - meaning that server is forcing one of the older ciphers/keyexch algorithms and local system (kodi/ssh/xbian/whatever) is refusing to use it for communication.

reverting to the short copy&paste log above, client logs kex error what is keyexchange alg problem. in that specific case would be needed:
```
ssh -Q kex
```
copy the list, remove from it the one obsolete, edit ssh_config by putting
```
KexAlgorithms diffie-hellman-group-exchange-sha256,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,curve25519-sha256@libssh.org
```