xbianonpi/xbian #753: Raspberry Pi Jessie switch

CurlyMoo, 2015-08-18:
I've asked this nummerous times, but let's try again here. Why haven't we switched to Jessie yet? I see all kinds of commits from various developers, but none of them relate to Jessie. As far as i know, switching should be easy and should also be the highest priority atm.

So, what's up with the whole Jessie switch:
1. Are we going to enforce users to switch through our updates?
2. Are we going to provide Jessie in the weekly images and not as an upgrade path?

In either case, can at least somebody tell me what the status of this is and the estimated roadmap. I especially addressing @mk01.

The-Exnor, 2015-08-18:
I surely hope is via updates.... i hate to have to re-configure my systems (already have a lot of that at work :P)

mkreisl, 2015-08-18:
@CurlyMoo 
I'm working on it since two weeks. But unfortunately, I didn't find a good way to switch from Wheezy to Jessie automatically via xbian-update without ugly hooks (dpkg can not called recursively)
Last weekend I started building images for RPi1 and RPi2 with Jessie and Kodi 15.1rc1. 
I will have finished it until friday, so RPi Images with Jessie/Kodi15.1rc1 should be available at Saturday

CurlyMoo, 2015-08-18:
Can you describe the workflow you are trying to achieve?

mkreisl, 2015-08-18:
First of all, I wrote a script which does the distribution upgrade which I started manually. This works now, I called it /usr/local/sbin/xbian-distupgrade and is part of xbian-update. 
Next, I tried to call this script via xbian-update postinst procedure, I only found one way to get it work is to make an ugly hook, expanding the already existing apt-get hook. But the disadvantage of this hook is that you have to relogin to activate this hook
```
diff --git a/content-tpl/DEBIAN/postinst b/content-tpl/DEBIAN/postinst
index a16e9ed..8364758 100755
--- a/content-tpl/DEBIAN/postinst
+++ b/content-tpl/DEBIAN/postinst
@@ -925,6 +925,11 @@ that risk?\n \
                 fi
         else
                 command apt-get "$@"
+                if [ -e /run/apt-get-hook ]; then
+                    . /run/apt-get-hook
+                    rm -f /run/apt-get-hook
+                    [ -x $APT_POST_HOOK ] && $APT_POST_HOOK
+                fi
         fi
 }
 EOF
@@ -945,6 +950,13 @@ EOF
                fi
        fi
 
+        APT_POST_HOOK="/usr/local/sbin/xbian-distupgrade"
+        if [ -x $APT_POST_HOOK ]; then
+            if $APT_POST_HOOK test > /dev/null; then
+                echo "APT_POST_HOOK=$APT_POST_HOOK" > /run/apt-get-hook
+            fi
+        fi
+
        if ! status -q xbian-update-reboot; then
 cat << \EOF > /tmp/xbian-update-reboot
 #!/bin/sh
```
I also tried to make it work vie `Dpkg::Post-Invoke` without success

CurlyMoo, 2015-08-18:
What is in `/usr/local/sbin/xbian-distupgrade`?

mkreisl, 2015-08-18:
```
#!/bin/bash

LOG="/tmp/xbian-distupgrade.log"

# Step 1: Check version
grep -q BCM2708 /proc/cpuinfo 2>/dev/null && config_platform=rpi
grep -q BCM2836 /proc/cpuinfo 2>/dev/null && config_platform=rpi
grep -q BCM2709 /proc/cpuinfo 2>/dev/null && config_platform=rpi2
grep -q "i.MX6" /proc/cpuinfo 2>/dev/null && config_platform=imx6
grep -q sun7i   /proc/cpuinfo 2>/dev/null && config_platform=bpi
                    
version="";
if which lsb_release >/dev/null; then
    version=$(lsb_release -c -s);
elif [ -f /etc/debian_version ]; then
    grep -q "7\..*" /etc/debian_version && version="wheezy"
    grep -q "8\..*" /etc/debian_version && version="jessie"
else
    echo "Could not determine running OS version"
    exit 1
fi

echo "platform=$config_platform, version=$version"

if [ "$version" = jessie ]; then
    echo "Already on Jessie"
    exit 1
fi

if [ "$1" = test ]; then
    exit 0
fi

# Step 2: Modify /etc/apt/sources.list ...
FILES="/etc/apt/sources.list /etc/apt/sources.list.d/xbian.list"
for f in $FILES; do
    [ -e $f.dpkg-$version ] || cp -a $f $f.dpkg-$version
    sed -i s/wheezy/jessie/g $f
    if [ ! "$config_platform" = rpi ]; then
        sed -i s%deb.*raspbian.*%deb\ http:\/\/http.debian.net\/debian\/\ jessie\ main\ contrib\ non-free%g $f
    fi
done

# Step 3: Pinning of systemd-sysv
if ! cat /etc/apt/preferences.d/xbian | grep -q systemd; then
    echo "Pinning of systemd-sysv"
    echo ""                      >> /etc/apt/preferences.d/xbian
    echo "Package: systemd-sysv" >> /etc/apt/preferences.d/xbian
    echo "Pin: release o=Debian" >> /etc/apt/preferences.d/xbian
    echo "Pin-Priority: -1"      >> /etc/apt/preferences.d/xbian
fi

#echo "XBian Distribution Upgrade"
#exit 0

# Step 4: Check if enough space is available
av=$(df -BM /var/cache/apt | grep -m1 / | awk '{ printf "%d", $4 }')
apt-get update 2>&1 | tee $LOG 
nd=50
for i in $(LC_ALL=C apt-get -o APT::Get::Trivial-Only=true dist-upgrade | grep " MB" | awk '{ printf "%d ", $4 }'); do
    nd=$((nd + i))
done

echo "available=$av needed=$nd"

if (( $nd > $av )) ; then
    echo "Not enough space available"
    exit 2
fi

# Step 5 deinstall packages which breaks upgrade
REINST=""
if dpkg -l | grep -q "\ watchdog\ "; then
    echo "Uninstalling watchdog package"
    #REINST+="watchdog" ### watchdog omitted, does not install w/o error
    dpkg -r watchdog 2>&1 | tee -a $LOG
fi 

# Step 6 do download and upgrade
DEBIAN_FRONTEND=noninteractive apt-get -y -o Dpkg::Options::="--force-confdef" -o Dpkg::Options::="--force-confold" dist-upgrade 2>&1 | tee -a $LOG

# Step 7 install previous uninstalled packages
for p in $REINST; do
    echo "Reinstalling $p package"
    apt-get install -y $p 2>&1 | tee -a $LOG
done

# Step 8 update initramfs if necessary
if -e /boot/initramfs.gz; then
    echo "Update initramfs"
    /etc/xbian-initramfs/update-initramfs.sh > $LOG 2>&1
fi

sync
```

CurlyMoo, 2015-08-18:
You're thinking way too difficult. The steps are:
1. Install the Jessie sources through `xbian-package-sources`.
2. Make sure that all packages are properly set on hold, or prioritized for XBian (e.g. watchdog).

3. As soon as apt is updated an xbian-update should be in place on the XBian Jessie sources that finetunes the last few parts.
4. No more updates should be done on the wheezy sources.

The most complex part is step 2. Making sure everything is set right in `/etc/apt/preferences.d/xbian`.

mkreisl, 2015-08-18:
Seems that I don't have the knowledge base for doing this
What do you mean with `xbian-package-sources.` Can't find a package with this name

CurlyMoo, 2015-08-18:
> Seems that I don't have the knowledge base for doing this

That's also why i addressed @mk01 

> What do you mean with xbian-package-sources. Can't find a package with this name

I meant this one `https://github.com/xbianonpi/xbian-package-repo`

mkreisl, 2015-08-18:
Yes, ok. I assumed that you meant this repo and yes, I know that @mk01 knows all this stuff better than I do.

airend, 2015-08-20:
I've been looking forward to jessie-based images :+1:  Is the August 20th RPi2 image (in http://sourceforge.net/projects/xbian/files/release) built with the jessie, 15.1rc, 4.1.5, etc?

mkreisl, 2015-08-20:
yes

mkreisl, 2015-08-20:
@CurlyMoo @mk01 
Building of Jessie images for RPI's is now working :smile: 

If no one has anything about it, I will now going to prepare our build server (vps01) for (auto?)building of the new packages (xbian-package-kernel, xbian-package-cec, xbian-package-xbmc)

airend, 2015-08-20:
Has anyone else tested `XBian_2015.08.20_rpi2.img`? It doesn't seem to boot properly off a completely clean install. My image MD5 sum is `4e74e5c80f676ea88648b34ab527b435`, just in case.

mkreisl, 2015-08-20:
Yes I did. Just done again. No issue

airend, 2015-08-20:
Very odd, maybe it's another CEC problem. Can you confirm the image MD5 hash?

mkreisl, 2015-08-20:
Yes, is the same
What exactly happens?

airend, 2015-08-20:
Basically, no video output, auto-switching to the Kodi source, or CEC entry to be found. The LEDs do indicate booting, reading media, etc, but have no other visual indication. Since network is not yet configured (relying on WLAN), don't really have any mean to debug things, short of serial console.

mkreisl, 2015-08-20:
So, in summary: nothing works. Very strange
Your sd-card is ok?

airend, 2015-08-20:
I think so… Actually, I suspect most things work based on LED patterns, but the CEC connection is broken somehow, so my TV is never instructed to switch to the proper source. I will re-route my HDMI connections, to force specific connections.

mkreisl, 2015-08-24:
> If no one has anything about it, I will now going to prepare our build server (vps01) for (auto?)building of the new packages (xbian-package-kernel, xbian-package-cec, xbian-package-xbmc)

Short report about current state:

- [x] Building of images:
  - [x] RPi1: Jessie and Kodi 15.1 - build and tested -- Ok
  - [x] RPi2: Jessie and Kodi 15.1 - build and tested -- Ok
  - [ ] iMX6: Jessie and Kodi 15.1 rc1 - build successfully but untested

- [x] Building of xbian-package-kernel on buildserver (vps01):
  - [x] RPi1: build and tested -- Ok
  - [x] RPi2: build and tested -- Ok
  - [ ] iMX6: not done

- [x] Building of xbian-package-cec on buildserver (vps01/Jenkins)
  - [x] RPi1: cec3/Jessie - build and tested -- Ok
  - [x] RPi2: cec3/Jessie - build and tested -- Ok
  - [ ] iMX6: cec3/Jessie - build successfully but untested

- [x] Building of xbian-package-xbmc on buildserver (vps01/Jenkins)
  - [x] RPi: Kodi 15.1/Jessie - build and tested -- Ok
  - [x] RPi2: Kodi 15.1/Jessie - build and tested -- Ok
  - [ ] iMX6: Kodi 15./Jessie - build successfully but untested (put to staging)

(more packages will follow)

mkreisl, 2015-08-24:
@CurlyMoo 
Don't have write access on xbian-package-tvheadend. Could you please correct this - thanks

CurlyMoo, 2015-09-12:
The updates to xbian-package-repo and xbian-update to allow the Jessie switch are working flawlessly. I finally switched and are running Jessie and Isengard.

Fabio72, 2015-09-13:
I tried to update using xbian-config but I have now a systemd installation.
Kodi starts at boot then close letting me with some x11 messages on screen.
If I restarts Kodi by console then everything seems working, but at every boot I have to start xbmc manually
I'm getting the "failed to start user@1000.service" like error on dmesg

Also watchdog cannot install claiming errors on MAKEDEV

So I reverted back for now

mkreisl, 2015-09-13:
@Fabio72 
That's what I expected for some users updating to Jessie :worried:

I've built my version of updating Wheezy systems, but unfortunately i was knocked out last week (heavy migraine) and I was not able to complete and commit my work, so @mk01 was faster so I stopped working and testing on it (and, of course,  all my RPi's are already running Jessie now).

If you want to try my version, you can download it from here: http://apt.xbian.org/pool/devel/main/x/xbian-package-repo/xbian-package-repo_20150904-5_armhf.deb

mk01, 2015-09-18:
@Fabio72 

- presence of systemd is no indication of systemd being an init loader. Since systemd took over projects like udevd, it became inseparable from base install.
- MAKEDEV's "/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation." error is not an error, it is standard message informing udevd is active / MAKEDEV is useless. if watchdog is making trouble, uninstall it prior update, install back after.
```
Preparing to unpack .../watchdog_5.14-3_armhf.deb ...
Unpacking watchdog (5.14-3) ...
Setting up watchdog (5.14-3) ...
/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation.
/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation.
Processing triggers for systemd (215-17+deb8u2) ...
```
- "failed to start user@1000.service" = https://www.google.sk/search?client=safari&rls=en&q=debian+failed+to+start+user@1000.service&ie=UTF-8&oe=UTF-8&gfe_rd=cr&ei=z4P7VerGFNjANJ3gqbAH 
- if those messages are "Setting up X socket directories... /tmp/.X11-unix /tmp/.ICE-unix.", then it is just runlevel 2 service start console output, not relevant to the unsuccessful Kodi start

once you decide moving further with the upgrade and you end up with same situation (Kodi not starting) - post those logs:
```
tailf -500 /var/log/upstart/upstart-ev.log | pastebinit
xll | pastebinit
```

YoYoost, 2015-10-05:
As I had the 'kodi wont start' issue after I updated to jessie I figured I would reinstall my complete RPI2 (model B). I used the latest RPI2 release (XBian_2015.10.03_rpi2.img.gz) and this seemed to boot fine. However after I installed all the applications I use (NzbGet, Transmission, SickRage, Couchpotato) I noticed Kodi only boots sometimes. Is this related to this issue and should I have waited for a stable release or am I doing something wrong? I used HowTo's from the forums to install the applications and they only involved some apt-get installs.

closed 2016-06-17 by mkreisl