xbianonpi/xbian #754: Latest updates breaks I2S support (part 2).

coderpop, 2015-08-18:
After today's upgrade, I2S support no longer works (again).

I have reverted the file xbian-package-xbmc-scripts=1.2.0 which was needed to get Xbain working with Kodi booting at all.

Now, even though Alsa output is enabled in the system meny, the output is always on HDMI.

Also, the meny display refresh has gone back to 60Hz from 50Hz.

Does any testing happen before these releases?

CurlyMoo, 2015-08-18:
No, we automatically compile our packages. We offer a multiversion apt repository to fallback to previous version in case of errors and offer snapshot functionality to rollback a system altogether.

coderpop, 2015-08-19:
Rollback not working and today's 'upgrades' seem to make things worse.

mkreisl, 2015-08-19:
How did you do the rollback?

coderpop, 2015-08-19:
http://forum.xbian.org/thread-3087.html

mkreisl, 2015-08-19:
Ah, interesting. New command for me :smile: 
I'm always using the hard way:
````
sudo mount /dev/root /mnt
cd /mnt/root
sudo mv @ @-bad
sudo mv @-<good-snapshot> @
sudo sync
sudo reboot
````
Never had problems with this commands. And believe me, I use it very often

mkreisl, 2015-08-19:
I looked into the code of the script, it does the same - only hundreds of more lines of code, works pretty well on my RPi2

coderpop, 2015-08-19:
Both methods result in a 'file not found' message. Whilst this is off topic, it's annoying !
Now I have no audio and the safe rollback won't rollback!
Thanks for the suggestion though, I know you are trying to help! :)

closed 2016-06-17 by mkreisl