xbianonpi/xbian #756: Isengard breaks cec on rpi2

hackedbellini, 2015-08-20:
I tried upgrading `xbian-package-xbmc` to `15.1~RC1-1439988955`. After that, cec will not work on it anymore.

There's nothing strange on the logs. I even tried running `cec-client` and it does connect to my tv and recognize any button I press on the remote control.

Downgrading xbmc to `14.3~IsengardBackports-1437839259` makes work again.

mkreisl, 2015-08-20:
Hmm, sounds not good. 
I tested the RPI2 image with 15.1rc1, CEC works pretty well, no issue

hackedbellini, 2015-08-20:
@mkreisl what do you think is happening with my installation then? Because I have no problems at all with Helix, but Isengard will not work, no matter what I do.

Btw, forgot to mention, I'm using Jessie (were you using Wheezy?)

```
$ cat /etc/apt/sources.list
deb http://ftp.debian.org/debian/ jessie main contrib non-free
$ cat /etc/apt/sources.list.d/xbian.list 
deb mirror://apt.xbian.org/mirror.txt      stable  main rpi2-jessie
deb mirror://apt.xbian.org/mirror.txt      staging main rpi2-jessie
### deb mirror://apt.xbian.org/mirror.txt  devel   main rpi2-jessie
```

mkreisl, 2015-08-20:
@hackedbellini 
No, I have switched to Jesse for building images. And this new Image, containing Jessie, Kodi 15.1rc1, Kernel 4.1.5, cec 3.0.1 works without any issues on a RPi2

mkreisl, 2015-08-20:
Did you already played with the cec settings in Kodi. 
Now I remember, I had to deactivate one option there when I switched to Kodi 151 rc1, which is available in xbmc-sources-xbmc. Before this option was not deactivated, cec also dont't work here

hackedbellini, 2015-08-20:
@mkreisl hrm ok, that makes sense. Let me try here and I'll let you know.

mkreisl, 2015-08-20:
It's the second option from top. Try to deactivate it please

mkreisl, 2015-08-20:
imo cec is still very buggy, for example my rpi's can never turn on my tv, turning off is ok

hackedbellini, 2015-08-20:
> It's the second option from top. Try to deactivate it please

It worked ;)

> imo cec is still very buggy, for example my rpi's can never turn on my tv, turning off is ok

Yeah, I [know what you mean](https://github.com/xbianonpi/xbian/issues/495)

mkreisl, 2015-08-20:
Ok, +1
There must be some lines in the xbian kodi sources which breaks cec with this option activated. 
I worked in the past with my rpi-nightly builds, which I always updated from xbmc github directly, and had never problems with this option.
But as I remember, on a fresh installed img this option is also enabled and cec works

closed 2016-06-17 by mkreisl