xbianonpi/xbian #757: Graphical glitches in latest version

renwique, 2015-08-29:
Wiped my SD card and installed 20150817-0. I have some weird graphical glitches in Kodi, where text does not show up, or shows with errors. Scrolling through lists/items seems to cause or exacerbate the issue. Screenshots below with yellow rectangles highlighting the problem areas.
[image]
[image]

renwique, 2015-08-29:
Also, upgrading from the included kernel (4.1.4) to 4.1.5 did not fix.

mkreisl, 2015-08-29:
Maybe modifying some settings may help:
http://kodi.wiki/view/Advancedsettings.xml#algorithmdirtyregions

er0l, 2015-08-30:
Same issue here. Removing "<visualizedirtyregions>false</visualizedirtyregions>" from advancedsettings.xml makes it better, but still there are some glitches here and there especially when browsing trough long lists.

mkreisl, 2015-08-30:
I can confirm the issue on a fresh installed img on a RPi2

The solution seems to be:
Update kernel to 4.1.6 and the issue is gone. Just tested

Btw, it does not happen on a RPi1

renwique, 2015-08-31:
I tried a fresh 20150829 install, which came with 4.1.6, but couldn't get Kodi to start, even with a manual call to ```service xbmc start```. It might have been a bad SD card flash, but I didn't try again.

I tried 20150822 with an ```apt-get upgrade```, updating the kernel from 4.1.5 to 4.1.6. I still have the issue, in almost the exact same places.

mkreisl, 2015-08-31:
Hmm sounds strange
After I updated the kernel the issue has gone. But I should say, I only tested it on kodi's main screen

renwique, 2015-08-31:
Playing with ```<algorithmdirtyregions>``` didn't help.
Playing with ```<nofliptimeout>``` caused some interesting problems, including screen the screen going blank after updates if set higher than 6ms.
Setting ```<visualizedirtyregions>true</visualizedirtyregions>``` didn't give me any useful information, other than to see tons of red squares (which I expected). When the red squares faded away, everything was rendered correctly. My guess, of course, is that the entire area is being refreshed as the blocks fade away.

I'm flashing back to my old image from April, and I'm going to see if I can try to replicate the ```<nofliptimeout>``` behavior, and see if the visualized regions look any different.

renwique, 2015-08-31:
On 20150407-0 running kernel 3.18.8, adding ```<nofliptimeout>1000</nofliptimeout>``` has no effect, the screen updates like normal and continues to display without problem.

On 20150822 with 4.1.6, adding ```<nofliptimeout>``` with a value higher than 6-7 caused screen blanking after the updates happened.

Maybe related? Not sure.

renwique, 2015-08-31:
Scratch the above, there were two ```<gui>``` stanzas in my config. Behavior between the two are the same; the screen blanks on both versions.

renwique, 2015-08-31:
Playing with ```<algorithmdirtyregions>``` on the old version didn't cause any problems, except for some slowness on 2, which was to be expected. 1 seems faster than the default (3), so I'm leaving that set.

My only guess is there must be some issue with the graphics code/driver somewhere in the 4.x kernel series. I'm not a developer, so I have no clue where to start with that.

I'm staying on 20150407-0 with 3.18.8 for now. I'd like some of the new features of Isengard, but this is fine for the time being.

Smultie, 2015-09-07:
I'm having the same issue. Paging @mk01

mkreisl, 2015-09-07:
We all have the same issue including @mk01. It is absolutely randomly. I already made some tests but at the moment, I can't reproduce it anymore.

er0l, 2015-09-07:
Leave gui tags in advancedsettings.xml empty and try to open any larger list whether movies or online resources after scrolling few pages moving up and down through the list. it almost always shows up in my setup.

mk01, 2015-09-12:
@popcornmix

can you please look briefly at this ?
basically it looks it started ~2w ago when I changed our tree to be rebase on top of pop/isengard_rbp_backports (instead of xbmc/Isengard as before). 

before I was merging your newclock4 into our tree. I will double check AGAIN that there are no leftovers in the tree I rebase on top of yours (from previously merging the newclock4, but in-between please take brief look, maybe you will come with an idea what's the issue.

thx

mk01, 2015-09-12:
ok, found few lines as leftover exactly relevant to rendering / dirty regions.

fixed source, pushed, asked for recompile. rpi2 pkg is already at staging repo, rpi compiling.

@renwique 
it is not unnecessary to re-image always. much faster would be just downgrade installed Kodi package.
available versions can one list via
```
apt-cache policy xbian-package-xbmc
```
and install specific version (15.1~RC1-1438235202 in that example case)
```
apt-get install xbian-package-xbmc=15.1~RC1-1438235202
```

mk01, 2015-09-12:
rpi package was pushed to staging too

once they would be confirmed as glitch free, they will go stable

closed 2015-09-23 by mkreisl