xbianonpi/xbian #758: Clean install of 20150829-0 has "Duplicate sources.list entry"

renwique, 2015-08-31:
Output of a clean install's ```apt-get update```:

```
xbian@xbian ~ $ sudo apt-get update

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for xbian: 
Hit http://ftp.us.debian.org jessie InRelease
Hit http://http.debian.net jessie InRelease      
Hit http://ftp.us.debian.org jessie/main armhf Packages                    
Get:1 http://apt2.xbian.org stable InRelease [5,318 B]                     
Hit http://ftp.us.debian.org jessie/non-free armhf Packages      
Hit http://ftp.us.debian.org jessie/main Translation-en
Hit http://ftp.us.debian.org jessie/non-free Translation-en                               
Get:2 http://apt2.xbian.org stable/main armhf Packages [14.1 kB]                          
Hit http://http.debian.net jessie/main armhf Packages
Get:3 http://apt2.xbian.org stable/rpi2-jessie armhf Packages [15.3 kB]              
Hit http://http.debian.net jessie/non-free armhf Packages                            
Hit http://http.debian.net jessie/contrib armhf Packages                                                                                          
Hit http://http.debian.net jessie/contrib Translation-en                                 
Hit http://http.debian.net jessie/main Translation-en              
Hit http://http.debian.net jessie/non-free Translation-en          
Ign http://apt2.xbian.org stable/main Translation-en_US
Ign http://apt2.xbian.org stable/main Translation-en
Ign http://apt2.xbian.org stable/rpi2-jessie Translation-en_US
Ign http://apt2.xbian.org stable/rpi2-jessie Translation-en
Fetched 34.7 kB in 45s (765 B/s)                                                                                                                            
Reading package lists... Done
W: Duplicate sources.list entry http://http.debian.net/debian/ jessie/non-free armhf Packages (/var/lib/apt/lists/http.debian.net_debian_dists_jessie_non-free_binary-armhf_Packages)
W: You may want to run apt-get update to correct these problems
```

Line two of ```/etc/apt/sources.list``` contains ```deb http://http.debian.net/debian jessie main non-free contrib non-free```, simply removing the last ```non-free``` seems to fix this issue.

I believe at least 20150817-0 had this issue too.

mkreisl, 2015-08-31:
i have already noticed this

mkreisl, 2015-08-31:
it seems to be a bug in jessie debootstrap
will see how to fix it

mkreisl, 2015-09-05:
Fixed today, new images are available tomorrow

closed 2015-09-05 by mkreisl

tafia, 2015-12-29:
Got this today too
```
sudo apt-get update && sudo apt-get upgrade
Ign http://http.debian.net jessie InRelease      
Get:1 http://http.debian.net jessie Release.gpg [2,373 B]
Hit http://http.debian.net jessie Release
Ign http://http.debian.net jessie Release
Hit http://http.debian.net jessie/main armhf Packages
Hit http://http.debian.net jessie/non-free armhf Packages
Get:2 http://http.debian.net jessie/main Translation-en [4,585 kB]
Hit http://xbian.serveftp.net stable InRelease                                                                                                       
Hit http://xbian.serveftp.net stable/main armhf Packages                                                                                             
Get:3 http://http.debian.net jessie/non-free Translation-en [72.2 kB]                                                                                
Hit http://xbian.serveftp.net stable/rpi2-jessie armhf Packages                                                                                      
Ign http://http.debian.net jessie/main Translation-en_US                                                                                             
Ign http://http.debian.net jessie/non-free Translation-en_US                                                                                         
Ign http://xbian.serveftp.net stable/main Translation-en_US                                                                                          
Ign http://xbian.serveftp.net stable/main Translation-en                                                                                             
Ign http://xbian.serveftp.net stable/rpi2-jessie Translation-en_US                                                                                   
Ign http://xbian.serveftp.net stable/rpi2-jessie Translation-en                                                                                      
Fetched 4,659 kB in 33s (140 kB/s)                                                                                                                   
Reading package lists... Error!
W: GPG error: http://http.debian.net jessie Release: The following signatures were invalid: BADSIG 8B48AD6246925553 Debian Archive Automatic Signing Key (7.0/wheezy) <ftpmaster@debian.org>
E: Read error - read (5: Input/output error)
E: Problem opening /var/lib/apt/lists/http.debian.net_debian_dists_jessie_non-free_binary-armhf_Packages
E: The package lists or status file could not be parsed or opened.
```

installed xbian on 26th dec.

tafia, 2015-12-30:
Fixed it by changing apt source list to some other mirror