xbianonpi/xbian #761: could not find any device /dev/loop#Bad address

Roxxor91, 2015-09-02:
Hi!
I installed a fresh 8.29 RPi2 build and tried to mount an .img file (file is fine checked by MD5).
When trying to mount an image by mount -o loop,ro,offset=xxxxxxx, I get the following return: mount failed: Unknown error -1 (the same command and offset is working fine on my Linux Mint 17.2 Desktop).
mkreisl was able to mount on a custom kernel via sudo kpartx -v -a ./xxxxxxxxx.img.
On my fresh install I got the following return:
mount: could not find any device /dev/loop#Bad address can't set up loop

mkreisl, 2015-09-02:
Can confirm the issue. 
It is a kernel issue. Taking my 'private' kernel and it worked ok
@mk01 any ideas?

er0l, 2015-09-03:
Did you check if you have loaded kernel module loop? If it is not present you can do it with "modprobe loop" before mounting.

mkreisl, 2015-09-03:
That's it. :thumbsup: 
Was too late yesterday for recognizing the differences between the kernel configs :sleeping:

closed 2015-09-03 by mkreisl