xbianonpi/xbian #762: OpenVPN crashing Kodi

Roxxor91, 2015-09-04:
Hi!
I wanted to enable a OpenVPN Server. I made a standard setup and had to additionally load the "tun" module.

When I then start OpenVPN, kodi crashes. OpenVPN works like expected.
Here an excerpt from the openvpn log.

Fri Sep  4 12:18:39 2015 OpenVPN 2.3.4 arm-unknown-linux-gnueabihf [SSL (OpenSSL)] [LZO] [EPOLL] [PKCS11] [MH] [IPv6] built on Dec  1 2014
Fri Sep  4 12:18:39 2015 library versions: OpenSSL 1.0.1k 8 Jan 2015, LZO 2.08
Fri Sep  4 12:18:39 2015 Diffie-Hellman initialized with 1024 bit key
Fri Sep  4 12:18:39 2015 Socket Buffers: R=[163840->131072] S=[163840->131072]
Fri Sep  4 12:18:39 2015 ROUTE_GATEWAY 192.168.2.1/255.255.255.0 IFACE=eth0 HWADDR=b8:27:eb:6a:28:ec
Fri Sep  4 12:18:39 2015 TUN/TAP device tun0 opened
Fri Sep  4 12:18:39 2015 TUN/TAP TX queue length set to 100
Fri Sep  4 12:18:39 2015 do_ifconfig, tt->ipv6=0, tt->did_ifconfig_ipv6_setup=0
Fri Sep  4 12:18:39 2015 /sbin/ip link set dev tun0 up mtu 1500
Fri Sep  4 12:18:39 2015 /sbin/ip addr add dev tun0 local 10.8.0.1 peer 10.8.0.2
Fri Sep  4 12:18:39 2015 /sbin/ip route add 10.8.0.0/24 via 10.8.0.2
Fri Sep  4 12:18:39 2015 GID set to nogroup
Fri Sep  4 12:18:39 2015 UID set to nobody
Fri Sep  4 12:18:39 2015 UDPv4 link local (bound): [undef]
Fri Sep  4 12:18:39 2015 UDPv4 link remote: [undef]
Fri Sep  4 12:18:39 2015 MULTI: multi_init called, r=256 v=256
Fri Sep  4 12:18:39 2015 IFCONFIG POOL: base=10.8.0.4 size=62, ipv6=0
Fri Sep  4 12:18:39 2015 Initialization Sequence Completed
Fri Sep  4 12:19:10 2015 event_wait : Interrupted system call (code=4)
Fri Sep  4 12:19:10 2015 /sbin/ip route del 10.8.0.0/24
RTNETLINK answers: Operation not permitted
Fri Sep  4 12:19:10 2015 ERROR: Linux route delete command failed: external program exited with error status: 2
Fri Sep  4 12:19:10 2015 Closing TUN/TAP interface
Fri Sep  4 12:19:10 2015 /sbin/ip addr del dev tun0 local 10.8.0.1 peer 10.8.0.2
RTNETLINK answers: Operation not permitted
Fri Sep  4 12:19:10 2015 Linux ip addr del failed: external program exited with error status: 2
Fri Sep  4 12:19:10 2015 SIGTERM[hard,] received, process exiting

I think the "external prgram exiting" is kodi?
Which kodi log do you need?

mk01, 2015-09-11:
@Roxxor91 

would it be possible to provide ANY 'demo' configuration which I would use for debugging?

Roxxor91, 2015-09-21:
@mk01 Sorry for answering so late. I was busy.

Here is the guide I followed with a fresh install of xbian: http://jankarres.de/2013/05/raspberry-pi-openvpn-vpn-server-installieren/
It is german, but I guess it should be possible to understand the commands.

mk01, 2015-09-22:
@Roxxor91 

ok, I should better read - you have problem starting VPN server on machine running Kodi - not connecting TO VPN server from you machine running Kodi.

I have openswan installed without issues. So have to uninstall it and install OpenVPN, but will try.

between - we started building DBG packages to Kodi. that means each "*.deb" of kodi, has counter part "*-dbg.deb"

if you install it and attach Kodi process before you start OpenVPN, you will get pretty detailed backtrace of actual problem.
```
gdb --pid=$(pgrep kodi)
```
will attach debugger.

bionemesis, 2015-10-03:
So I'm running into a very similar issue, but it's with OpenVPN client and it's only on startup. Kodi crashes on startup if an OpenVPN connection has started/finished. If I manually start OpenVPN, either via the openvpn command or via /etc/init.d/openvpn, no problem with Kodi. If I re(start) OpenVPN via xbian-config, no problem. But I reboot, and Kodi doesn't load.

Unfortunately, this also means I can't get a detailed backtrace from gdb, unless you can tell me how to get that enabled before Kodi starts?

The only unique things about my setup:
1) I manually added the openvpn service in xbian-config.
2) I had to add tun to /etc/modules so it loaded on startup.

flo741, 2015-10-10:
I have the same issue in my xbian installation: Fresh install from today. I have added tun to /etc/modules. This worked - no problem. But when starting openvpn in server mode with a configuration manually edited in the config dir, kodi crashes. It is definitely connected to the openvpn start with a tun device. Would be pleased for a fix.

Spekkie, 2015-10-12:
@bionemesis I have exactly the same problem. May I ask how you added the openvpn service to xbian-config?

mk01, 2015-10-19:
@Spekkie @Roxxor91  @flo741 

there is new XBMC compiling right now where (we suppose) this should be fixed. all versions (rpi2/rpi/imx6) should be available within next hours.

Spekkie, 2015-10-19:
@mk01 
Thanks for working out a fix. I'm eager to try it out and trying to update all day, but xbian config is still telling me that I'm completely up to date...

mkreisl, 2015-10-19:
@Spekkie 
A zombie process blocked out build server, so packages like xbmc-package-xbmc were not updated. I've fixed this and build is just in progress. Tomorrow morning latest XBMC should be available

Spekkie, 2015-10-20:
@mk01 @mkreisl 
new xbmc-package fixed the issue for me it seems, as both openvpn and Kodi are now properly running at boot. Thanks!

flo741, 2015-10-20:
@mk01 
Thanks for the update, which I have just installed. The issue is still where, means kodi crashes as soon as openvpn starts in server mode with a tun device.

zomborig, 2015-10-25:
I also updated and the issue persists for me.
With latest packages installed, I've ran a gdb session on the source, and got following result:
```
Program received signal SIGSEGV, Segmentation fault.
0x008bf73c in CNetworkLinux::queryInterfaceList (this=this@entry=0x1c5ccb8)
    at NetworkLinux.cpp:431
431     NetworkLinux.cpp: No such file or directory.
(gdb) bt
#0  0x008bf73c in CNetworkLinux::queryInterfaceList (this=this@entry=0x1c5ccb8)
    at NetworkLinux.cpp:431
#1  0x008bfe76 in CNetworkLinux::CNetworkLinux (this=0x1c5ccb8)
    at NetworkLinux.cpp:322
#2  0x007c43e8 in CApplication::SetupNetwork (this=this@entry=0x1c1b170)
    at Application.cpp:440
#3  0x007c5940 in CApplication::Create (this=0x1c1b170) at Application.cpp:452
#4  0x008415c8 in XBMC_Run (renderGUI=<optimized out>) at xbmc.cpp:60
#5  0x002f8f3e in main (argc=1, argv=0x7efff754) at main.cpp:77
(gdb) info locals
mask = <error reading variable: Cannot access memory at address 0xfffffff4>
macAddrRaw = "8\225S\001\240\025"
i = <optimized out>
addr = <error reading variable: Cannot access memory at address 0xfffffff5>
name = <error reading variable: Cannot access memory at address 0xfffffff4>
iface = <optimized out>
list = 0x1c5de70
change = <optimized out>
lock = {<XbmcThreads::UniqueLock<CCriticalSection>> = {<XbmcThreads::NonCopyable> = {<No data fields>}, mutex = @0x1c5ccbc, owns = true}, <No data fields>}
pos = warning: Unable to fetch general register.
<unavailable>
cur = warning: Unable to fetch general register.
<unavailable>
```

Unfortunatelly I could not match the right source from git (even, if I got the 15.2RC3 src). 
I suspect, the TUN interface is not digestable for the source, ifconfig brings this for tun0:

```
root@xbian /usr/lib/debug # ifconfig tun0
tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.20.30.1  P-t-P:10.20.30.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

Spekkie, 2015-10-26:
I might have spoken too soon as well, as I encounter problems again. However I didn't have time to properly investigate, so it might  as well be something else...
I'll report back asap with my findings....

OtakuWally, 2015-10-26:
I have the same issue and would like to try to solve it. Are there SOURCES for the current xbian-package-xmbc, I would like to try to solve the problem via a patch, since I have spent some years to program unix networking software.

Spekkie, 2015-10-26:
@OtakuWally As far as I can tell the sources are available here:
https://github.com/xbianonpi/xbian-package-xbmc

mkreisl, 2015-10-26:
@Spekkie @OtakuWally 
The sources of kodi are available here https://github.com/xbianonpi/xbian-sources-xbmc, branch `treeIsengard`

OtakuWally, 2015-10-27:
I will take a look into it today. Which development packages do I need have to be installed to compile the xbmc sources?

mk01, 2015-10-27:
@all

ok, seeing the outputs and line numbers - the problem is in NULL dereferencing some of the fields on reading interfaces list. I will have time to fix it in few days, for anyone who willing to fix sooner:

NetworkLinux.cpp, line ~430. all info is read from struct ifaddrs (see man pages or man getifaddrs) - some of the fields are pointers to other structs and might contain NULL (depending on interface type)).

accessing any of such structures (and sub-structures) requires validity test first (!= NULL). 

mk

OtakuWally, 2015-10-27:
Yeah it was pretty probable that it is some kind of null pointer problem. I will try to reproduce today in case that I will have no problems to build xbmc and if I succeed I will post a patch here.

OtakuWally, 2015-10-27:
How can specify the includes and libs under /opt/vc for the ./configure step?

```
./configure --libdir=/opt/vc/lib/ --includedir=/opt/vc/include/
```

This upper did not work ...

OtakuWally, 2015-10-28:
Well I will have to give up, if no one is willing to help :-( The bugfix should be pretty easy, but since I can not compile xbmc to check it ... PLEASE someone?

CurlyMoo, 2015-10-28:
https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages

OtakuWally, 2015-10-28:
Thanks ;-) Will have a look at it ASAP.

zomborig, 2015-11-02:
Maybe I'm far too lame to do it based on that description, but I don't seem to be able to create that build environment, ... yet. 

The most irritating thing is, that there are probably only missing the following two lines from the code, around the 430. line :
```C++
if(cur->ifa_addr == NULL)
       continue;
```
I'm still trying to get the build environment online, right now it seems, I get that message at the **$xbiangit --arch armhfjessie --action other --debug**
```
...
E: Unable to locate package firmware-imx
E: Unable to locate package xbian-package-cec-6q
E: Unable to locate package xbian-package-shairplay-6q
E: Unable to locate package libafpclient-dev
E: Unable to locate package libimx
E: Unable to locate package libfslvpuwrap
E: Unable to locate package libafpclient-dev
```

mkreisl, 2015-11-02:
@zomborig 
Probably the build routines are outdated. It is in my looooong todo list to check and update this. 

But for the moment, you can download and use my ready to use schroot environments from here:
https://www.mediafire.com/folder/uy6mv8mt8m6em/schroots

File `etc-schroots.tar.bz2` holds the configuation for all schroots, please unpack it to `/etc`.

You probably have to edit `/etc/schroot/schroot-fstab`' (two lines at the end) and in `/etc/schroot/chroot.d/*.conf` the location of the unpacked schroot (directory)

Tip: Building package xbmc for RPi2 and/or i.MX6 is much much faster using the cross environment.

Hope this helps

zomborig, 2015-11-02:
Sorry guys, this setup is still very new to me :(
Never the less, thank you, those files helped a lot!
Unfortunatelly I'm still blocked, and too tired to continue for today.

What I got:
1. **apt-get install pkg-config** was still necessary in the cross bootstrap environment, otherwise I got some messages, like
```
configure.ac:32: error: possibly undefined macro: AC_DEFINE
```
2. Even after that I still got an error : 
```
...
touch .installed-arm-linux-native
make: Leaving directory '/usr/src/xbian/xbian-package-xbmc/build/rpi2-cross-isengard/working/tools/depends/native/JsonSchemaBuilder'
### Running hook 01-compile-jsonschemabuilder from pre-configure.d finished OK ###
### Running hook 10-compile-libsquish from pre-configure.d ###
make: Entering directory '/usr/src/xbian/xbian-package-xbmc/build/rpi2-cross-isengard/working/tools/depends/native/libsquish-native'
make: Nothing to be done for 'all'.
make: Leaving directory '/usr/src/xbian/xbian-package-xbmc/build/rpi2-cross-isengard/working/tools/depends/native/libsquish-native'
### Running hook 10-compile-libsquish from pre-configure.d finished OK ###
### Running hook 20-compile-TexturePacker from pre-configure.d ###
make: Entering directory '/usr/src/xbian/xbian-package-xbmc/build/rpi2-cross-isengard/working/tools/depends/native/TexturePacker'
make: Nothing to be done for 'all'.
make: Leaving directory '/usr/src/xbian/xbian-package-xbmc/build/rpi2-cross-isengard/working/tools/depends/native/TexturePacker'
### Running hook 20-compile-TexturePacker from pre-configure.d finished OK ###
### Running hook clone-pvr-addons from pre-configure.d ###
### Running hook clone-pvr-addons from pre-configure.d finished OK ###
configure: WARNING: unrecognized options: --disable-sdl, --disable-xrandr
configure: loading cache config.cache
configure: error: cannot find install-sh, install.sh, or shtool in build-aux "."/build-aux
Warning: schroot -c cross -p -- ./configure -C --enable-texturepacker --with-ffmpeg --enable-gles --disable-sdl --disable-x11 --disable-xrandr --enable-optical-drive --enable-dvdcss --disable-joystick --enable-debug --enable-libbluray --disable-vtbdecoder --disable-vaapi --disable-vdpau --enable-rtmp --disable-projectm --with-platform=raspberry-pi2 --enable-libcec --enable-player=omxplayer --host=arm-linux-gnueabihf returned 1
```

Hopefully I'll get a chance tomorrow, to continue.

mkreisl, 2015-11-02:
Probably there are some package missing on your build host. 

But anyway, I tried to reproduce this issue (installed OpenVPN at Saturday and tested Kodi 15.2 this evening) and this issue also occours on my RPi2. 
So, @mk01 made a patch which I tested sucessfully. So, issue has been solved.

New packages are currently be build by our build-server and should be available tomorrow morning.

zomborig, 2015-11-03:
Affirmative, I've just upgraded, and works, like a charm! 
Both tunnel and Kodi are running.
Thank you!

OtakuWally, 2015-11-03:
Same for me, THANKS (though I really would have liked to help here ;-) )

mkreisl, 2015-11-03:
> Same for me, THANKS (though I really would have liked to help here ;-) )

You are welcome - help is grateful accepted. Perhaps next time :question: In the meantime you can try building xbmc package :smile:

CurlyMoo, 2015-11-03:
Or just join the team :wink:

closed 2015-11-06 by mkreisl