xbianonpi/xbian #763: xbian backup function to create image did not know the kodi mappings

Starwhooper, 2015-09-06:
xbian offers within the kodi frontend environment the possibility, to create a image file about the current   system.

The assistent to configure the destination location of the image file, works with the sources that defined within kodi itself.
But the creation of the image file runs in the linux Environment that did not know the mappings from kodi.

As example:
I mount a samba share in the kodi Environment and choose it as destination for the image file. In reason that the linux in the background did not know that mount, the files was not created. (as workaround i configured a mount to the samba share in linux itself and use the local mountpoint as destination for assistent in kodi)

mk01, 2015-09-11:
@Starwhooper 

yes, this is confirmed.
those Kodi internal mounts should be removed from the list provided to user.

Starwhooper, 2015-09-13:
OK, thats also a solution. But i hoped that the linux environment will be also able to use the kodi mappings. 
Otherwise its for a only-xbian user not able, to create images to a server.

mkreisl, 2016-06-20:
Fixed, support added

closed 2016-11-04 by mkreisl