xbianonpi/xbian #764: Add the Module dvb-usb-dvbsky to xbian kernel config

gitr0b, 2015-09-07:
Please add support for dvbsky dvb-s usb devices:
Add the Module dvb-usb-dvbsky to xbian kernel config. It is included in standard kernel sources since 3.19. 
This firmware is needed in /lib/firmware/:
http://www.dvbsky.net/download/linux/dvbsky-firmware.tar.gz

mkreisl, 2015-09-07:
@gitr0b 
I built a kernel for RPi1 with enabled dvb-usb-dvbsky module (for RPi2 and i.MX6 it was already enabled).

For testing you can download this kernel from here: http://apt.xbian.org/pool/stable/rpi-jessie/x/xbian-package-kernel/xbian-package-kernel_4.1.6+-1441469033_armhf.deb

Firmware has to be downloaded and installed manually.

Please let us know if it works

gitr0b, 2015-09-07:
I think there is only a remote control keymap driver in the kernel: 
xbian@xbian ~ $ find /lib/modules -iname '*dvbsky*'
/lib/modules/4.1.6+/kernel/drivers/media/rc/keymaps/rc-dvbsky.ko

is I.MX6 the same as  armv7l ?
We have this: 
xbian@xbian ~ $ uname -a
Linux xbian 4.1.6+ #520 SMP Tue Sep 1 11:11:45 CEST 2015 armv7l GNU/Linux

Please add this module: 
xbian@xbian ~ $ diff .config .config.old 
3629c3629
< CONFIG_DVB_USB_DVBSKY=m
---
> # CONFIG_DVB_USB_DVBSKY is not set

-> Device Drivers
  -> Multimedia support (MEDIA_SUPPORT [=m]
    -> Media USB Adapters (MEDIA_USB_SUPPORT [=y]
      -> Support for various USB DVB devices v2 (DVB_USB_V2 [=m])

mkreisl, 2015-09-07:
I did not answer my question on irc which kind of hardware you use
I looked to the config of RPi2, this module is enabled there

gitr0b, 2015-09-07:
sorry i fell into sleep. My answer was late...:
[07:18] <r0b> <mkreisl> My hardware is cubox-i.  So this is the .config?: https://github.com/xbianonpi/xbian-package-kernel/blob/master/extra-files/imx6-4.1.y/.config
[07:19] <r0b> Is the source just vanilla from kernel.org or are additional patches used?
[07:53] <r0b> I added new issue #764 asking to support one more dvb module in kernel config.

mkreisl, 2015-09-07:
Build new kernel for i.MX6. You can download it from http://apt.xbian.org/pool/staging/imx6-jessie/l/linux-image-armmp/ in about 3 hours.
The desired module should now be included.

gitr0b, 2015-09-08:
The usb dvb-s adapter DVBSky S960CI works fine with this kernel. 
The manually copied firmware is loaded (dmesg output): 
i2c i2c-3: m88ds3103: downloading firmware from file 'dvb-demod-m88ds3103.fw'
Devices are created in /dev/dvb/adapter0/ . 
TVHeadend works. 
I suggest adding to xbian kernel for all supported platforms, it is a common adapter. 
Thank you!

mkreisl, 2015-09-09:
@gitr0b 
Thank you for your feedback. I've moved this kernel deb to the stable area
This module should now be available for all hardware platforms (RPi1, RPi2 and i.MX6)

closed 2015-09-09 by mkreisl