xbianonpi/xbian #765: Kodi crashes when unknown codec detected

Stunherald, 2015-09-07:
I have one specific movie and when i try to play it the CPU0 fires up to 100% and after ~1min the whole kodi reloads. From the debug (see links bellow) i see there is some issue with the Stream 4 - subtitles. By checking the mkv via MediaInfo both subtitle streams (3 and 4) looks the same, but by checking the codec info via VLC i can see some difference...

Kodi debug log: http://pastebin.com/bA7N80wV
MediaFile codec info: http://pastebin.com/xHUYGxzL
VLC codec info: http://pastebin.com/hnY6mQRQ

HW used: Raspberry PI 2
XBian image: XBian_2015.07.04

I guess in case like this there should be something like popup saying "Unable to load stream <n>" and then maybe something like "Ignore" and "Cancel" buttons rather that the whole kodi crash'n'load.

If more info is needed just let me know.

popcornmix, 2015-09-07:
A sample file will be needed.
e.g. cut a 2 minute chunk using mmg from [mkvtoolnix](https://www.bunkus.org/videotools/mkvtoolnix) and confirm the problem still occurs and upload to a hosting site (e.g. google drive or dropbox).

Stunherald, 2015-09-07:
Ok so looks like the issue was not with the subtitles but with some "attachment" type. I've removed both subtitle streams (3 and 4) and kodi crashed again on stream 2 (audio), so i've tried to remove the unusual x-rar attachment and bingo, the video worked flawlessly.

I've tried to cut a part of the file via MKVToolNix but in case of small file kodi will not crash but throw an error. I tried with 120s, 5min, 15min, 30min and for all of them i got error like there was an error with the video playback. Thing is that bigger the file is the longer it takes to get the error message. 30min cut was 2GB big. So i believe it is related to the size. If its too big, kodi will just not handle it and crash/restart.

So bellow is the 5sec cut. It is playable via VLC or Zoom player, but Kodi is having some issues with it. Getting error message is still better than a crash.

https://mega.nz/#!SgsiWS5a!laOJe-WRBsUKHrMPHu-RRB_TE6IjXXdrkDNHNlqlsYg

[image]

popcornmix, 2015-09-07:
We get a failure from new with pStream->codec->extradata_size=251658240:
https://github.com/xbmc/xbmc/blob/master/xbmc/cores/dvdplayer/DVDDemuxers/DVDDemuxFFmpeg.cpp#L1306
codec_type = AVMEDIA_TYPE_ATTACHMENT
codec_id = AV_CODEC_ID_NONE

I see this error:
```
00:15:14 17766.025391 T:1557132320   ERROR: Action - thread DVDPlayer, Unhandled exception caught in thread process, aborting. auto delete: 0
```
The file behaves badly on a linux machine. It takes about 30 seconds to start playback, although this machine does have enough RAM for multiple 250M buffers to be copied around.

I'd suggest you open a [trac ticket](http://trac.kodi.tv/) including the file and see if any devs have a suggestion for the best solution. I suspect ignoring codec_type = AVMEDIA_TYPE_ATTACHMENT may be sensible.

popcornmix, 2015-09-07:
https://github.com/popcornmix/xbmc/commit/bbc8d321002602b85ff7fb71e123e005d17d412c allows the file to play. I'll include it on my test branch a while and see if there are any issues.

mk01, 2015-09-13:
@Stunherald 

the fix was merged and next issue of -xbmc package will contain it.

Stunherald, 2015-09-13:
Perfect, thanks !!

mk01, 2015-10-19:
no more feedback, assuming fixed

closed 2015-10-19 by mk01