xbianonpi/xbian #766: Recent update on CuBox leads to non starting Kodi

didlich, 2015-09-11:
I've update my system today. After a rebboot kodi doesn't start any more.

kodi.log:

00:06:23 T:3027697664   ERROR: DBus: Error org.freedesktop.UDisks.Error.PermissionDenied - Not Authorized
00:06:23 T:3027697664   ERROR: DBus: Error org.freedesktop.UDisks.Error.Failed - Error mounting: mount exited with exit code 1: helper failed with:
                                            mount: only root can mount /dev/root on /home

got also kodi error:

** Error in `/usr/local/lib/kodi/kodi.bin': free(): invalid pointer: 0x015e7b80 ***

mk01, 2015-09-12:
this looks like galcore kernel driver / userspace firmware-imx libs would be of different generation.

what is your firmware-imx version and kernel ?
```
dpkg-query -W linux-image-armmp firmware-imx
```

didlich, 2015-09-12:
this is the result of the command on my CuBox:
```
firmware-imx    2.1.2
linux-image-armmp       4.1.6+-1441745602
```

mk01, 2015-09-12:
@didlich 

can you please post 
```
gunzip -cd < /proc/config.gz | grep VIV 
```

mk01, 2015-09-12:
@didlich 

*602 kernel had bad .config during compile, new is on the go. 
(you can revert to previous or wait until next repo update - in 2h - for new version)

didlich, 2015-09-12:
```
gunzip -cd < /proc/config.gz | grep VIV
```
here it is:

```
CONFIG_DRM_VIVANTE=m
# CONFIG_MXC_GPU_VIV is not set
CONFIG_MXC_GPU_VIV_V5=y
```

didlich, 2015-09-12:
I've just updated my system,
@mk01 the problem is **solved**, thanks for your investigation

closed 2015-09-12 by didlich