xbianonpi/xbian #767: Can't initialize iptables table `nat'

m-urban, 2015-09-12:
Hi there,

as of the last update of my RPi B+ to Kernel 4.1.6, iptables don't work for me anymore.

```# iptables -A POSTROUTING -t nat -o eth0 -j MASQUERADE```
```iptables v1.4.21: can't initialize iptables table `nat': Table does not exist (do you need to ins mod?)```
```Perhaps iptables or your kernel needs to be upgraded.```

I wonder if this might be the reason:
https://github.com/xbianonpi/xbian-package-kernel/blob/master/extra-files/rpi-4.1.y/.config#L820

Thanks for taking a look into this!

mk01, 2015-09-12:
@m-urban 

yes, ... obviously a bug, not intention.
changed. 
new kernel on next repo rebuild

m-urban, 2015-09-12:
Great, thank you!

m-urban, 2015-09-13:
The new build seems to have fixed this. Thanks again!

closed 2015-09-13 by m-urban