xbianonpi/xbian #768: xbian-package-xbmc 15.2~RC2-1442111406 excess logging

er0l, 2015-09-13:
Anyone else having issues with DEBUG logging being turned on with this version?

Log file keeps filling up with DEBUG messages no matter what setting it is in advancedsettings.xml.

NOTICE: Disabled debug logging due to GUI setting. Level -1

Still logging is enabled.

er0l, 2015-09-13:
Staging package xbian-package-xbmc_15.2~RC2-1442072724_armhf.deb does not have this issue.

CurlyMoo, 2015-09-15:
Version 15.2~RC2-1442111406 got this core issue:
```
write(2, "[vcsm_malloc_cache]: [25603] [Mo"..., 74[vcsm_malloc_cache]: [25603] [Mouse pointer]: NULL size or invalid device!) = 74
write(2, "\n", 1
)                       = 1
write(2, "kodi.bin: RBP.cpp:405: int gpu_m"..., 118kodi.bin: RBP.cpp:405: int gpu_malloc_uncached_internal(int, GPU_MEM_PTR_T*, int): Assertion `p->vcsm_handle' failed.
) = 118
```

mkreisl, 2015-09-15:
@CurlyMoo 
Do you have a mouse connected to your RPi?
What says `ls -la /dev/vc* | grep -v tty`? 

The output should be like this
```
root@kmxbilr2 /etc/apt/sources.list.d # ls -la /dev/vc* | grep -v tty
crw-rw---- 1 root video 251,   0 Sep 12 20:45 /dev/vc-cma
crw-rw---- 1 root video 248,   0 Sep 12 20:45 /dev/vchiq
crw-rw---- 1 root video 249,   0 Sep 12 20:45 /dev/vcio
crw-rw---- 1 root video 250,   0 Sep 12 20:45 /dev/vc-mem
crw-rw---- 1 root video 246,   0 Sep 12 20:45 /dev/vcsm
```
xbian-package-firmware >= 2.5.0 is required

CurlyMoo, 2015-09-16:
I don't. Reverting to a previous version worked, so the dev interfaces are there (because not affected by kodi).

er0l, 2015-09-20:
New package 15.2~RC2-1442550617 has DEBUG mode turned off which is step forward, but like the first RC2 package there is an issue when you pause and play any video audio stream is running at stuttering speed, like it is being fast forwarded and after few seconds it gets to normal speed and in sync with video.

CurlyMoo, 2015-09-26:
The rest of the 15.2 package also contain the same error:
```
ioctl(15, 0xc40c, 0x170002)             = 0
ioctl(15, 0x400cc404, 0x7ee94144)       = 0
ioctl(15, 0xc010c408, 0x7ee94138)       = 4
ioctl(15, 0xc40d, 0x170002)             = 0
write(2, "usrvcsm: ", 9usrvcsm: )                = 9
write(2, "[vcsm_malloc_cache]: [28276] [Mo"..., 74[vcsm_malloc_cache]: [28276] [Mouse pointer]: NULL size or invalid device!) = 74
write(2, "\n", 1
)                       = 1
write(2, "kodi.bin: RBP.cpp:405: int gpu_m"..., 118kodi.bin: RBP.cpp:405: int gpu_malloc_uncached_internal(int, GPU_MEM_PTR_T*, int): Assertion `p->vcsm_handle' failed.
) = 118
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76ff4000
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
tgkill(28276, 28276, SIGABRT)           = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=28276, si_uid=0} ---
+++ killed by SIGABRT +++
Aborted
```

mk01, 2015-10-01:
@CurlyMoo 

previous versions had no need of those NEW dev interfaces - so no , you don't have them. as was told earlier, xbian-package-firmware > 2.5.0 at min.

no future package, update, magic or power of thought will change that historical fact.

CurlyMoo, 2015-10-01:
@mk01:
```
# apt-cache policy xbian-package-firmware
xbian-package-firmware:
  Installed: 2.7.0-1442559202
  Candidate: 2.7.0-1442559202
  Version table:
 *** 2.7.0-1442559202 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
        100 /var/lib/dpkg/status
```
I'm already on it.

closed 2016-06-20 by mkreisl