xbianonpi/xbian #769: /usr/lib/avahi/avahi-daemon-check-dns.sh fails

m-urban, 2015-09-13:
Hi there,

as of the recent xbian / kernel 4.1.6 update, I receive the following error message on `ifup eth0`

`/usr/lib/avahi/avahi-daemon-check-dns.sh: 81: /usr/lib/avahi/avahi-daemon-check-dns.sh: sort: not found
`

The script is run by `/etc/network/if-up.d/avahi-daemon`

Is there anything we can do about it?
Thanks!

mk01, 2015-09-18:
can you provide
```
which sort
dpkg-query -W coreutils
lsb_release -c
apt-get install -s -f
```

m-urban, 2015-09-20:
Hi @mk01,

    root@mrp ~ # which sort
    /usr/bin/sort

    root@mrp ~ # dpkg-query -W coreutils
    coreutils	8.23-4
    
    root@mrp ~ # lsb_release -c
    Codename:	jessie

    root@mrp ~ # apt-get install -s -f
    Paketlisten werden gelesen... Fertig
    Abhängigkeitsbaum wird aufgebaut.       
    Statusinformationen werden eingelesen.... Fertig
    0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
    [none updated, installed, removed or not updated]

mk01, 2015-09-20:
@m-urban 

and the error is still present ???

m-urban, 2015-09-20:
It is, yes.

mk01, 2015-09-20:
@m-urban 

so even now you do ifdown eth0 / ifup eth0 and you get this error, right ? 
and when you try
```
for f in b d 7 jd a; do echo $f; done  | sort
```
you get 
```
7
a
b
d
jd
```
and no error ?

m-urban, 2015-09-20:
Correct. But while digging a little deeper, it appears that not sort is the culprit, but rather the destination of the redirected output. This is a quick excerpt from the said file `/usr/lib/avahi/avahi-daemon-check-dns.sh`

    TMP_CACHE="${NS_CACHE}.$$"
    cat /etc/resolv.conf | grep "nameserver" | sort > ${TMP_CACHE} || return 0

It seems that something is wrong with this TMP_CACHE variable. The redirection shouldn't fail with a "not found", though. I wonder if this might be a permission problem. Can I track this down any further? I always ran the `ifup`/`ifdown` as root so far.

mk01, 2015-09-20:
@m-urban 

you can. put 
```
set -x
```

early in the script. this will echo all command be executed in the script to stdout - including variable values and results of the commands.

mk01, 2015-09-20:
@m-urban 

and run ifup / ifdown with "-v" param
this will allow output of spawned processes (one is the mdns script)

m-urban, 2015-09-22:
Ok, I finally found out what's going wrong here: in /usr/lib/avahi/avahi-daemon-check-dns.sh:6 there is an instruction

    PATH=/bin:/usr/bin:/sbin:/usr/sbin

which was commented out. Once I included it, everything went back to normal. I am rather sure I didn't modify the line myself, so I wonder if any of the recently ran updates were messing with it.

mk01, 2015-09-23:
@m-urban 

believe it or not, history shown that it was myself in year 2014!
(adding the code into xbian-update changing that).

perhaps it needed to prevent this script from ```overwriting``` system standard PATH (or deliberately set by owner) with this mini one.
what changed now is most probably a caller script, removing std environment.
xbian-update will be updated.

m-urban, 2015-09-23:
Great! Thanks for patching it. I assume we can close this, then.

closed 2015-09-23 by m-urban