xbianonpi/xbian #770: Latest update stutters sound on starting a track.....

coderpop, 2015-09-15:
Fully up-to-date upgrade from yesterday and all went well until I played some music.
A track starts by playing about 0.5 seconds of sound, then silence for a second or two and then starts the track. Sometimes it can do this a few times, others only once and I believe it didn't do it at all at one point.

Here's the log....

http://pastebin.com/26yd0KYH

CurlyMoo, 2015-09-15:
It also occasionally stutters while playing a full album (e.g. music for a longer period of time).

coderpop, 2015-09-15:
I have not seen that yet CM, it seems that once it's playing it's OK.
I'll try some streaming radio and more full albums today and see how I get on.

coderpop, 2015-09-15:
Three hours + of streaming radio and no problem. Back to local server and problem is back. I think it might be a samba issue as that's the only thing I can think of that would be different between streaming from the net and from the server. I am using smb/cifs mount but never had issues with it before and nothing else has changed.

Smultie, 2015-09-15:
Could you try FTP or NFS, just to exclude...?

mkreisl, 2015-09-15:
Sounds like a buffering problem (that's also different between local and internet streaming)

Could you please check these settings? http://kodi.wiki/view/HOW-TO:Modify_the_cache
Maybe `<buffermode>1</buffermode>` could solve your issue

coderpop, 2015-09-15:
I can confirm that CM's observation is correct and stuttering (or drop out) happens randomly as well.

@Smultie  - If I knew how I could, but I can't at the moment.

@mkreisl  Here is what is in the advanced xml file right now.....

[term]
<advancedsettings>
  <network>
    <cachemembuffersize>20971520</cachemembuffersize>
    <readbufferfactor>4.0</readbufferfactor>
  </network>
  <samba>
    <clienttimeout>10</clienttimeout>
  </samba>
   <fanartres>720</fanartres>
   <imageres>540</imageres>
  <fullscreen>true</fullscreen>
  <libass>
    <styleoverrides>
      <style name="Shadow" value="0" />
      <style name="Blur" value="0" />
      <style name="BorderStyle" value="0" />
    </styleoverrides>
  </libass>
    <gui>
        
        
    </gui>
</advancedsettings>
[/term]

So should I add the buffer command to this or replace the one that's already there?

mkreisl, 2015-09-15:
@coderpop 
add

coderpop, 2015-09-15:
I tried both options (adding and repalcing) and it didn't solve the problem!

coderpop, 2015-10-13:
Seems to be cured with todays update too... :)

closed 2015-10-19 by mk01