xbianonpi/xbian #771: IPv6 module on 20150911-3

YoYoost, 2015-09-15:
I used to be able to use IPv6 when I added 'ipv6' to /etc/modules but this doesn't work anymore

I use IPv6 via an AICCU tunnel that now says: "No IPv6 Stack found!"

When I try:

    sudo modprobe ipv6

I get an error:

    modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.1.6+/modules.dep.bin'

I actually had this issue with 4.1.2 as well but it still persisted after I updated to 4.1.6

mkreisl, 2015-09-15:
@YoYoost , 3 questions:
Which hardware do you have?
Is this file (`/lib/modules/4.1.6+/modules.dep.bin`) availilable?
Can you copy this file, for example `cp /lib/modules/4.1.6+/modules.dep.bin /tmp`?

YoYoost, 2015-09-15:
This is on a Raspberry Pi 2 model B
I only have `/lib/modules/3.18.8+/` now that I look at it.
the copy command gives this error: `cp: cannot stat ‘/lib/modules/4.1.6+/modules.dep.bin’: No such file or directory`

YoYoost, 2015-09-15:
ohhh, Is this the problem?

````xbian@xbian ~ $ sudo apt-cache policy xbian-package-kernel
xbian-package-kernel:
  Installed: (none)
  Candidate: 4.1.6+-1442079217
  Version table:
     4.1.6+-1442079217 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     ...
xbian@xbian ~ $ sudo apt-cache policy xbian-package-kernel-headers
xbian-package-kernel-headers:
  Installed: (none)
  Candidate: 4.1.6+-1442079217
 Version table:
     4.1.6+-1442079217 0
 500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages````

I have had problems with kernels on RPi2 before, what should I install?

mkreisl, 2015-09-15:
You have a RPi2, the kernel package on this target is called `linux-image-bcm2836`, so do NEVER install `xbian-package-kernel` package, it is for RPi1.

What says `dpkg -l | grep bcm2836`?

YoYoost, 2015-09-15:
Ah, I was just reading about that.

    xbian@xbian ~ $ dpkg -l | grep bcm2836
    ii  linux-image-bcm2836               3.18.8+-1425522430            armhf        Latest XBian kernel (rpi2/3.18.y 3.18.8+)

mkreisl, 2015-09-15:
Hmmm, very strange is your error message in your first post.
I think it is the best if you update your kernel to 4.1.6

YoYoost, 2015-09-15:
I'm afraid I don't know how could you direct me to some documentation?

    xbian@xbian ~ $ sudo apt-cache policy linux-image-bcm2836
    linux-image-bcm2836:
    Installed: 3.18.8+-1425522430
    Candidate: 3.18.8+-1425522430
    Version table:
    *** 3.18.8+-1425522430 0
        100 /var/lib/dpkg/status

mkreisl, 2015-09-15:
When did you made `apt-get update` the last time?

mkreisl, 2015-09-15:
You can do all of this with the command `xbian-config`

YoYoost, 2015-09-15:
I just did apt-get update\upgrade without any new packages or a change in the `apt-cache policy linux-image-bcm2836`. I changed `deb http://http.debian.net/debian stable main contrib non-free` to `deb http://http.debian.net/debian jessie main contrib non-free`, this didn't differ.
xbian-config (with `update mirrors` and `apt-get update`) gives, you have installed all packages.
I tried rebooting

CurlyMoo, 2015-09-15:
What's in your `/etc/apt/sources.list.d/xbian.list`

YoYoost, 2015-09-15:
xbian@xbian ~ $ cat /etc/apt/sources.list
    deb http://http.debian.net/debian jessie main contrib non-free
    xbian@xbian ~ $ cat /etc/apt/sources.list.d/xbian.list
    deb mirror://apt.xbian.org/mirror.txt     stable   main    rpi-jessie
    ### deb mirror://apt.xbian.org/mirror.txt     staging   main    rpi-jessie
    ### deb mirror://apt.xbian.org/mirror.txt     devel   main    rpi-jessie

I tried `apt-get update` and no kernel package even when **staging** was uncommented.

mkreisl, 2015-09-15:
is wrong
should looks like this
```
deb mirror://apt.xbian.org/mirror.txt      stable  main rpi2-jessie
### deb mirror://apt.xbian.org/mirror.txt     staging   main    rpi2-jessie
### deb mirror://apt.xbian.org/mirror.txt     devel   main    rpi2-jessie
```

CurlyMoo, 2015-09-15:
Hmm, mine is wrong as well :confused:

YoYoost, 2015-09-15:
After changing `/etc/apt/sources.list.d/xbian.list` and `apt-get update` xbian-config gives a lot of new packages, among them `linux-image-bcm2836 4.1.6+`

mkreisl, 2015-09-15:
Ok, sounds much better :smile:

YoYoost, 2015-09-15:
Nvm, I will just install all packages 1 by one and reinstall the 'newer' packages.

YoYoost, 2015-09-15:
Actually, after I updated the kernel (not the rest) Kodi wouldnt start anymore and `dmesg` says `Failed to start user service: Unknown unit: user@1000.service`
Should I start a new thread for this?

mkreisl, 2015-09-15:
Probably the same issue here https://github.com/xbianonpi/xbian/issues/753#issuecomment-139866149

closed 2015-09-15 by YoYoost