xbianonpi/xbian #772: RPi2 no I2S sound output after todays full update/upgrade.

coderpop, 2015-09-15:
Kodi still shows alsa output selected, but no sound at all appears.

coderpop, 2015-09-16:
I have looked at /boot/config and the DToverlay is still there. Also, the /etc/modules contains all the same elements as before, so it is puzzling as to what has happened here. But I am not the only one with this issue as reported on the Xbian forum.

mkreisl, 2015-09-16:
I'm not the ALSA expert, so it is hard for me helping you.
Did you already checked if the output channels are muted? If they are, perhaps it helps by using alsamixer to unmute it.

coderpop, 2015-09-16:
Replacing kernel to 4.1.3+ and sound again! So likely a kernel issue???  I'm not a expert either! ;)

mk01, 2015-09-18:
@coderpop 

can you please test kernel just posted to staging =4.1.6+-1442543695 ?

coderpop, 2015-09-19:
mk01,
I would have tested your kernel, but I spent nearly all day trying to resolve cron and i2c with Xbian. Plus the pi2 was flashed back to wheezy working version. I will have to do all the updates/upgrades again before I get close to testing it. Maybe I will get time tomorrow.

coderpop, 2015-09-23:
Finally I get some time to try this....Here is the result....

[term]
xbian@cinema /tmp $ sudo dpkg -i xbian-package-kernel_4.1.6+-1441645427_armhf.deb
Selecting previously unselected package xbian-package-kernel.
dpkg: regarding xbian-package-kernel_4.1.6+-1441645427_armhf.deb containing xbian-package-kernel:
 linux-image-bcm2836 conflicts with xbian-package-kernel
  xbian-package-kernel (version 4.1.6+-1441645427) is to be installed.

dpkg: error processing archive xbian-package-kernel_4.1.6+-1441645427_armhf.deb (--install):
 conflicting packages - not installing xbian-package-kernel
Errors were encountered while processing:
 xbian-package-kernel_4.1.6+-1441645427_armhf.deb
[/term]

According to xbain-config all packages are up-to-date.

mkreisl, 2015-09-23:
@coderpop 
You tried installing a kernel for RPi1 (xbian-package-kernel) on a RPi2. This can not work.
Please use a Kernel for RPi2 (linux-image-bcm2836), @mk01 meant this one http://apt.xbian.org/pool/staging/rpi2-jessie/l/linux-image-bcm2836/linux-image-bcm2836_4.1.6+-1442543695_armhf.deb

coderpop, 2015-09-23:
Yeah, I just realised while looking agian that the file was the wrong one - Doh!

SO now I did it with the correct file and install went smoothly. Did a reboot and still no sound on I2S.
Alsa is selected in the system menu - do you want any logs?

As an aside, I also had to recalibrate the overscan to get the screen looking right again, never had that issue before.

coderpop, 2015-09-23:
While I think of it, what firmware would you suggest to run with the above? Maybe it would help?

coderpop, 2015-10-13:
and todays update makes it work again! :)

mk01, 2015-10-19:
@coderpop 

thanks for confirmation.

closed 2015-10-19 by mk01