xbianonpi/xbian #773: ASS subtitles in MKV containers dont work by default

pynej, 2015-09-16:
On Raspberry Pi B, current release as of Sep 10, 2015.

Minor thing but the embedded ASS tracks in MKV containers will show as tracks but the subtitles themselves will not actually render when watching a video.  Any MKV with embedded ASS track should reproduce.

Fix is trivial, the libass5 package should be installed by default.  

Manually fix and re-test with:
sudo apt-get install libass5

mkreisl, 2015-09-16:
Did you made a full upgrade from Wheezy to Jessie? 
Then libass5 should be installed automatically

pynej, 2015-09-16:
Hrm, it should be the latest.  I did a fresh download/install the other day and then ran the updates and had the issue.  Forgetting how to pul the current release/rev at the moment.

pynej, 2015-09-16:
It has both libbers and libbers packages available and 4 was installed by default.  It does look like there are some other updates that I haven't installed as this is a fresh download from the other day, but it doesn't list libass5 as new package to be added when I do an upgrade.  I didn't actually run through the upgrade though as it's working in my case now with both versions installed.

mkreisl, 2015-09-16:
So, what you did is download a recent image, wrote it to sd-card and then made an update?

I think I found the culprit. The image builder loads libass4 instead of libass5 (don't know why) and xbmc-package-xbmc requires libass4 OR libass5.

pynej, 2015-09-16:
Yah, it was with the SD card installer for OS X on http://www.xbian.org/getxbian/ targeting the Pi.  From what I recall I just selected the latest image from the list and installed.  Not 100% sure if i did an apt-get upgrade after it was done as i had to start over once.

Sound like you found it though as it did have v4 installed instead.  I haven't removed v4 so I don't know if other things will break if I do.

mkreisl, 2015-09-16:
You can remove libass4

closed 2015-09-23 by mkreisl