xbianonpi/xbian #774: xbian-package-xbmc should depend on libnfs4

ssakatausa, 2015-09-18:
I have found that the NFS feature does not work in Kodi provided in XBian_2015.09.12_rpi.img.gz. The log file suggested that Kodi could not handle NFS, failing to load libnfs.so.4, which was not included the 2015.09.12 release. After installing the libnfs4 package, the NFS feature started working. So, it is probably a good idea to make xbian-package-xbmc dependent on libnfs4.

Thanks.

mkreisl, 2015-09-18:
Done. Thank you for investigation

closed 2015-09-23 by mkreisl

palswim, 2015-11-02:
`xbian-package-xbmc` does not appear to depend on `libnfs4` yet, at least coming from the apt.xbian.org/mirror.txt, `rpi-wheezy` repository.

mkreisl, 2015-11-02:
`rpi-wheezy` is obsolete and won't be updated anymore. You should upgrade your rpi installation to Jessie