xbianonpi/xbian #775: Edimax AC600 Wifi not working out of the box (EW-7811UTC)

chrismhinds, 2015-09-29:
Unfortunately I only have a partial transcript of our IRC chat, but you walked me through this via IRC.  The gist is that the current build doesn't include the Edimax AC600, but after compiling it, Xbian now includes the drivers for my wifi USB stick.

The solution included installing build tools, adding a workaround to tell Makefile how to build the package, making the package, and then rebooting.

If you have the remainder of the convo, this should be an easy "issue" to resolve for future users who also want to use the Edimax AC600 (EW-7811UTC) wifi USB.

[13:04] <Brad_> Hi - does XBian work with Edimax AC600 USB wifi?  (EW-7811UTC)
[13:04] <Brad_> or do you have a suggestion of an 802.11ac wifi USB?
[13:13] <mk01> Brad: if that's supported in 4.1.x kernel, then it will work
[13:14] <mk01> brad: ok, support is not mainlined in kernel yet - but http://ubuntuforums.org/showthread.php?t=2228244
[13:15] <mk01> so on XBian, you just install kernel-headers package and copy & paste the same commands
[13:15] <mk01> your driver will build
[13:16] <Brad_> using that link?
[13:16] <mk01> yes
[13:17] <mk01> sudo apt-get update
[13:17] <mk01> sudo apt-get install linux-headers-armmp
[13:17] <mk01> git clone https://github.com/gnab/rtl8812au.git
[13:17] <mk01> cd ~/rtl8812au
[13:17] <mk01> make
[13:17] <mk01> sudo make install
[13:17] <mk01> sudo modprobe 8812au
[13:17] <Brad_> yah ok
[13:17] <mk01> ((if on RPI2))
[13:17] <Brad_> I'll give it a go.  Thanks!  :)
[13:17] <mk01> if on RPI1, change linux-headers-armmp -> xbian-package-kernel-headers
[13:18] <mk01> if you test & appove the driver as working ok, we can add it into std build of kernel
[13:18] <mk01> just let us know
[13:18] <Brad_> ok, thanks.  I'm updating now
[13:20] <mk01> ah so you have the stick already ?
[13:20] <Brad_> yes
[13:20] <mk01> :D
[13:20] <mk01> ok
[13:20] <Brad_> and am currently connected via 1000-baseT
[13:21] <mk01> ok, let me know later ,,, how it worked
[13:23] <Brad_> make: command not found .... hmm
[13:23] <mk01> my bad
[13:24] <mk01> you need some basic building tools too
[13:24] <mk01> wait
[13:24] <mk01> apt-get install build-essential
[13:29] <Brad_> make[1]: *** /lib/modules/4.1.6+/build: No such file or directory.  Stop.
[13:29] <Brad_> is there another package I should get
[13:31] <mk01> no, just reboot normally (on start this ./build directory is created as link to /usr/src/linux-.....)
[13:31] <mk01> so do it manually
[13:31] <mk01> list /usr/src/
[13:31] <mk01> ls -la /usr/src/
[13:31] <mk01> there will be linux-headers-xxxxx
[13:31] <mk01> and do
[13:32] <Brad_> k
[13:32] <mk01> ln -s /usr/src/linux-headers-xxxx /lib/modules/4.1.6+/build
[13:32] <mk01> so if you have 4.1.6+, then perhaps
[13:32] <mk01> ln -s /usr/src/linux-headers-4.1.6+ /lib/modules/4.1.6+/build
[13:33] <mk01> ((but this missing link should be fixed anyhow))
[13:34] <mk01> if you don't forget, open please issue on github , copy & paste this discussion. or just write "after headers installation, link /lib/modules/XXX/build is not instantly created - reboot is needed'
[13:34] <Brad_> k

CurlyMoo, 2015-09-30:
Same issue as posted a short moment later: https://github.com/xbianonpi/xbian/issues/776

mk01, 2015-09-30:
fixed

closed 2015-09-30 by mk01