xbianonpi/xbian #777: Xibain update held back

chrisofthewhites, 2015-09-29:
Don't know why i get 
`The following packages have been kept back:`
  `xbian-update`
(don't know how to put a new line within code...hence my messy formatting)

`dpkg-query -W xbian-\*`

Gives the below details

`xbian-package-cec       9.2.2.0-1-1430117049`
`xbian-package-cec-6q`
`xbian-package-config`
`xbian-package-config-shell      20150429-0`
`xbian-package-config-xbmc       20150306-0`
`xbian-package-firmware  2.7.0-1442559202`
`xbian-package-firmware-6q`
`xbian-package-initramfs-tools   20150918-1`
`xbian-package-kernel    4.1.7+-1442921806`
`xbian-package-kernel3.6.11`
`xbian-package-libshairport`
`xbian-package-libtag    1.0-3`
`xbian-package-lirc      1.5.1`
`xbian-package-networkfs`
`xbian-package-repo      20150919-0`
`xbian-package-samba     20150219-0`
`xbian-package-shairplay 1.9.0-1440513689`
`xbian-package-splash    2.2.0`
`xbian-package-upstart-xbmc-bridge       1.5.3`
`xbian-package-usbmount  2.0.1`
`xbian-package-vnc-server        2.0.0fix`
`xbian-package-xbianhome 1.0.7-1423098214`
`xbian-package-xbmc      15.2~RC2-1442546683`
`xbian-package-xbmc-6q`
`xbian-package-xbmc-alsa`
`xbian-package-xbmc-gotham-nightly`
`xbian-package-xbmc-nightly`
`xbian-package-xbmc-nightly-6q`
`xbian-package-xbmc-scripts      1.2.5`
`xbian-package-zram-swap 1.0.6`
`xbian-update    20150817-0`

Can anybody spot anything that would cause the xbian update not to upgrade?

Thanks,
Chris

CurlyMoo, 2015-09-30:
What's the full output of the failing upgrade when running `apt-get upgrade` from the console?

chrisofthewhites, 2015-09-30:
I'm not at home now, originally it listed some things that weren't needed and could be removed using apt-get autoremove (I think) so i removed them using autoremove as it said. Than the only other thing mentioned was the one application held back.

I was trying it a few times, and once there was something beginning with b which seem to be upgraded, and then downgraded the next time...but then wasn't mentioned the last time I tried it. 

Sorry for lack of info, I'll run `apt-get upgrade` again tonight and give you a more useful update.

Thanks for your interest.

mk01, 2015-09-30:
@chrisofthewhites 

run 
```
apt-get dist-upgrade
```

(just this one time only)

chrisofthewhites, 2015-09-30:
hmm it's going now, still kept that package back, 
`The following packages will be REMOVED:
  xbian-package-cec
The following NEW packages will be installed:
  init
The following packages have been kept back:
  xbian-update
0 upgraded, 1 newly installed, 1 to remove and 1 not upgraded.
Need to get 4,684 B of archives.
`
Only thing i can think of , I hardly had held that package for some reason, to stop it upgrading? maybe there was an issue with that upgrade at one time and I had read somewhere not to upgrade it? I don't know how to check that....
(how do i put a new line in the code, or copy and paste it so it's formatted?)
Thanks,
Chris

mkreisl, 2015-09-30:
> (how do i put a new line in the code, or copy and paste it so it's formatted?)

Use of three ` at the beginning and at the end of your text block

chrisofthewhites, 2015-09-30:
thanks mkreisl
System rebooting now after upgrades, fingers crossed!

closed 2015-09-30 by chrisofthewhites