xbianonpi/xbian #778: RPi too slow lauching videos from library in recent updates

jpataias, 2015-10-06:
In recente updates I noticed that it is taking much longer to launch videos from the library in Kodi. It can take from just a few seconds to several minutes. This is really annoying when trying to watch something.

mkreisl, 2015-10-06:
You should give us a little bit more of information
- Log files
- Where are the videos stored
Sounds for me like a DNS or networking problem

jpataias, 2015-10-06:
I don't know what log files should I get.

The files however are in a external drive connected to a USB port. The hardware configuration hasn't changed, but since a month ago this has started to happen.

jpataias, 2015-10-08:
Here it took 2 minutes to start the video playback:

22:10:17 274356.875000 T:3019870208  NOTICE: Previous line repeats 5 times.
22:10:17 274356.875000 T:3019870208  NOTICE: DVDPlayer: Opening: video.mkv
22:10:17 274356.875000 T:3019870208 WARNING: CDVDMessageQueue(player)::Put MSGQ_NOT_INITIALIZED
22:10:17 274356.875000 T:2619339696  NOTICE: Thread DVDPlayer start, auto delete: false
22:10:17 274356.875000 T:2619339696  NOTICE: Creating InputStream
22:10:17 274356.875000 T:2619339696  NOTICE: Creating Demuxer
22:10:17 274357.031250 T:2619339696  NOTICE: Opening stream: 0 source: 256
22:10:17 274357.093750 T:2619339696  NOTICE: Creating video thread
22:10:17 274357.093750 T:2619339696  NOTICE: Opening stream: 1 source: 256
22:10:17 274357.093750 T:2619339696  NOTICE: Creating audio thread
22:10:17 274357.093750 T:2610951088  NOTICE: Thread OMXPlayerVideo start, auto delete: false
22:10:17 274357.093750 T:2602562480  NOTICE: Thread OMXPlayerAudio start, auto delete: false
22:12:00 274459.968750 T:2619339696  NOTICE: OMXClock using audio as reference
22:12:00 274460.406250 T:2610951088  NOTICE: Display resolution ADJUST : 1920x1080 (1920x1080) @ 23.98 - Full Screen (28) (weight: 0.000)

mkreisl, 2015-10-08:
Did you edit your logfile?

On my system it looks like:
```
... T:1954202176  NOTICE: DVDPlayer: Opening: /media/usb500g/videos/external/video.mkv
```
But anyway, your USB Disk is ok, no bad blocks which could cause read timeouts? 
And you could execute following command `btrfs scrub start -B /` to see if your SD card is ok

jpataias, 2015-10-09:
Yeah sorry, I was changing the name of the video and ended up removing all the path. it should be:
```
... T:3019870208 NOTICE: DVDPlayer: Opening: /media/Elements/Biblioteca/Series/Name/Subname/video.mkv
```

The command add the following output:
```
scrub done for 6988a0d8-6562-4073-a18d-33c5492b01a1
        scrub started at Fri Oct  9 10:17:54 2015 and finished after 139 seconds                                                                                                                                     
        total bytes scrubbed: 1.69GiB with 0 errors
```

The disk should be ok as I don't have any timeouts when reading the files through the network. I will try to check for badblocks later.

mk01, 2015-10-19:
@jpataias 

so you have the delays only if opening files from USB ???
can you send output of "mount" ?

jpataias, 2015-10-19:
It is essentially always from usb. I use samba in xbian to share the contents of the hard drive in the network.

mk01, 2015-10-19:
ok, send back output
```
mount
```

after.

jpataias, 2015-10-21:
I ran the chkdisk and got the following result:
'''
The type of the file system is NTFS.
Volume label is Elements.

Stage 1: Examining basic file system structure ...
  29232 file records processed.
File verification completed.
  702 large file records processed.
  0 bad file records processed.

Stage 2: Examining file name linkage ...
  32840 index entries processed.
Index verification completed.
  0 unindexed files scanned.
  0 unindexed files recovered to lost and found.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
  1804 data files processed.

Stage 4: Looking for bad clusters in user file data ...
  29216 files processed.
File data verification completed.

Stage 5: Looking for bad, free clusters ...
  175269104 free clusters processed.
Free space verification is complete.
CHKDSK discovered free space marked as allocated in the volume bitmap.

Windows has made corrections to the file system.
No further action is required.

   2861584 MB total disk space.
   2176667 MB in 26705 files.
      5088 KB in 1806 indexes.
         0 KB in bad sectors.
    273672 KB in use by the system.
     65536 KB occupied by the log file.
 701076416 KB available on disk.

      4096 bytes in each allocation unit.
 732565759 total allocation units on disk.
 175269104 allocation units available on disk.
'''

However, right now I am not able to boot xbian, I think it has something to do with the SD card. I will try with another card and report back later.

jpataias, 2015-10-22:
For a moment I thought that everything was fixed by the new SD card, but sadly that was not the case.

Here is the log when mounting the hard drive:
```
09:58:31 1647.509033 T:3020894208   DEBUG: UDisks: Is not able to mount DeviceUDI /org/freedesktop/UDisks/devices/sda: IsFileSystem false HasFileSystem  IsSystemInternal false IsMounted false IsRemovable false IsPartition false IsOptical false
09:58:31 1648.046265 T:3020894208   DEBUG: UDisks: Mounting /org/freedesktop/UDisks/devices/sda1
```

jpataias, 2015-11-01:
I have found that when I play a video on a PC through the network, then there was a higher chance of the video (any video) being played instantly on xbian. That however doesn't always happens.

I have now found that if the video doesn't start to play, if I press pause (while xbian is trying to launch the video) and the play, the video will instantly start, every time.