xbianonpi/xbian #779: Army of dhclients

matega, 2015-10-18:
Hardware: Raspberry Pi 2
XBian version: 2015-10-17, 2015-10-10
Special: Running from external hard drive, BTRFS compression=lzo

## The problem

I upgraded to the latest image about a week ago. (This should be the 2015-10-10 version, although I can't be sure.) From then on, a new instance of `dhclient` started about every 2 minutes, to the point of them consuming all the RAM and the OOM killer killing other processes. A reboot doesn't help. Since then I reinstalled the whole system to the 2015-10-17 version (I replaced the HDD under the whole thing, and as the update screwed up my `/etc/fstab`, I suspected that this too could be an update artifact), but the problem persisted.

## Installed packages

The standard ones, plus the following and their dependencies:
  * transmission-daemon
  * nginx
  * php5-fpm
  * php5-
    * gd
    * curl
    * apcu
  * curl
  * mariadb
  * ssmtp
  * rsyslog

## What I tried so far

I was stumped (I didn't edit any DHCP-related config files even before the upgrade), so I read a lot of `dmesg`, but I found no mention of `dhclient`. I also installed `rsyslog` and looked into `/var/log/syslog`, but apart from each `dhclient` happily announcing its birth and getting the same (static) lease, I couldn't find anything related to the issue. I've made a cronjob to `pkill dhclient` every hour, so it's ugly but stable.

mk01, 2015-10-19:
@matega 

no idea on this. this can be anything (dhcp server changed, isc-dhcp-client pkg changed ...). for now remove isc-dhcp-client (what XBian preinstalls by default as safe option). XBian contains lightweight own dhcp client, which will be used after you remove isc-dhcp-client. DL isc-dhcp-client deb pkg as contingency you won't get ip, and try restarting eth0 - like this:
```
sudo -s
cd /tmp
apt-get download isc-dhcp-client
apt-get remove isc-dhcp-client
ifdown --force eth0; ifup eth0
```

if all well, let's discuss again in a month or so... (maybe something will come up)
if you don't get IP, configure manually and try any other dhcp client - udhcpd, dhcpcd5 ... you can configure eth0 manually with:
```
ip a add IP/NETMASK dev eth0
ip r add default via IPOFGW dev eth0
```
if you don't know the data, install back  isc-dhcp-client, ifdown/ifup eth0 and try the others.
...

Spekkie, 2015-10-20:
I'm also having this issue, though I thought is was due to openVPN or SSH...
Since the pi ran fine overnight and as soon as I started logging in over openVPN+SSH to update xbian for the openVPN crashing KODI issue the dhclients started to stack until the point where the ssh connection was kicked out and I was locked out of my pi...
Problem is the pi is stationed at my parents house to provide backups to my server, so when I'm locked out it's hard to get back in since one of my folks need to powercycle the pi...

mk01, 2015-10-20:
@Spekkie 

that's why one should use (exactly as written before - one line)
```
ifdown --force eth0; ifup eth0
```

sure there is risk of eth0 no comming up, then change that for:
```
ifdown --force eth0; ifup eth0 || { ip a add IP/NETMASK dev eth0; ip r add default via IPOFGW dev eth0; }
```

sure, there is never 100% assurance of success. but currently we can't help more.

Spekkie, 2015-10-20:
@mk01 
removing the isc-dhcp-client package seems to have fixed the issue for me, however a reboot was the easiest way to get rid of the remaining dhclient processes...

mk01, 2015-10-20:
@Spekkie 
thanks for info

matega, 2015-10-22:
Thanks for your reply. Unfortunately I don't have physical access to my Pi now, but I'll try this once I get there.