xbianonpi/xbian #780: Rpi 2 SBS 3D not working !

coderpop, 2015-10-18:
Hi

Not had much time to look into this, but 3D SBS on a rpi 2 is not playig right.

I has some play, then pixelated screen, then stuttered play, then pixelated screen etc......

I will try with Rpi 1 later in the week (hopefully).

Have not looked at logs yet, just thought it best to highlight the issue first and look into it later.

CP

mk01, 2015-10-19:
@coderpop 

there were some 3d related commits in the latest code.
update and retest please. then we can approach @popcornmix if this will still be valid.

also please make sure you have the latest firmware (xbian version 2.8.0-1444638638, upstream from Oct-11-2015).

coderpop, 2015-11-06:
Here's the result of trying to update the firmware with stable and staging enabled....

xbian-package-firmware is already the newest version.
The following packages were automatically installed and are no longer required:
  g++-4.6 guile-1.8-libs libass4 libavcodec53 libavdevice53 libavfilter2
  libavformat53 libbind9-80 libdns88 libffi5 libgettextpo0 libgssglue1
  libisc84 libisccc80 libisccfg82 liblwres80 libnfs1 libopencv-core2.3
  libopencv-imgproc2.3 libopenjpeg2 libparted0debian1 libplist1 libpostproc52
  libreadline5 librtmp0 libstdc++6-4.6-dev libswscale2 libsystemd-login0
  libtiff4 libts-0.0-0 libx264-123 python-central tsconf
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Is it really safe to remove all those packages?
3D is still broken.

CurlyMoo, 2015-11-06:
Nothing is using those pacakges so yes.

mkreisl, 2015-11-06:
@coderpop 
Where are your 3D movies stored? 
Just tested SBS and HOU (1080p) with my RPi1 - everything works perfect
Did you already try to use another sound output? Maybe I2S together with 3D has an issue.

coderpop, 2015-11-06:
I'll look into it. I am on firmware 2.9.0 now BTW.
@Curley Mo - Thanks! :)

@Mkreisel -  I2S sound is OK, just stutters and stops with the video. 
Also, the transport controls won't stop the video no matter how many times I try. I have to SSH in and reboot.

I might be able to test with RPi 1 this weekend, but for now only have '2' in the cinema.

mkreisl, 2015-11-06:
Tested again with RPi2 - no problem. Tried both h/w accelleration omx and mmal - both works

coderpop, 2016-01-04:
Tried again with 3D ISO stored on server. Worked before upgrading yesterday. Now I have problem with left half of screen in left eye and right half of screen in right eye.

Changing sync bit on projector made no difference. Strange one. I will look at if I can make it into a SBS mkv file and see if the problem persists (works fine with 3d bluray player though).