xbianonpi/xbian #781: RPI2 ir-keytable not supporting NEC protocol on rc-it913x-v1

nikto-es, 2015-10-19:
I'm trying to use my IT913x-based USB DVB-T tuner IR remote.

Problem comes when ir-keytable fails to enable its protocol, NEC:

```
ir-keytable 
Found /sys/class/rc/rc0/ (/dev/input/event0) with:
	Driver dvb_usb_af9035, table rc-it913x-v1
	Supported protocols: NEC 
	Enabled protocols: 
	Name: ITE 9135 Generic
	bus: 3, vendor/product: 048d:9135, version: 0x0200
	Repeat delay = 500 ms, repeat period = 125 ms
```

```
ir-keytable -p NEC -d /dev/input/event0 
/sys/class/rc/rc0//protocols: Invalid argument
Couldn't change the IR protocols
```

Lower-case fails as well.

```
ir-keytable -p nec -d /dev/input/event0 
Invalid protocols selected
Couldn't change the IR protocols
```

Any idea what's going on?

nikto-es, 2015-10-19:
Needless to say, irw shows nothing.

ir-keytable does show debug activity:
```
# ir-keytable -d /dev/input/event0 -t -v
Opening /dev/input/event0
Input Protocol version: 0x00010001
Testing events. Please, press CTRL-C to abort.
1445263921.175918: event type EV_MSC(0x04): scancode = 0x19
1445263921.175918: event type EV_SYN(0x00).
1445263921.675923: event type EV_MSC(0x04): scancode = 0x19
1445263921.675923: event type EV_SYN(0x00).
1445263922.675918: event type EV_MSC(0x04): scancode = 0x19
1445263922.675918: event type EV_SYN(0x00).
1445263923.175911: event type EV_MSC(0x04): scancode = 0x19
1445263923.175911: event type EV_SYN(0x00).
```

Here some relevant syslog messages:

```
[    8.284203] lirc_dev: IR Remote Control driver registered, major 244 
[    8.330861] lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
[    8.606752] usb 1-1.5: dvb_usb_v2: downloading firmware from file 'dvb-usb-it9135-01.fw'
[    8.779392] usb 1-1.5: dvb_usb_af9035: firmware version=12.54.14.0
[    8.875566] af9033 3-0038: firmware version: LINK 12.54.14.0 - OFDM 2.47.14.0
[    9.212360] lirc_rpi: auto-detected active high receiver on GPIO pin 18
[    9.213022] lirc_rpi lirc_rpi: lirc_dev: driver lirc_rpi registered at minor = 0
[    9.213040] lirc_rpi: driver registered!
```

mk01, 2015-10-19:
@nikto-es 

does
```
modprobe ir-nec-decoder; modprobe ir-rc6-decoder 
```

make any difference ?

closed 2018-09-25 by f1vefour