xbianonpi/xbian #782: Build script troubles?

cmichal2, 2015-10-22:
I'm trying to build an xbian kernel for a cubox-i with some custom patches.
I'm trying to follow the instructions at:
https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages
but run into some troubles.
First,  $xbiangit --arch armhfjessie --action build
fails because it refers to a version number of xbian-package-repo (1.0.6) that doesn't exist. After replacing the version number in create.buildroot with one that exists, the same command fails due to a dependency problem with xbian-package-initramfs-tools: 

The following packages have unmet dependencies:
 xbian-package-repo : Depends: xbian-package-initramfs-tools (>= 20150918) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

Are there more recent build instructions somewhere else?

mkreisl, 2015-10-22:
@cmichal2 
I'm not sure if there is a more recent documentation about this.

But - for building kernel it is not necessary having a schroot enviromnent (` $xbiangit --arch armhfjessie --action build` does this), you only need a cross compiler if you don't use an arm system for building the kernel. 
In my case, I'm building my special kernels in a debian8 vm without having a schroot environment.

cmichal2, 2015-10-24:
Thanks. I did manage to build the imx6-4.1.y kernel. I'd love to try imx6-4.3.y, but the build fails with a bunch of unresolved symbols related to cec and edid. Is there a solution known for this, or is that a work in progress?

mk01, 2015-11-03:
@cmichal2 

4.3.y is (was) under development... best start will be with default xbian config - imx_v7_cbi_hb_xbian_defconfig.
```
ARCH=arm make imx_v7_cbi_hb_xbian_defconfig
ARCH=arm make menuconfig
.
.
.
```

cmichal2, 2015-11-05:
Thanks. The build does still fail with cec and edid issues, but perhaps that's just where it stands.

mk01, 2015-11-05:
@cmichal2 

due to dependencies mxc_edid & drm turned to M. change them back to Y.

```
CONFIG_FB_MXC_EDID=y
CONFIG_DRM=y
```