xbianonpi/xbian #783: No GUI audio after update

ilgrank, 2015-10-27:
Hi
I was on 14.2 since May, but since 15.2 was out, i did an upgrade all packages from the xbian ssh text interface.
Everything went smooth, but now I have no gui audio even if gui audio is enabled in the system->settings->system->Audio Output

I'm running on a Raspberry pi 2, connected via HDMI.
Can someone please help me?
Thanks! :)

mkreisl, 2015-11-06:
What's the setting of `System-Settings->Audio output->Keep audio device alive` and `Play GUI sounds`?

ilgrank, 2015-11-09:
Hi mkreisl
Play GUI sounds is set to "Only when playback is stopped" 
and Keep Audio device Alive is set to 1 minute 

Thanks for your reply

mkreisl, 2015-11-09:
@ilgrank 
Same settings here but I have GUI sounds

Are you using passthrough? If yes, then probably your volume settings is too low for hearing GUI sound - so disable passthrough - increase volume and try again please. 

My settings are -14/3 db

ilgrank, 2015-11-09:
I tried both with passthrough enabled and with passtrough disabled, same result.
Also tried setting the volume of the tv to the max (50/50), still no sound.
I tried also resetting the values to default (hdmi out, 2.0, no passthrough) and same result (no gui audio)
is there any log I should look in?
Btw, the audio of video files is fine.
Thanks

mkreisl, 2015-11-18:
No, unfortunately not.
The only thing you can try is starting kodi with a fresh environment:
```
sudo stop xbmc
cd /home/xbian
mv .kodi .kodi.bak
sudo start xbmc
```
and see if you have a GUI sound or not.

mkreisl, 2015-11-18:
@ilgrank 
Just tested with RPi2 and fresh environment (see my previous post). 

Works perfectly

closed 2017-10-31 by mkreisl