xbianonpi/xbian #785: Can't upgrade old image pn Rpi 2

coderpop, 2015-10-29:
After the latest upgrade broke I2S audio, I re-flashed the SD card to an image from a month ago.
It didn't have all the Jessie updates and now when I try to update hte 388 packages available I get the error "There are problems and -y was used without --force-yes" - Even tried one package at a time and still the same problem.

How do I proceed to get this updated and working again please?

Thanks.

mkreisl, 2015-10-29:
what says `apt-get install -f`?

coderpop, 2015-10-30:
Thanks again for your time and help.

It says....
sudo apt-get install -f
[sudo] password for xbian:
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 389 not upgraded.

Next I give you the following as it might help....

cat /etc/apt/sources.list.d/xbian.list
deb mirror://apt.xbian.org/mirror.txt  stable  main rpi2-jessie
deb mirror://apt.xbian.org/mirror.txt  staging main rpi2-jessie
deb mirror://apt.xbian.org/mirror.txt  devel   main rpi2-jessie

cat /etc/apt/sources list....
deb http://http.debian.net/debian stable main non-free

This looked wrong and if you look at http://www.xbian.org/faq/ it states it should be....
deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi

So I tried that and only got about 20 packages to update/upgrade but it also gives even more errors......

I'm sure it's something simple, but I don't know what to do next!

mkreisl, 2015-10-30:
First, i would deactivate staging and devel repo
```
deb mirror://apt.xbian.org/mirror.txt stable main rpi2-jessie
### deb mirror://apt.xbian.org/mirror.txt staging main rpi2-jessie
### deb mirror://apt.xbian.org/mirror.txt devel main rpi2-jessie
```
Second, you are using RPi2, so 
```
deb http://http.debian.net/debian stable main non-free
```
is correct

And, assuming you are already on Jessie, don't use `rpi2-wheezy`, because this repo has never exist

Are you using `xbian-config` for doing updates or do you use `apt-get update/apt-get upgrade`?
I would prefer 2nd method.

What says `lsb_release -cs`?

coderpop, 2015-10-30:
lsb_release -cs says wheezy!

Have returned settings as you said above, the Xbian site needs to be updated by admins to reflect this or others might have the same problems....

coderpop, 2015-10-31:
Fully upgraded now.

Only issue is with I2S audio. I get frequent drop outs on the sound (several times per track).

Let me know if I can help.

mkreisl, 2015-10-31:
I'm a little bit confused. 

You told here https://github.com/xbianonpi/xbian/issues/770#issuecomment-147859818 everything is find, what has been changed?

coderpop, 2015-11-01:
Something odd is happening. Sound via I2S is fine now , only rebooted a couple of times. But, strangely, the I2S seems to cut off when the projector connected to the HDMI is turned off. It only just happened and I will try and test it some more tomorrow.

coderpop, 2015-11-05:
Upgraded today and all issues seem to be fixed (at least the ones that were affecting my installation).

Will test some more, but looking good via ssh right now!

Thanks for a good job team!

closed 2015-11-06 by coderpop