xbianonpi/xbian #787: Empty Packages in apt repository of rpi-jessie

agijsberts, 2015-11-08:
Due to the horror that is USB wifi on linux, I'm in the process of putting a 2 month-old backup on my RPi-B. Unfortunately, `apt-get upgrade` is not listing updates for kodi (among others):

    root@rpi:~# apt-cache policy xbian-package-xbmc
    xbian-package-xbmc:
      Installed: 14.2-1431026245
      Candidate: 14.2-1431026245
      Version table:
     *** 14.2-1431026245 0
            100 /var/lib/dpkg/status

Apt sources seem to be fine:
```
    root@rpi:~# cat /etc/apt/sources.list /etc/apt/sources.list.d/xbian.list
    deb http://mirrordirector.raspbian.org/raspbian/ stable main non-free
    deb mirror://apt.xbian.org/mirror.txt      stable  main rpi-jessie
```

While investigating, I noticed that the Packages file in the `rpi-jessie` distribution is empty (for instance: http://apt.xbian.org/dists/stable/rpi-jessie/binary-armhf/Packages). Comparing with the other distributions (e.g., `rpi2-jessie`), it seems that this source should contain the xbian specific packages. Is this assumption correct, and if so, what can I do about it?

mkreisl, 2015-11-08:
@agijsberts 
Fixed

agijsberts, 2015-11-08:
Much appreciated!

closed 2015-11-09 by mkreisl