xbianonpi/xbian #789: Kodi aborts

kev1807, 2015-11-21:
Hi It's actually pretty annoying: 

I update and upgrade my Pi B, running Xbian ( normally it is up to date...) and after some days it keeps stopping kodi from running. 

It starts running and after some time kodi stops and a terminal opens. 

´´´
login: done.
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...done.
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...done.
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...pidof [4254]: can't get program name from /proc/4247/stat

done.
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...

´´´
I tried downgrading to an earlier version using
apt-get install xbian-update=20150919-1

now nothing opens up at all... 

please help me & fix the reoccuring problem.

best wishes, 
Kevin

mkreisl, 2015-11-21:
@kev1807 
You can try using an older snapshot, see http://wiki.xbian.org/doku.php/snapshots

sdoney, 2015-11-25:
I get the same behavior on new install and composite. See issue #790 for kodi log.

daleye, 2015-12-06:
I have this same issue with the latest build on RPI B+, Kodi runs for about 5-10 minutes and then aborts.

mkreisl, 2015-12-06:
Full Kodi debug logs might be useful. To enable this, add 
```
# Enable full debug
DEBUG='--debug'
```
to file `/etc/default/xbmc`

The logfile kodi.log can be found in `/home/xbian/.kodi/temp`

mk01, 2015-12-11:
@mkreisl 

and last boot (~150lines) from /var/log/upstart/upstart-ev.log
(there would be logged the xbmc-loaded - which is missing -> kodi is killed)

rlogiacco, 2015-12-13:
I've tried everything on my RPi B+ previously working nicely, but now it just crashes after a couple of minutes since startup completed going into black screen. Logging in via SSH and issuing a reboot doesn't work: the thing is completely stuck, even if still responding to SSH.
Is this something being looked into or should we just move on?
I've tried multiple setups, going backward to Halloween release (31/10) still with no joy. The only time the rpi runs nicely is after first boot. Any reboot brings it to this stall.

mkreisl, 2015-12-13:
@rlogiacco @daleye @sdoney 
No debug logs -> no help, our crystal balls are all in maintance

rlogiacco, 2015-12-13:
Here is the full content of the above mentioned file:
```
17:07:40  62.713142 T:3020648448   DEBUG: Previous line repeats 19 times.
17:07:40  62.713547 T:3020648448  NOTICE: special://profile/ is mapped to: special://masterprofile/
17:07:40  62.713821 T:3020648448  NOTICE: -----------------------------------------------------------------------
17:07:40  62.714214 T:3020648448  NOTICE: Starting Kodi (15.2 Git:2015-10-27-b8129c6). Platform: Linux ARM 32-bit
17:07:40  62.714489 T:3020648448  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
17:07:40  62.714767 T:3020648448  NOTICE: Kodi compiled Oct 27 2015 by GCC 4.9.2 for Linux ARM 32-bit version 3.18.8 (201224)
17:07:40  62.715797 T:3020648448  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 4.1.11+
17:07:40  62.716202 T:3020648448  NOTICE: FFmpeg statically linked, version: 2.6.4-kodi-2.6.4-Isengard
17:07:40  62.716465 T:3020648448  NOTICE: Host CPU: ARMv6-compatible processor rev 7 (v6l), 1 core available
17:07:40  62.716698 T:3020648448  NOTICE: ARM Features: Neon disabled
17:07:40  62.717049 T:3020648448  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
17:07:40  62.717308 T:3020648448  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
17:07:40  62.717548 T:3020648448  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
17:07:40  62.717793 T:3020648448  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
17:07:40  62.718109 T:3020648448  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
17:07:40  62.718487 T:3020648448  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
17:07:40  62.718952 T:3020648448  NOTICE: Local hostname: xbian
17:07:40  62.719231 T:3020648448  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
17:07:40  62.719666 T:3020648448  NOTICE: -----------------------------------------------------------------------
17:07:44  66.777748 T:3020648448   DEBUG: UPower: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
17:07:44  66.910248 T:3020648448   DEBUG: DeviceKit.Power: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.DeviceKit.Disks was not provided by any .service files
17:07:44  66.922401 T:3020648448   DEBUG: UPower: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
17:07:44  66.923332 T:3020648448  NOTICE: load settings...
17:07:44  67.120331 T:3020648448   DEBUG: CSettings: loaded settings definition from special://xbmc/system/settings/settings.xml
17:07:44  67.234505 T:3020648448   DEBUG: CSettings: loaded settings definition from special://xbmc/system/settings/rbp.xml
17:07:44  67.247864 T:3010458544  NOTICE: Thread FDEventMonitor start, auto delete: false
17:07:44  67.315880 T:3020648448    INFO: CAESinkALSA - Unable to open device "surround71" for playback
17:07:44  67.320900 T:3020648448    INFO: CAESinkALSA - Unable to open device "surround51" for playback
17:07:44  67.321442 T:3020648448    INFO: CAESinkALSA - Unable to open device "surround71" for playback
17:07:44  67.326530 T:3020648448    INFO: CAESinkALSA - Unable to open device "surround40" for playback
17:07:44  67.327072 T:3020648448    INFO: CAESinkALSA - Unable to open device "surround51" for playback
17:07:44  67.327522 T:3020648448    INFO: CAESinkALSA - Unable to open device "surround71" for playback
17:07:44  67.436684 T:3020648448  NOTICE: Found 2 Lists of Devices
17:07:44  67.437065 T:3020648448  NOTICE: Enumerated PI devices:
17:07:44  67.437309 T:3020648448  NOTICE:     Device 1
17:07:44  67.438889 T:3020648448  NOTICE:         m_deviceName      : HDMI
17:07:44  67.439217 T:3020648448  NOTICE:         m_displayName     : HDMI
17:07:44  67.439468 T:3020648448  NOTICE:         m_displayNameExtra:
17:07:44  67.439713 T:3020648448  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
17:07:44  67.439949 T:3020648448  NOTICE:         m_channels        : FL,FR
17:07:44  67.440300 T:3020648448  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
17:07:44  67.440636 T:3020648448  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
17:07:44  67.441048 T:3020648448  NOTICE:     Device 2
17:07:44  67.441528 T:3020648448  NOTICE:         m_deviceName      : Analogue
17:07:44  67.441795 T:3020648448  NOTICE:         m_displayName     : Analogue
17:07:44  67.442123 T:3020648448  NOTICE:         m_displayNameExtra:
17:07:44  67.442390 T:3020648448  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
17:07:44  67.442642 T:3020648448  NOTICE:         m_channels        : FL,FR
17:07:44  67.442879 T:3020648448  NOTICE:         m_sampleRates     : 48000
17:07:44  67.443130 T:3020648448  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
17:07:44  67.443390 T:3020648448  NOTICE:     Device 3
17:07:44  67.443825 T:3020648448  NOTICE:         m_deviceName      : Both
17:07:44  67.444168 T:3020648448  NOTICE:         m_displayName     : HDMI and Analogue
17:07:44  67.444427 T:3020648448  NOTICE:         m_displayNameExtra:
17:07:44  67.444786 T:3020648448  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
[... cut ...]
18:09:44 151.762894 T:3020648448    INFO: CheckIdle - Closing session to http://xml.superrepo.org (easy=0x9e622908, multi=0xb2d7b3d8)
18:09:54 161.078537 T:3020648448   DEBUG: checkVTchange : out VT being deactivated
18:09:56 163.435715 T:2710905776   DEBUG: Thread JobWorker 2710905776 terminating (autodelete)
18:09:58 165.611816 T:2786403248   DEBUG: Thread JobWorker 2786403248 terminating (autodelete)
```

This is xbian 31/10 running on RPi B+ with eminence skin and no other plugins other than Yahoo Weather

mkreisl, 2015-12-13:
@rlogiacco Thanks
Did you do anything before Kodi crashes? 
```
18:09:54 161.078537 T:3020648448   DEBUG: checkVTchange : out VT being deactivated
```
It seems that there is a strange VT change, but who is doing that???. I investigated my logfiles, I've never seen this message. @mk01 Could this be the culprit?

rlogiacco, 2015-12-13:
> Did you do anything before Kodi crashes?
Not even touching the keyboard, just logged in via SSH, but after the crash.
What is a VT change?

I actually thought the issue could be related to the plugin directory failed read...

After another couple of restarts now instead of hanging into a black screen it fails down to console login and repeatedly printing
```
[ok]Setting up X socket directories .../tmp/.x11-unix/tmp/.ICE-unix.
[ok]Setting preliminary keymap...
```

I downgraded to the mentioned version due to suggestions related to this issue, which is still present on release 31/10...

mkreisl, 2015-12-13:
VT --> Virtual Terminal
you can switch the virtual termial by pressing ctrl-alt-[1...5] (if a keyboard is connected) or `chvt <VTnumber>` by command (AFAIK Kodi runs on VT 5, 1-4 are text consoles).
But in the past, on RPi's this does not working without errors. Not sure if this has been already fixed

> and last boot (~150lines) from /var/log/upstart/upstart-ev.log
> (there would be logged the xbmc-loaded - which is missing -> kodi is killed)

Now this log could have been more information why kodi does not come up (and of course kodi.log)

rlogiacco, 2015-12-13:
I'm not sure I do understand: are you asking for that `/var/log/upstart/upstart-ev.lo` as well?

mkreisl, 2015-12-13:
Yes

rlogiacco, 2015-12-13:
Here it is. It's a subsequent reboot, so no time correlation is possible, and it has been running for a while now.
I've tried to switch terminal using CTRL+ALT+F# with # in 1-5, but I can see consoles only at 1 and 2, while 3 and 5 show a blinking cursor and 4 a black screen without any cursor....

```
Sun Dec 13 20:24:08 UTC 2015 163.32 0.15 failsafe-boot                            14.31 5.74 2.12 5/216 2154
Sun Dec 13 20:24:11 UTC 2015 166.19 0.15 started zram-swap                        13.40 5.70 2.13 8/228 2199
Sun Dec 13 20:24:11 UTC 2015 166.20 0.15 started tty1                             13.40 5.70 2.13 7/225 2199
Sun Dec 13 20:24:11 UTC 2015 166.27 0.15 started tty2                             13.40 5.70 2.13 6/222 2199
Sun Dec 13 20:24:13 UTC 2015 167.67 0.15 stopped xbmc-preload                     13.40 5.70 2.13 4/219 2213
Sun Dec 13 20:24:15 UTC 2015 169.84 0.15 starting mountall                        13.40 5.70 2.13 6/225 2230
Sun Dec 13 20:24:15 UTC 2015 169.85 0.15 started xbian-run                        13.40 5.70 2.13 5/221 2230
Sun Dec 13 20:24:19 UTC 2015 173.63 0.15 stopped hostname.sh                      13.05 5.75 2.17 8/232 2259
Sun Dec 13 20:24:19 UTC 2015 173.59 0.15 started hostname.sh                      13.05 5.75 2.17 8/232 2259
Sun Dec 13 20:24:19 UTC 2015 173.64 0.15 started network-interface dummy0         13.05 5.75 2.17 5/226 2259
Sun Dec 13 20:24:20 UTC 2015 174.77 0.15 started wait-for-state  xbmc-exits       13.05 5.75 2.17 4/222 2269
Sun Dec 13 20:24:22 UTC 2015 176.77 0.15 started mountall                         12.57 5.77 2.19 6/225 2283
Sun Dec 13 20:24:28 UTC 2015 183.35 0.15 xbmc-rc                                  12.20 5.81 2.22 7/230 2335
Sun Dec 13 20:24:28 UTC 2015 183.35 0.15 stopped mountall                         12.20 5.81 2.22 6/227 2335
Sun Dec 13 20:24:29 UTC 2015 184.31 0.15 started xbmc-preload                     12.20 5.81 2.22 6/226 2341
Sun Dec 13 20:24:31 UTC 2015 186.22 0.15 stopped wait-for-state  xbmc-exits       11.54 5.78 2.23 6/220 2359
Sun Dec 13 20:24:32 UTC 2015 187.22 0.15 started xbmc-kill-splash                 11.54 5.78 2.23 5/225 2385
Sun Dec 13 20:24:32 UTC 2015 187.23 0.15 stopped xbmc-preload                     11.54 5.78 2.23 5/222 2386
Sun Dec 13 20:24:34 UTC 2015 188.37 0.15 started xbian-chvt                       11.54 5.78 2.23 6/221 2397
Sun Dec 13 20:24:34 UTC 2015 188.74 0.15 started xbmc-failed-start                11.54 5.78 2.23 4/220 2403
Sun Dec 13 20:24:34 UTC 2015 189.14 0.15 stopped xbian-chvt                       11.54 5.78 2.23 4/218 2409
Sun Dec 13 20:24:35 UTC 2015 189.60 0.15 stopped xbmc-kill-splash                 11.54 5.78 2.23 4/218 2416
Sun Dec 13 20:24:58 UTC 2015 213.01 0.15 stopping xbmc                            8.43 5.51 2.24 3/214 2497
Sun Dec 13 20:25:00 UTC 2015 214.56 0.15 starting xbmc-exits                      8.43 5.51 2.24 7/191 2543
Sun Dec 13 20:25:00 UTC 2015 214.57 0.15 started xbmc-changesettings              8.43 5.51 2.24 8/191 2543
Sun Dec 13 20:25:00 UTC 2015 214.55 0.15 stopped xbmc                             8.43 5.51 2.24 6/188 2543
Sun Dec 13 20:25:00 UTC 2015 214.87 0.15 stopped xbian-xbmc-cpufreq               8.43 5.51 2.24 5/182 2548
Sun Dec 13 20:25:01 UTC 2015 215.52 0.15 stopped tvonoff-xbmc                     8.08 5.48 2.25 4/180 2555
Sun Dec 13 20:25:01 UTC 2015 216.13 0.15 stopped xbmc-changesettings              8.08 5.48 2.25 5/182 2566
Sun Dec 13 20:25:02 UTC 2015 216.54 0.15 started xbmc-exits                       8.08 5.48 2.25 4/180 2579
Sun Dec 13 20:25:03 UTC 2015 217.41 0.15 stopped xbmc-exits                       8.08 5.48 2.25 3/178 2593
Sun Dec 13 20:25:08 UTC 2015 222.92 0.15 stopped setupcon                         7.59 5.43 2.25 2/176 2612
Sun Dec 13 20:25:12 UTC 2015 226.83 0.15 stopped rc-sysinit                       7.06 5.35 2.24 6/181 2698
Sun Dec 13 20:25:12 UTC 2015 226.84 0.15 runlevel                                 7.06 5.35 2.24 6/181 2698
Sun Dec 13 20:25:12 UTC 2015 227.15 0.15 started rc                               7.06 5.35 2.24 5/177 2709
Sun Dec 13 20:25:13 UTC 2015 227.59 0.15 started anacron                          7.06 5.35 2.24 5/182 2728
Sun Dec 13 20:25:13 UTC 2015 227.60 0.15 started cron                             7.06 5.35 2.24 5/181 2728
Sun Dec 13 20:25:13 UTC 2015 227.88 0.15 stopped anacron                          7.06 5.35 2.24 4/179 2740
Sun Dec 13 20:25:15 UTC 2015 230.14 0.15 started xbian-inotifyd-bridge            6.98 5.36 2.26 3/180 2793
Sun Dec 13 20:25:16 UTC 2015 230.43 0.15 stopped restore-home                     6.98 5.36 2.26 2/178 2800
Sun Dec 13 20:25:23 UTC 2015 237.35 1.09 stopped rc                               6.50 5.29 2.25 1/174 2861
Sun Dec 13 20:25:30 UTC 2015 244.49 7.31 net-device-up  eth0                      5.98 5.20 2.24 2/177 2895
Sun Dec 13 20:25:31 UTC 2015 245.26 7.35 stopped failsafe-boot-started            5.58 5.13 2.23 1/171 2933
Sun Dec 13 20:26:14 UTC 2015 288.83 46.37 failsafe-boot                            2.91 4.50 2.14 2/175 3030
Sun Dec 13 20:26:14 UTC 2015 289.02 46.37 started rc-sysinit                       2.91 4.50 2.14 4/176 3039
Sun Dec 13 20:26:15 UTC 2015 289.89 46.37 starting mountall                        2.91 4.50 2.14 6/190 3077
Sun Dec 13 20:26:15 UTC 2015 289.88 46.37 started xbian-run                        2.91 4.50 2.14 6/190 3077
[... cut ...]
Sun Dec 13 21:33:45 UTC 2015 4339.35 1661.25 stopped rc-sysinit                       2.02 1.50 1.38 5/218 29526
Sun Dec 13 21:33:45 UTC 2015 4339.35 1661.25 runlevel                                 2.02 1.50 1.38 5/218 29526
Sun Dec 13 21:33:45 UTC 2015 4339.60 1661.25 started rc                               2.02 1.50 1.38 4/215 29537
Sun Dec 13 21:33:45 UTC 2015 4339.88 1661.25 started anacron                          2.02 1.50 1.38 4/214 29547
Sun Dec 13 21:33:45 UTC 2015 4340.11 1661.25 stopped anacron                          2.02 1.50 1.38 4/216 29558
```

rlogiacco, 2015-12-24:
Any news on this? I'm about to switch distribution, my media center cannot remain down during holidays....

daleye, 2015-12-24:
I know I never posted a log but I just loaded osmc and my issue has gone away. Works for my application which is extremely basic.

rlogiacco, 2015-12-24:
well, I have been happy with xbian for quite a while, I would prefer to stay with xbian if possible. this is an open source project, the guys don't get paid to work on this or provide us a working distro, so we cannot pretend anything, but before switching I wish to double check with the guys here. on the other end it's tough to explain this all to a 6 y/o boy asking to watch his favorite movie during xmas holidays....

LauwS, 2016-01-04:
This has happened a fem times for me as well. I have to remote in and do a reboot before i can use kodi again.

closed 2016-06-17 by mkreisl