xbianonpi/xbian #791: Webserver and JSON not starting after upgrade

er0l, 2015-12-04:
After upgrade to xbian-package-xbmc_15.2-1448685125_armhf.deb and today also to xbian-package-xbmc_15.2-1449200414_armhf.deb webserver and json are not started.

Issue also reported on the forums:
http://xbian.org/forum/thread-3233.html

Even though I tried to force this in the advancedsettings.xml it is not getting read. 

If I make the change manually from the GUI, for example it is showing that web server is enabled ti is not working, but if I disable it and enable again it gets working again, but this change will not survive a restart.

It seems that settings do not get written or read from guisettings.xml properly since I tried even changing them manually.

er0l, 2015-12-04:
Reverting back to 15.2-1446623926 seems to get it working again.

apt-get install xbian-package-xbmc=15.2-1446623926

tcp 0 0 0.0.0.0:8080 0.0.0.0:* LISTEN 15403/kodi.bin
tcp 0 0 0.0.0.0:9090 0.0.0.0:* LISTEN 15403/kodi.bin

mk01, 2015-12-04:
@er0l 

yup, right
...
new packages are being compiled http://jenkins.xbian.org/job/xbian-sources-xbmc/238/changes

The-Exnor, 2015-12-04:
Same problem here... did not reverted yet.

er0l, 2015-12-05:
I can confirm that latest package 15.2-1449275614 fixes this issue.

thanks

The-Exnor, 2015-12-06:
Same here. Today's update have resolved the issue (Model B Pi 1)

closed 2015-12-06 by CurlyMoo