xbianonpi/xbian #792: WakeOnAccess/Ping problem on RPi2 with xbian_2015.11.29.img

rgls, 2015-12-04:
Just installed a Raspberry Pi2 with the latest xbian image. Couldn't get WakeOnAccess to work. It never found my fileserver, even when it was already running. A little sniffing with Wireshark revealed that xbian didn't try to ping 192.168.100.5 (my fileservers IP) but 5.100.168.192! A quick verification by renaming ping and putting a shell script with some logging in it's place confirmed this. This script allowed a workaround by flipping the address bytes using awk.
Maybe the recent changes to support IPv6 broke the endianness in the IPv4 code?

mk01, 2015-12-04:
@rgls 

yes, there was conversion twice (once in wakeonlan, also in new networking code). I reviewed wakeonlan and fixed - now it works on my Mac and IMX6, RPIs should too.

there is new builds already running http://jenkins.xbian.org/job/xbian-sources-xbmc/239/, please report back after test.
we have to push then to Jarvis branch and to upstream xbmc too.

thx

rgls, 2015-12-05:
Jenkins seems to have finished the build. Where do I find the resulting binaries/packages/image?
I would prefer a binary or package to a complete image, as this installation has already been configured and tweaked quite a bit.
Which reminds me: One of the tweaks was to rename /usr/local/bin/ping as this is a link to busybox which works only as ping if you are root. This shadowed the /bin/ping which is from iputils and works for non-root users (as xbian) as well.

mk01, 2015-12-05:
APT repo inventory is updated in 4h intervals. so few minutes after 4:00am CET (in 2h) the new packages will become "available".

mk01, 2015-12-05:
@rgls 

regarding the ping, ... yes, ping is normally setUID flagged, if busybox is, works then. so we can do that. we can install iputils-ping. we can .... ? any more idea? do you have preference ?

rgls, 2015-12-05:
SetUID on busybox is actually not such a good idea, as it would mean **EVERY** command that is a link to busybox is executed as root. So please don't do that. My suggestion is: Ditch busybox-ping and use iputils-ping.
Regarding the nightly build: I'll download it later and hopefully have the time to test it this weekend. I'll let you know as soon as I did.
This is the url of the package containing the fix, right?
http://apt.xbian.org/pool/stable/rpi2-jessie/x/xbian-package-xbmc/xbian-package-xbmc_15.2-1449275614_armhf.deb

mk01, 2015-12-05:
@rgls 

I pushed the fix into treeIsengard directly and that was built to stable repo, so you can just update regular way.

rgls, 2015-12-06:
I just updated to the latest version in the stable repo. With this version I can't see any ping attempts at all. Neither in the log of my script nor in the capture written by the sniffer.

rgls, 2015-12-06:
Furthermore the behaviour has changed.
With the old version and without the flip script it was:
-Waiting for network 1-100%
-Waiting for server to wake up 1-100%
-Extended wait for server to wake up 1-100%
This is expected as it never got a reply to the ping. Kodi thought the server was still down and showed the message that the movie is not available.

Old version + flip script:
-Waiting for network 1-100%
-Waiting for server to wake up 1-85%
-Waiting for services 1-100%
The server woke up, the additional delay for the services to come up went by and the movie started.

New version:
-Waiting for network 1-100%
-Waiting for server to wake up 1%
Kodi shows the "Movie not available" message immediately after starting the wait for the server, so probably something goes wrong when it tries the first ping.
Maybe writing the commandline to the (debug) log before calling system() could clear things up.