xbianonpi/xbian #793: Installer GUI

Zac610, 2015-12-08:
On xubuntu 14.04, the graphic installer has some GUI glitches:
- installer version is too near the window border
- the downloading textfield is too little for the text inside

here is the screenshot:
[image]

(this is an easy issue ;)

bairdy, 2015-12-08:
I have a fix to be pushed as well as an update to be able to restore a backed up image. I picked up on it testing restoring images in different Linux distro's and osx.

skyway22, 2015-12-24:
Also sometimes on ubuntu 14.04, the installer freezes my pc. I only flash from terminal..

bairdy, 2016-01-07:
@Zac610 ive fixed the GUI and have checked on standard ubuntu 14.04 and on windows you can get a compiled copy from here https://www.dropbox.com/s/bb9lrkdtola81i8/Xbian-Installer-Linux-1.3.zip?dl=0 to try

mk01, 2016-01-23:
@bairdy 

All is perfect now, even tried running in remote X session running on Mac and proportions / sizes have no single issue (perhaps the only X app running this way I use)

:+1:

Zac610, 2016-04-05:
Problem solved, great job!

[image]

closed 2016-04-05 by Zac610