xbianonpi/xbian #795: imx6 cubox-i no audio output

Implexus, 2015-12-27:
There is no audio output for HDMI, analog, also SPDIF since kernel update linux-image-armmp 4.3.0+-1449787808

w00ty, 2016-01-07:
Same issue here. I have reverted to an older build. Also tried clean install of the newer build, also no sound. I have not tried any builds after december. Does anyone knows if the issue is solved in the latest build?

mkreisl, 2016-01-07:
I does not have such issues. Used self compiled 4.3.3 and version from APT.

Sometimes I have no sound when starting a video, so I have to stop and start again. And sometimes I have for some seconds no GUI sound after stopping video. But this happens with all used kernels (4.1.x, 4.3.x, 4.4.x)

Your sound settings are correct?

Implexus, 2016-01-08:
Hello,

yes, all my sound settings are correct since it's working without any issues with the older kernel.

I've just tried the latest kernel linux-image-armmp 4.3.3+-1450885118 and the issue still persists. Going to revert back to the latest kernel functional version 4.1.13+-1449274499.

mkreisl, 2016-01-09:
I can confrm the issue.
It happens on my Hummingboard Pro after a longer power off (> 10s). All was fine if no power loss occours, all kernels worked before.

@mk01 It seems that some hardware parts are not enabled in dtb file for kernel 4.3.x and 4.4.x. Can you please test it? Thanks

For the moment  I have removed all 4.3.x kernel packets from repositories and a new image will be build tonight

mk01, 2016-01-12:
@Implexus @mkreisl @w00ty 

SPDIF got broken some DT updates from vanilla upstream. It is fixed in tree 4.3 and 4.4 as well now.
Regarding the other reports - just put for me down some detailed steps - so I can reproduce.

w00ty, 2016-01-12:
@mk01 @mkreisl  Thanks! Will try it out later this week. Have to find some free time first! :)

mkreisl, 2016-01-12:
@mk01 
At the moment sound is ok,. Don't ask me what is different. Sound has survived several power offs, reboots, and even multiple changes of sd-card [1] did not bring back the issue.

[1] It seems that kernel 4.4.0-rc8 corrupts the btrfs fs. It happens twice during after an apt-get update command. apt-get stopped updating and a btrfs scrub / reported errors, stopped and no access to / was possible. This happend twice so I returned to 4.3.3 the problem disappeared.

thewebexpert, 2016-01-12:
unsubscribe

mk01, 2016-01-12:
@mkreisl 

that's why perhaps why there are Betas and RCs. RC 4.4 kernels had broken SD/MMC. I was reporting that a week/two ago somewhere.

but since early Sunday 4.4.0 stable is out. I rewrote lot of imx6 things for 4.4, most of them back-ported to 4.3. branch and re-released this as well (last release of 4.3 from my side). 4.4 is now new LTS.

mkreisl, 2016-01-12:
So, can/should I give 4.4 a new try  :question: 

> that's why perhaps why there are Betas and RCs. RC 4.4 kernels had broken SD/MMC. I was reporting

I'm wondering why those extremely important things are always changed from version to version :disappointed: 

> 4.4 is now new LTS.

I know :smile:

mk01, 2016-01-12:
@mkreisl 
heh funny question if you 'can' :) sure you can, after you buy unlimited CPU software license. as per CPU we sell only up to 3xcpu, per seat we don't at all with 4.4.y.

mkreisl, 2016-01-12:
Where can I buy such license :blush: Or can I remove one core from the CPU :smiley:

mk01, 2016-01-13:
@mkreisl 
give me few more minutes, just registering company on cayman islands.

CurlyMoo, 2016-01-13:
> I'm wondering why those extremely important things are always changed from version to version :disappointed: 

Because nobody takes the time to write a XBian manual just as i created the pilight manual... https://manual.pilight.org/

mk01, 2016-01-13:
@CurlyMoo 
yes, one day we will document even upstream vanilla kernel development in XBian manual.
or maybe share those parts with kernel documentation in pilight manual ;)

we can agree on details when maintainers will be rewriting kernel for 4,5 release.

Implexus, 2016-01-15:
@mk01 thank you, sound output is working again!

closed 2016-01-15 by Implexus

mk01, 2016-01-23:
@Implexus 

just to have it covered informationally - what was your resolution ? 4.4.y usage, or 4.3 fixed or another?

Implexus, 2016-01-23:
@mk01 
I have updated to 4.4.x kernel, but I can try also 4.3 if you want.

mk01, 2016-01-23:
@Implexus 

no, all fine. thanks.