xbianonpi/xbian #796: Composite / S-Video - Blackscreen problem

suuhm, 2015-12-27:
HDMI is working but on the composite i've got a black screen and only sound is working but no screen

here my logs:

20:58:36  48.150745 T:3020312576  NOTICE: -----------------------------------------------------------------------
20:58:36  48.151047 T:3020312576  NOTICE: Starting Kodi (15.2-RC3 Git:2015-09-29-57da20e). Platform: Linux ARM 32-bit
20:58:36  48.151287 T:3020312576  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
20:58:36  48.151554 T:3020312576  NOTICE: Kodi compiled Sep 30 2015 by GCC 4.9.2 for Linux ARM 32-bit version 3.18.8 (201224)
20:58:36  48.152596 T:3020312576  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 4.1.6+
20:58:36  48.152897 T:3020312576  NOTICE: FFmpeg statically linked, version: 2.6.4-kodi-2.6.4-Isengard
20:58:36  48.153130 T:3020312576  NOTICE: Host CPU: ARMv6-compatible processor rev 7 (v6l), 1 core available
20:58:36  48.153343 T:3020312576  NOTICE: ARM Features: Neon disabled
20:58:36  48.153572 T:3020312576  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
20:58:36  48.153809 T:3020312576  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
20:58:36  48.154198 T:3020312576  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
20:58:36  48.154453 T:3020312576  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
20:58:36  48.154682 T:3020312576  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
20:58:36  48.155006 T:3020312576  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
20:58:36  48.155437 T:3020312576  NOTICE: Local hostname: xb
20:58:36  48.155701 T:3020312576  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
20:58:36  48.156113 T:3020312576  NOTICE: -----------------------------------------------------------------------

cat /home/xbian/.xbmc/temp/kodi.log | grep ERROR ;tail -n 99 /var/log/upstart/upstart-ev.log
20:58:42  53.915012 T:3020312576   ERROR: DBus: Error org.freedesktop.UDisks.Error.Failed - Error mounting: mount exited with exit code 1: helper failed with:
20:58:43  54.846584 T:3020312576   ERROR: UpdateResolutions: Fatal Error, GetPreferredResolution failed
20:58:45  56.891598 T:3020312576   ERROR: Couldn't load font name: font10(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.100471 T:3020312576   ERROR: Couldn't load font name:font10 file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.102718 T:3020312576   ERROR: Couldn't load font name: font12(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.107708 T:3020312576   ERROR: Couldn't load font name:font12 file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.109936 T:3020312576   ERROR: Couldn't load font name: font13(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.114876 T:3020312576   ERROR: Couldn't load font name:font13 file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.117210 T:3020312576   ERROR: Couldn't load font name: font14(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.122143 T:3020312576   ERROR: Couldn't load font name:font14 file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.124489 T:3020312576   ERROR: Couldn't load font name: font16(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.129395 T:3020312576   ERROR: Couldn't load font name:font16 file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.131756 T:3020312576   ERROR: Couldn't load font name: font30(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.136658 T:3020312576   ERROR: Couldn't load font name:font30 file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.138885 T:3020312576   ERROR: Couldn't load font name: fontContextMenu(roboto-regular.ttf), trying to substitute arial.ttf
20:58:45  57.143898 T:3020312576   ERROR: Couldn't load font name:fontContextMenu file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.159332 T:3020312576   ERROR: Couldn't load font name: font10_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.164402 T:3020312576   ERROR: Couldn't load font name:font10_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.166637 T:3020312576   ERROR: Couldn't load font name: font12_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.171734 T:3020312576   ERROR: Couldn't load font name:font12_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.173954 T:3020312576   ERROR: Couldn't load font name: font13_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.178989 T:3020312576   ERROR: Couldn't load font name:font13_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.181328 T:3020312576   ERROR: Couldn't load font name: font24_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.186234 T:3020312576   ERROR: Couldn't load font name:font24_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.188583 T:3020312576   ERROR: Couldn't load font name: font28_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.193497 T:3020312576   ERROR: Couldn't load font name:font28_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.195831 T:3020312576   ERROR: Couldn't load font name: font30_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.200768 T:3020312576   ERROR: Couldn't load font name:font30_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.202999 T:3020312576   ERROR: Couldn't load font name: font35_title(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.207958 T:3020312576   ERROR: Couldn't load font name:font35_title file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.210262 T:3020312576   ERROR: Couldn't load font name: font_MainMenu(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.215256 T:3020312576   ERROR: Couldn't load font name:font_MainMenu file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:45  57.217499 T:3020312576   ERROR: Couldn't load font name: WeatherTemp(roboto-bold.ttf), trying to substitute arial.ttf
20:58:45  57.222538 T:3020312576   ERROR: Couldn't load font name:WeatherTemp file:/usr/local/share/kodi/media/Fonts/arial.ttf
20:58:49  61.061028 T:3020312576   ERROR: ExecuteAsync - Not executing non-existing script /home/xbian/.kodi/addons/script.navi-x/service.py
20:59:11  83.053375 T:2812978096   ERROR: CCurlFile::Open failed with code 404 for http://repo.skinquantum.de/nerdsrepo/allgemein/addons.xml.md5
20:59:11  83.053825 T:2812978096   ERROR: Open - failed to open source <http://repo.skinquantum.de/nerdsrepo/allgemein/addons.xml.md5>

Sun Dec 27 20:57:55 CET 2015 6.91 0.15 stopped hwclock-systemd-disable          0.16 0.03 0.01 9/90 408
Sun Dec 27 20:57:55 CET 2015 6.93 0.15 started portmap                          0.16 0.03 0.01 9/88 408
Sun Dec 27 20:57:55 CET 2015 6.94 0.15 starting network-interface eth0          0.16 0.03 0.01 9/88 408
Sun Dec 27 20:57:55 CET 2015 7.41 0.15 started xbian-inotifyd-bridge            0.16 0.03 0.01 4/78 420
Sun Dec 27 20:57:57 CET 2015 8.76 0.15 xbmc-rc                                  0.16 0.03 0.01 9/89 468
Sun Dec 27 20:57:57 CET 2015 9.23 0.15 stopped failsafe                         0.16 0.03 0.01 6/89 483
Sun Dec 27 20:57:58 CET 2015 9.80 0.15 started cgmanager                        0.16 0.03 0.01 8/101 512
Sun Dec 27 20:57:58 CET 2015 9.81 0.15 started autofs                           0.16 0.03 0.01 7/101 511
Sun Dec 27 20:57:58 CET 2015 9.80 0.15 autofs                                   0.16 0.03 0.01 7/101 511
[... 65 lines cut ...]
Sun Dec 27 20:59:08 CET 2015 80.17 0.15 stopped xbmc-loaded                      6.18 1.82 0.63 3/210 1742
Sun Dec 27 20:59:29 CET 2015 101.15 0.15 started mounted-tmp                      8.97 2.72 0.95 10/230 1824
Sun Dec 27 20:59:29 CET 2015 101.16 0.15 started xbmc-done                        8.97 2.72 0.95 10/230 1823
Sun Dec 27 20:59:29 CET 2015 101.14 0.15 started screen-cleanup                   8.97 2.72 0.95 10/230 1823
Sun Dec 27 20:59:29 CET 2015 101.15 0.15 started ntp                              8.97 2.72 0.95 7/224 1824
Sun Dec 27 20:59:31 CET 2015 103.57 0.15 stopped xbmc-failed-start                8.97 2.72 0.95 10/223 1850
Sun Dec 27 20:59:31 CET 2015 103.56 0.15 stopped wait-for-state  xbmc-done        9.45 2.93 1.03 8/220 1850
Sun Dec 27 20:59:42 CET 2015 114.24 0.15 stopped screen-cleanup                   10.06 3.16 1.11 6/215 1873
[... cut ...]
Sun Dec 27 21:01:58 CET 2015 212.54 0.15 stopped rc                               10.79 5.75 2.27 1/206 2368
Sun Dec 27 21:02:28 CET 2015 242.32 22.78 started xbmc-screensaver                 6.98 5.30 2.23 4/208 2453
Sun Dec 27 21:02:28 CET 2015 242.70 22.78 stopped xbian-xbmc-cpufreq               6.98 5.30 2.23 4/211 2468
Sun Dec 27 21:02:29 CET 2015 243.04 22.78 started xbmc-priority                    6.98 5.30 2.23 2/213 2489
Sun Dec 27 21:02:29 CET 2015 243.69 22.78 stopped xbmc-priority                    6.98 5.30 2.23 4/206 2506

mkreisl, 2015-12-27:
Please update your Kodi. This issue has been already fixed here https://github.com/xbianonpi/xbian/issues/790

suuhm, 2015-12-27:
OKay, thanks, but ive already seen this posting. i cannot reinstall or upgrade any package. its already up to date
should is use the Development-Repository?

mkreisl, 2015-12-27:
No, there are a lot of 15.2 packages (without RC3) on the server in stable repo.
Did you already run `sudo apt-get update`?
What about `sudo apt-cache policy xbian-package-xbmc`?

suuhm, 2015-12-27:
Yes i did. there arent any updates

apt-cache policy xbian-package-xbmc says:

´´´
xbian-package-xbmc:
  Installiert:           15.2~RC3-1443590769
  Installationskandidat: 15.2~RC3-1443590769
  Versionstabelle:
 *** 15.2~RC3-1443590769 0
        100 /var/lib/dpkg/status
´´´

mkreisl, 2015-12-27:
So this is absolutely wrong. I got:
```
  Installationskandidat: 15.2-1449270576
  Versionstabelle:
     15.2-1449270576 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     15.2-1449196858 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     15.2-1448682632 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     15.2-1446621231 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     15.2-1446505745 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     15.2-1445985905 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
     15.2~RC3-1445287766 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-jessie armhf Packages
```
Seems you don't have a proper /etc/apt/sources.list.d/xbian.list
Should be:
```
deb mirror://apt.xbian.org/mirror.txt      stable  main rpi-jessie
###deb mirror://apt.xbian.org/mirror.txt      staging main rpi-jessie
###deb mirror://apt.xbian.org/mirror.txt  devel   main rpi-jessie
```

suuhm, 2015-12-27:
ok yes, im using the staging level:

deb mirror://apt.xbian.org/mirror.txt      staging main rpi-jessie

mkreisl, 2015-12-27:
add the stable and all is ok

suuhm, 2015-12-27:
adding the stable to the staging?
or comment out the staging line?

and after that should i redo and apt-get upgrade to downgrade all packages?

mkreisl, 2015-12-27:
i wrote ADD
if you also want packages from staging, leave staging uncommented, if not, comment out staging
and after that please do `sudo apt-get update;sudo apt-get upgrade`

suuhm, 2015-12-27:
ok, this had fixed my issue. but it was a little bit confusing, I thought at staging repo was the newest package...

closed 2015-12-27 by suuhm