xbianonpi/xbian #797: remote control commands executed twice

Marsu31, 2015-12-29:
Hi,

Hardware : raspberry 2 with GPIO IR receiver (TSOP4838), Logitech Harmony emulating MCE-1039 remote.
Software : xbian up to date 

The MCE-1039 is recognized as "mceusb" remote.
When I use it the moves are executed twice, this is the kodi.log for one click on right arrow :
~~~
DEBUG: Keyboard: scancode: 0x6a, sym: 0x0113, unicode: 0x0000, modifier: 0x0
DEBUG: OnKey: right (0xf083) pressed, action is Right
DEBUG: LIRC: Update - NEW at 39604:000000037ff07bde 00 KEY_RIGHT mceusb (KEY_RIGHT)
DEBUG: OnKey: 168 (0xa8) pressed, action is Right
~~~

The "OK" is not executed twice : 
~~~
DEBUG: LIRC: Update - NEW at 3389751:000000037ff07bdd 00 KEY_OK mceusb (KEY_OK)
DEBUG: OnKey: 11 (0x0b) pressed, action is Select
~~~

If I restart lirc the kodi.log becomes :
~~~
DEBUG: Keyboard: scancode: 0x6a, sym: 0x0113, unicode: 0x0000, modifier: 0x0
DEBUG: OnKey: right (0xf083) pressed, action is Right
~~~

but, in this case, the key "OK" doesn't work anymore.

I used *irw* to see what lirc understands and it returns exactly the keys I pressed, only once, not twice. It's like if the KEY_RIGHT was read twice, from 2 paths ? or 2 threads ?

is this a bug or a configuration issue ?

regards
(sorry for my English)

jbkt, 2016-03-22:
Hi,

I have the same issue, latest version xbian running on a PI2, using HP MCE remote (usb connected 
IR), like you irw is right but xbmc move twice when using left or right arrow.

mkreisl, 2016-03-22:
Yes, there should be.

Please change (comment out) in all `.conf files` in folder **/etc/lirc/hardware** the parameter 
`LIRCD_ARGS="-u"` to `#LIRCD_ARGS="-u"` if there is one.

And then restart lirc and xbmc 
```
sudo restart lirc
sudo restart xbmc
```
and try again. 

For more information please type `man lircd` to see what the -u parameter does (Kodi reads some keys doubled from LIRC and standard Input events)

jbkt, 2016-03-22:
Thx mkreisl, but it did not work.

I had this:
`xbian@xbian /etc/lirc/hardware $ grep LIRCD_ARGS *.conf
custom.conf:#LIRCD_ARGS=""
default.conf:#LIRCD_ARGS=""
lirc_hb.conf:LIRCD_ARGS="-u"
lirc_rpi.conf:LIRCD_ARGS="-u"
logitech_ultrax.conf:#LIRCD_ARGS=""
mceusb.conf:#LIRCD_ARGS=""
mceusb_event0.conf:#LIRCD_ARGS=""
mceusb_event1.conf:#LIRCD_ARGS=""
mceusb_event2.conf:#LIRCD_ARGS=""
srm7500libusb.conf:LIRCD_ARGS=""
x10.conf:#LIRCD_ARGS=""
xbox.conf:#LIRCD_ARGS=""
`
So it was already diasblaed for all the MCE confs.
I tried to change lirc_hb.conf and lirc_rpi.conf anyway, rebooted, but without success.

mkreisl, 2016-03-22:
@jbkt 
What happens if you disable lirc completely?

jbkt, 2016-03-22:
I haven't tried, but if I do this:
`xbian@xbian ~ $ sudo ir-keytable -p RC-6
[sudo] password for xbian:
Protocols changed to RC-6
xbian@xbian ~ $ ir-keytable
Found /sys/class/rc/rc0/ (/dev/input/event3) with:
        Driver mceusb, table rc-rc6-mce
        Supported protocols: NEC RC-5 RC-6 JVC SONY SANYO LIRC RC-5-SZ SHARP XMP other
        Enabled protocols: RC-6
        Name: Media Center Ed. eHome Infrared
        bus: 3, vendor/product: 0471:0815, version: 0x0000
        Repeat delay = 500 ms, repeat period = 125 ms`
It works !

Now I habe to find out how to do this permanent.

jbkt, 2016-03-22:
By defaut I have this:

xbian@xbian ~ $ ir-keytable
Found /sys/class/rc/rc0/ (/dev/input/event3) with:
        Driver mceusb, table rc-rc6-mce
        Supported protocols: NEC RC-5 RC-6 JVC SONY SANYO LIRC RC-5-SZ SHARP XMP other
        Enabled protocols: RC-6 LIRC
        Name: Media Center Ed. eHome Infrared
        bus: 3, vendor/product: 0471:0815, version: 0x0000
        Repeat delay = 500 ms, repeat period = 125 ms

It seems that is LIRC protocol that is the problem.

mkreisl, 2016-03-22:
hmm, maybe you can add this command here (in file /etc/init/lirc.conf)
```
               if [ "$START_LIRCD" = "true" ]; then
                        echo "Starting remote control daemon(s) : LIRC "
                        LIRCD_ARGS=`build_args $LIRCD_ARGS`
  
                        if [ ! -z "$LIRCD_ARGS" ]; then
                                set -- $LIRCD_ARGS
                                /usr/local/sbin/lircd $@
                                                  # <--- put it in here
                        fi
                fi

                if [ "$START_LIRCMD" = "true" ]; then
  ```

mkreisl, 2016-03-22:
Unfortunately I don't have such an IR receiver. 
I would assume that if you disable the LIRC protocol LIRC stops working ...

jbkt, 2016-03-23:
You were right: if I stop lirc (`sudo stop lirc`) it is the same as if I do `sudo ir-keytable -p RC-6`: directional keys are working properly. But in both case the OK key is not working...

jbkt, 2016-03-23:
This a true MCE remote (similar to this one: http://www.amazon.com/gp/product/B00KDNIRYA) that I bought a long time ago as it was working out of the box on XBMC. It still is on Kodi over Ubuntu or in Openelec.

jbkt, 2016-03-23:
I have just checked on my Ubuntu box: LIRC protocol is used. So I have tried `sudo ir-keytable -p LIRC` and it work the same as RC-6. So it semms that using tow protocols as the same time was the problem: twice the events sent to Kodi.
Now OK (and perhaps other keys) is still not working even if I see the key pressed using irw, so I guess it is due to another configuration problem.

mkreisl, 2016-03-23:
> So it semms that using tow protocols as the same time was the problem: twice the events sent to Kod

This would make sense :smile: 

> Now OK (and perhaps other keys) is still not working even if I see the key pressed using irw, so I guess it is due to another configuration problem.

Yes, this is a configuration problem in Kodi. You have to edit the Lircmap.xml. To do this, please copy the file **/usr/local/share/kodi/system/Lircmap.xml** to **/home/xbian/.kodi/userdata/** end edit it there (do never edit the original, it will be overwritten when next update comes)

jbkt, 2016-03-28:
I have copied **/usr/share/kodi/system/Lircmap.xml** from ubuntu into** /home/xbian/.kodi/userdata/** and the OK button works, but I still have the events in double. if I diable LIRC ``sudo ir-keytable -p RC-6` the events are not in double but the Lircmap.xml is ignored.

So what is managing RC events when LIRC is off?  Caqn I disable that and use LIRC only?

mkreisl, 2016-03-28:
> So what is managing RC events when LIRC is off? Caqn I disable that and use LIRC only?

Kodi reads them directly from (u)input events Try unloading uinput module (sudo rmmod uinput)

mkreisl, 2016-03-28:
Another possible solution would be using inputlirc instead of lirc

See also here: http://kodi.wiki/view/HOW-TO:Set_up_an_MCE_remote_control_in_Linux

ccfiel, 2017-03-05:
I have this problem :( but this problem is not present in openelec.

ccfiel, 2017-03-05:
@mkreisl I tried to so sudo rmod uinput it does not have double key entry but the enter/ok button does not work. I tried to do irw and press ok button in the remote lirc detects it 

```
xbian@xbian ~ $ irw
000000037ff07bdd 00 KEY_OK mceusb
000000037ff07bdd 01 KEY_OK mceusb
000000037ff07bdd 00 KEY_OK mceusb
000000037ff07bdd 01 KEY_OK mceusb
```

but there is no action in kodi.

ccfiel, 2017-03-05:
@mkreisl It works. 

```sudo ir-keytable -p LIRC```
I just put it  /etc/init/lirc.conf and now its working! Thanks!

mkreisl, 2017-03-09:
@ccfiel 
which version of xbian-package-lirc are you using?
and could you please show me the place where you added this command?

and need some infos for integration into standard package, because /etc/init/lirc.conf would be overwitten by next update of package xbian-package-lirc:

is module mceusb loaded?

If yes, I could make a modification like this in /etc/init/lirc.conf:

```

                        if [ ! -z "$LIRCD_ARGS" ]; then
                                set -- $LIRCD_ARGS
                                /usr/local/sbin/lircd $@
                        fi

                        if lsmod | grep -qw mceusb && [ -e /sys/class/rc ]; then
                                ir-keytable -p LIRC || :
                        fi
                fi

                if [ "$START_LIRCMD" = "true" ]; then
```

Would be nice if you could test this, thanks :smile:

eyal0, 2017-03-27:
@ccfiel Your solution worked for me, too.

As I understand it, the problem is that the same key press was getting interpreted twice: once as a keyboard and once as the remote.  That's why it was in the log once each way.

When lirc was disabled, only the keyboard continued to work and only the directional keys.  The OK button and all the others didn't work.

For some reason, the ir-keytable command fixed it for me.  I can see in kodi.log that there are no more keyboard lines.  Great!

mkreisl, 2017-03-27:
@eyal0 
Still got no answer from @ccfiel, could you please answer my questions, thanks

eyal0, 2017-03-27:
@mkreisl I tried your patch, it didn't work.  What else would you like to know?

eyal0, 2017-03-27:
@mkreisl There is no "mceusb" in the output of lsmod

xbian@xbmc /etc/init $ sudo lsmod
Module                  Size  Used by
ir_lirc_codec           3984  3 
lirc_dev                6726  1 ir_lirc_codec
rc_rc6_mce              1264  0 
brcmfmac              167271  0 
brcmutil                4305  1 brcmfmac
cfg80211              333350  1 brcmfmac
uio_pdrv_genirq         2769  0 
gpio_ir_recv            2621  0 
uio                     6043  1 uio_pdrv_genirq
uinput                  6081  0

mkreisl, 2017-03-27:
@eyal0 
which version of xbian-package-lirc are you using?

no module mceusb loaded, ok. so my patch can not work

mkreisl, 2017-03-27:
@eyal0 
could you please try this patch? changed `mceusb` to `rc_rc6_mce`
```

                        if [ ! -z "$LIRCD_ARGS" ]; then
                                set -- $LIRCD_ARGS
                                /usr/local/sbin/lircd $@
                        fi

                        if lsmod | grep -qw  rc_rc6_mce && [ -e /sys/class/rc ]; then
                                ir-keytable -p LIRC || :
                        fi
                fi

                if [ "$START_LIRCMD" = "true" ]; then
```

eyal0, 2017-03-27:
Regarding version:
```
xbian@xbmc ~ $ sudo apt-cache policy xbian-package-lirc
xbian-package-lirc:
  Installed: 1.5.1
  Candidate: 1.5.1
  Version table:
 *** 1.5.1 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
        100 /var/lib/dpkg/status
     1.5.0-3 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     1.5.0-2 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages

```
The modified patch works.  Remote works as expected, arrow keys and OK button, too.  Here's the patch:

```
xbian@xbmc /etc/init $ diff -u lirc.conf.bak lirc.conf
--- lirc.conf.bak       2017-03-27 20:50:09.382284006 +0000
+++ lirc.conf   2017-03-27 21:10:13.833252867 +0000
@@ -89,6 +89,10 @@
                                set -- $LIRCD_ARGS
                                /usr/local/sbin/lircd $@
                        fi
+
+                        if lsmod | grep -qw rc_rc6_mce && [ -e /sys/class/rc ]; then
+                                ir-keytable -p LIRC || :
+                        fi
                fi
 
                if [ "$START_LIRCMD" = "true" ]; then

```

Here's the log with debug enabled when I press down:

```
00:14:08.678 207.610794 T:1954705984   DEBUG: LIRC: Update - NEW at 192342:000000037ff07be0 00 KEY_DOWN mceusb (KEY_DOWN)
00:14:08.679 207.611160 T:1954705984   DEBUG: OnKey: 167 (0xa7, obc88) pressed, action is Down

```

Previously I'd had the "Keyboard" line as in the first comment.

mkreisl, 2017-03-27:
Great. So i'll modify package, could you please test new version tomorrow?

I'm posting the link for downloading new package here tomorrow afternoon.

Thanks in advance

eyal0, 2017-03-27:
Will I need to reverse the above patch?

Just detail what steps do to do install it and I'll try it, no problem.

It's very late here so I'm signing off.  I'll be in front of a computer again in about 9 hours.  Thanks for your help, it was very prompt!

mkreisl, 2017-03-27:
Package for download and testing is ready :smile: 

> Will I need to reverse the above patch?

No, just download this package http://apt.xbian.org/pool/devel/rpi2-jessie/x/xbian-package-lirc/xbian-package-lirc_20170327_armhf.deb and install it with `dpkg -i xbian-package-lirc_20170327_armhf.deb`

ccfiel, 2017-03-27:
Sorry for the late reply :( @mkreisl yes it work :) I really do not know what I did. What do you want me to check?

mkreisl, 2017-03-27:
No problem, @eyal0 already answered my question.

If you want you could test new lirc package, see my previous post here

ccfiel, 2017-03-27:
@mkreisl ok I will try. :) by the way great project. Keep it up! :+1:

eyal0, 2017-03-28:
@mkreisl Sorry to say but now the remote isn't working at all!  None of the buttons have any effect.  Also, when I enable debug logging, I don't see any button presses in `kodi.log`.  I'm able to control my system using the Kore app (Kodi remote control for Android phone).  Here's a log of some stuff that I tried.

```
xbian@xbmc ~ $ sudo irw
[sudo] password for xbian: 
connect: Connection refused
xbian@xbmc ~ $ sudo irw /var/run/lirc/lircd
connect: Connection refused
xbian@xbmc ~ $ sudo ir-keytable
Found /sys/class/rc/rc0/ (/dev/input/event1) with:
        Driver gpio-rc-recv, table rc-rc6-mce
        Supported protocols: NEC RC-5 RC-6 JVC SONY SANYO LIRC RC-5-SZ SHARP XMP other 
        Enabled protocols: LIRC 
        Name: gpio_ir_recv
        bus: 25, vendor/product: 0001:0001, version: 0x0100
        Repeat delay = 500 ms, repeat period = 125 ms

```

Do you know what the issue might be?

Also, can you point me to an older version so that I can get my remote working in the meanwhile?  Thanks!

mkreisl, 2017-03-28:
Bad news :disappointed: 

to install older version, please run

```
sudo apt-cache policy xbian-package-lirc
audo apt-get install --reinstall xbian-package-lirc=1.5.1
```

Hard for me to debug, unfortunately do not have any mce rc :disappointed:

eyal0, 2017-03-28:
The problem might have been other changes in that package and not the above few lines?  Because thoes worked for me as a patch.

I'll try to do some debugging myself.

mkreisl, 2017-03-28:
Thanks for your help. Must be another issue

eyal0, 2017-03-28:
Here's what I know so far:

If I install the new version that you made yesterday, my remote stops working entirely.  irw doesn't show anything.  Using apt-cache and apt-get commands above to revert to lirc version 1.5.1 works and lets irw see my button presses again.  However, there are the double-presses of the arrow keys again, back to where I start.

I tried this:

1. Install new version.
2. Copy /etc/init/lirc.conf to /etc/init/lirc.conf.new .
3. Copy /etc/lirc to /etc/lirc.new (the whole directory).
4. Reinstall the old version with the apt-get command above.
5. Copy over /etc/init/lirc.conf with lirc.conf.new .
6. Copy of /etc/lirc with /etc/lirc.new .
7. Restart the system.

That leaves me in a start when I've got 1.5.1 installed but lirc.conf and /etc/lirc are all from the new version that you supplied.  In this situation, I'm able to use my remote and there are no double-presses, thanks to the `ir-keytable -p LIRC` in the new files.

So there is still something in the new version that isn't working with my setup but it isn't lirc.conf nor is it in /etc/lirc .  What else is that new package doing?  I dunno.

mkreisl, 2017-03-29:
@eyal0 
That's absolutely confusing me

All changes in lirc package were in `/etc/lirc` and oc `/etc/init/lirc.conf`. All binaries are  rebuilt (that's part of the build process), but from same source file.

Just installed that version on my RPi3, seems to work with my X10 device. `sudo irw` shows all my key I press on the rc

Could you replace the binaries from `/usr/local/bin` and `/usr/local/sbin/`

Btw, useful commands are `apt-get download` to download package and
`dpkg -x` extracts all files from given deb to any folder you want

So you can download both versions, extract them in different folders and change files one after the other.

And again, thanks for your help

mkreisl, 2017-03-29:
@eyal0 

Oooops, i forgit to ask: what hardware do you have. That new package only runs on RPi2 or RPi3

eyal0, 2017-03-31:
@mkreisl Haha, none of those!  I'm on a [cubox](https://www.solid-run.com/freescale-imx6-family/cubox-i/), about 3 years old maybe.

mkreisl, 2017-03-31:
I see, but that is same problem. cubox has cortex-a9, and rpi2,3 has cortex-a7 core, means that this package built for rpi2 (for cortex-a7) is not running on cubox. Will make package for i.MX6

mkreisl, 2017-03-31:
@eyal0 
Here we go, xbian-package-lirc for i.MX6: http://apt.xbian.org/pool/devel/imx6-jessie/x/xbian-package-lirc/xbian-package-lirc_20170331_armhf.deb

Checked on my Hummingboard - seems to work :smile: