xbianonpi/xbian #799: Newest update breaks SQL database on Raspberry PI

LauwS, 2016-01-04:
I thought this was an issue with my PI, so i reinstalled with latest image, created advancedsettings.xml and filled out the same details i always do.
Restarted the PI and everything looked fine.
Saw there was updates in xbian-config, so i installed those.
Rebooted and now all shows and movies are gone.

The model is Raspberry PI B.

mkreisl, 2016-01-04:
@LauwS 
Which version of *xbian-package-xbmc* are you using? And which repos do you have enabled (stable, staging, devel) ?

LauwS, 2016-01-04:
Ehm... where / how do i check that?

-------- Oprindelig meddelelse --------

Dato: 04/01/2016 18.54 (GMT+01:00)
Til: xbianonpi/xbian <xbian@noreply.github.com>
Cc: "René M. Laursen" <mester_lauw@hotmail.com>
Emne: Re: [xbian] Newest update breaks SQL database on Raspberry PI (#799)

mkreisl, 2016-01-04:
Enabled repo's: `cat /etc/apt/sources.list.d/xbian.list`
Installed package: `dpkg-query -l xbian-package-xbmc`

Your mysql server is running?

LauwS, 2016-01-04:
Enabled repo´s:
deb mirror://apt.xbian.org/mirror.txt      stable  main rpi-jessie
### deb mirror://apt.xbian.org/mirror.txt  staging main rpi-jessie
### deb mirror://apt.xbian.org/mirror.txt  devel   main rpi-jessie

Installed package:
ii  xbian-package-xbmc      15.2-1449270576  armhf            XBMC (rpi/isengard)

And yes the SQL runs, it works on another PI and on my PC :-)

LauwS, 2016-01-04:
Just did the same on the PI where it works, and it´s the same.
Difference in those 2 PI´s are:
Failing one - Model B Revision 2.0 - Mounting holes
Working one - Model B+

According to http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/

LauwS, 2016-01-04:
Did some more testing... and if i clean install and add a video smb folder and choose TV series, it can fine update library.
As soon as i add my advancedsettings.xml and reboot. Then there is no library and i cannot update the added folder.

LauwS, 2016-01-04:
Okay, nevermind. It was an mysql issue, had to do a mysqladmin flush-hosts.
Now it works like a charm, as before :+1:

closed 2016-01-04 by LauwS