xbianonpi/xbian #801: CEC update issue.

coderpop, 2016-01-11:
HI 

Posted this problem on the Xbian forum a week ago but no response so I thought there might be someone here who can help. I have two Rpi1 with latest Xbian on it, but 4 CEC packages refuse to update. I tried apt-get clean but still get the same problem.

The packages to update are libcec, libcec3, libcec-dev and cec-utils.

I have tried all at once and even individually. When I tried individual updates here is the response I get.....

install cec-utils = Depends libcec3
install libcec    = Depends python-libcec
install libcec3   = Already got newer verion of this package installed
install libcec-dev= Depends libcec3

I also tried an install of python-libcec with the following response...
install python-libcec = newest version already installed

Any idea how to get this sorted and what might have caused the problem?

mkreisl, 2016-01-11:
Did you already tried `sudo apt-get install -f` ?

coderpop, 2016-01-11:
Tried just now (hope I did it right)!

xbian@kitchen ~ $ sudo apt-get install -f
[sudo] password for xbian:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  libgif4
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
xbian@kitchen ~ $ sudo apt-get install -f python-libcec
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-libcec is already the newest version.
python-libcec set to manually installed.
The following package was automatically installed and is no longer required:
  libgif4
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
xbian@kitchen ~ $ sudo apt-get autoremove
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  libgif4
0 upgraded, 0 newly installed, 1 to remove and 4 not upgraded.
After this operation, 56.3 kB disk space will be freed.
Do you want to continue? [Y/n] y
Create a snapshot of '/run/btrfs-snap/root/@' in '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-01-11-2056'
Transaction commit: none (default)
Delete subvolume '/run/btrfs-snap/root/@last_good_known'
Create a snapshot of '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-01-11-2056' in '/run/btrfs-snap/root/@last_good_known'
Transaction commit: none (default)
Delete subvolume '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-01-01-1040'
@btrfs-auto-snap_apt-run-2016-01-11-2056, 1 created snapshots, 1 destroyed snapshots, 0 warnings.
(Reading database ... 32866 files and directories currently installed.)
Removing libgif4:armhf (4.1.6-11) ...
Processing triggers for libc-bin (2.19-18+deb8u1) ...
xbian@kitchen ~ $ sudo apt-get install -f libcec
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcec : Depends: python-libcec but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
xbian@kitchen ~ $ sudo apt-get install -f python-libcec
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-libcec is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
xbian@kitchen ~ $ sudo apt-get install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
xbian@kitchen ~ $ sudo apt-get install -f libcec3
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  autofs cec-utils cifs-utils exfat-fuse exfat-utils fontconfig-config hfsplus
  hfsprogs hfsutils ifplugd ir-keytable libaacs0 libafpclient0 libass5
  libbluetooth3 libbluray1 libcec-dev libcurl3 libenca0 libfontconfig1
  libfribidi0 libgif7 libgraphite2-3 libharfbuzz0b libhfsp0 libiw30 libmad0
  libmicrohttpd10 libmpeg2-4 libmysqlclient18 libnfs4 libnl-3-200
  libnl-genl-3-200 libpcrecpp0 libsmbclient libsqlite0 libssh-4 libtiff4
  libtiff5 libtinyxml2.6.2 libvorbisfile3 libxslt1.1 libyajl2 mysql-common pv
  python-sqlite sudo usbutils wireless-tools wpasupplicant
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  cec-utils libcec-dev
The following packages will be REMOVED:
  libcec python-libcec xbian-package-config-xbmc xbian-package-xbmc
  xbian-update
The following packages will be DOWNGRADED:
  cec-utils libcec-dev libcec3
WARNING: The following essential packages will be removed.
This should NOT be done unless you know exactly what you are doing!
  xbian-update xbian-package-xbmc (due to xbian-update) libcec (due to
  xbian-update)
0 upgraded, 0 newly installed, 3 downgraded, 5 to remove and 0 not upgraded.
Need to get 259 kB of archives.
After this operation, 74.7 MB disk space will be freed.
You are about to do something potentially harmful.
To continue type in the phrase 'Yes, do as I say!'
 ?]
Abort.
xbian@kitchen ~ $

SO not sure how to proceed from here.

mkreisl, 2016-01-11:
Hmmm, looks strange. Which repos do you have enabled?
```
cat /etc/apt/sources.list.d/xbian.list
```
Seems you have a mixed install from stable/staging/devel.

What I would do is the hard way:
Removing all cec packages (ignoring dependecies) and reinstalilng all of them

```
sudo dpkg -r --force-all cec-utils libcec libcec-dev libcec3:armhf python-libcec
sudo apt-get install libcec
```

coderpop, 2016-01-12:
Hi,
I have 2x Rpi1 with this issue and both are on stable only. I did try enabling staging but it didn't work.

I will try your solution tomorrow as been busy with a MS machine all evening.

Thanks, will let you know how it goes!

Happy New Year.

coderpop, 2016-01-12:
sudo dpkg -r --force-all cec-utils libcec libcec-dev libcec3:armhf python-libcec and a reboot.

Now I have no SSH into that pi,  so until I can get a monitor on it I'm stuck.

mk01, 2016-01-12:
@coderpop 

isn't just boot stuck due some other reason ? can you ping RPI ? if so, can you try 
```
telnet [my.rpi.ip]
```

mkreisl, 2016-01-12:
@coderpop 
You did a reboot before reinstalling cec? I've never talked about a reboot. So I would assume some services are not up because of missing depenencies.

coderpop, 2016-01-12:
Yes I did reboot before reinstalling, guess it's too much windoes experience! 

But now it will not connect by SSH after a 'soft' reboot from SSH link. 

To get SSH access again I need to do a hard reset. After that SSH works until I reboot via SSH and then it won't connect any more.

Glad I only did one of the 2 Rpi now! ;-)

mkreisl, 2016-01-12:
Sounds more like a hardware issue. What about cec now?

mk01, 2016-01-12:
@coderpop @mkreisl 

regarding the cec pkgs update - what always worked at last was to specify ALL cec related pkgs on cmd line (maybe even including xbmc, perhaps xbian-update). with libcec3 I created somewhere conflict or circular dependency (and never returned back to correct it). so depending on apt-get's complains, I would start with:
```
apt-get install cec-utils libcec libcec-dev libcec3 python-libcec
```
+ eventually adding "xbian-package-xbmc xbian-update"

or your forced approach but other way:
```
apt-get download cec-utils libcec libcec-dev libcec3 python-libcec
dpkg -i --force-all *.deb
dpkg --configure -a
```

mk01, 2016-01-12:
@coderpop 

until you settle the packages / apt state use this for reboot:
```
sync && sudo reboot -nf
```

coderpop, 2016-01-14:
I'll pull the card and find a backup and start again.

As an aside, todays 2 upgrades for Rpi2 (shairplay and xbianupdate) breaks I2S audio for me. 

The good news is that btrfs finally does what it should and I could revert to a working pi again :)

mkreisl, 2016-01-14:
I would assume that xbian-updates always writes `dtoverlay=lirc-rpi` into /boot/config.txt
Yesterday I got a Hifiberry Amp+ for my RPi2 and figured out that this setting does not work with the Hifiberry (found an entry in the forum)

coderpop, 2016-01-14:
I was just about to add more info... you are so fast!

2 things I noticed on all updates lately is not only what you mentioned but also updating to xbian 1.0 beta 2 is always shown, is that right?

As for the audio, I have lirc-rpi hashed out. It doesn't seem to get overwritten for me. In Kodi alsa is selected and aplay -a shows the device as expected. But this is a recurring issue for me on most upgrades recently.

I believe that the lirc uses the same stable PWM pin on the pi that the dac needs and when enabled will 'steal' it for it's own use. Lirc doesn't really need a PWM that good and it could be moved to another pin, or at least that is my understanding. Would you agree with that?

mkreisl, 2016-01-14:
> 2 things I noticed on all updates lately is not only what you mentioned but also updating to xbian 1.0 beta 2 is always shown, is that right?

Yes

> I believe that the lirc uses the same stable PWM pin on the pi that the dac needs and when enabled will 'steal' it for it's own use. Lirc doesn't really need a PWM that good and it could be moved to another pin, or at least that is my understanding. Would you agree with that?

I did not have a look which pins are using for lirc and i2s and if there is an overlapping. But, if `dtoverlay=lirc-rpi` is in, no sound comes out, but all alsa devices are there (kernel 4.1). Have no idea why this does not work (older kernel 3.18 did not require uncommenting this parameter)

mkreisl, 2016-01-14:
But anyway, I have an up2date RPi2 (stable and staging repo enabled) and sound still comes out of the speakers :smile:

mkreisl, 2016-01-14:
Yes, seems you're right - Pin 12 (GPIO_18) is used for both, WTF :rage:
So, yes,  I would agree

mkreisl, 2016-01-14:
@coderpop 
Have remapped LIRC pins, now seens it's working together:

That is my setting in /boot/config.txt
```
# GPIO pins have to be remapped for LIRC for working together with Hifiberry
dtoverlay=lirc-rpi,gpio_out_pin=16,gpio_in_pin=17

# Hifiberry
dtoverlay=hifiberry-amp
```

coderpop, 2016-01-14:
That's exactly what I thought. Well done, another issue solved.

Shall I try and upgrade again now?

mkreisl, 2016-01-14:
Sure :smile: Now you know how to revert to a prior snapshot

coderpop, 2016-01-14:
Nope! - Still slience here with todays upgrades. Only stable selected, so should work, but it doesn't.

coderpop, 2016-01-14:
HEELLLLPPPP! - 

OK, this time around the snapshot the worked last time had _rollback at the end of it, so I entered it just like that and still have no sound!   ARRGGHHHH........

coderpop, 2016-01-14:
What is going on here? Is it me?

xbian@cinema ~ $ sudo btrfs-auto-snapshot list root
root/@
root/@btrfs-auto-snap_apt-run-2016-01-11-1137
root/@btrfs-auto-snap_apt-run-2016-01-14-1801
root/@btrfs-auto-snap_apt-run-2016-01-14-1929
root/@btrfs-auto-snap_daily-2015-07-23-1030_rollback
root/@btrfs-auto-snap_daily-2016-01-12-0801
root/@btrfs-auto-snap_daily-2016-01-13-0900
root/@btrfs-auto-snap_daily-2016-01-14-0757_rollback_rollback
root/@btrfs-auto-snap_weekly-2016-01-02-1037
root/@btrfs-auto-snap_weekly-2016-01-09-0917
root/@last_good_known
xbian@cinema ~ $ sudo btrfs-auto-snapshot rollback root/@btrfs-auto-snap_daily-2016-01-14-0757
Error: Source name does not exists!
xbian@cinema ~ $ sudo btrfs-auto-snapshot rollback root/@btrfs-auto-snap_daily-2016-01-13-0900
Error: Source name does not exists!
xbian@cinema ~ $ sudo btrfs-auto-snapshot rollback root/@btrfs-auto-snap_apt-run-2016-01-14-1801
Error: Source name does not exists!

mkreisl, 2016-01-14:
Strange.
So, what packages exactly were updated?

coderpop, 2016-01-14:
shairplay and xbianupdate

mkreisl, 2016-01-14:
Hmmm, unbelievable.
And your audio device is there but no sound.
Did you checked /boot/config.txt?

coderpop, 2016-01-14:
config.txt is still as it was with rpi-lirc commented out and hifiberry still there.

coderpop, 2016-01-14:
I don't think the problem is with config.txt but rather it is somewhere else.

coderpop, 2016-01-14:
Just tries another reboot and now it won't boot.

Tried to pull the power and reapply power, still nothing.

So removing the card now and will flash to some old working version (wish I had taken a back up today when it was working)...

mkreisl, 2016-01-14:
Hmmm, so we newer know what the culprit is

coderpop, 2016-01-14:
The btrfs issue has happened to me many times before, that's why I was happy today when it finally worked as it should. Not sure what happens to get it to the point it will list a shapshot and then adamently deny it exists when I try to restore it!  :-0

You never know, at some point in the future I might fell brave/suicidal/drunk enough to try another upgrade! :)

coderpop, 2016-01-14:
Luckily I took an image 8 days ago, so hopefully not all is lost.

mkreisl, 2016-01-14:
Ok

coderpop, 2016-01-14:
Any idea or thoughts on the btrfs thing?
How or why would it list a backup and then say it doesn't exist when I try to restore it?

mkreisl, 2016-01-14:
No (I don't use this command)

I'm always doing the hard way:
````
sudo mount /dev/root  /mnt
cd /mnt/root
ls -l
sudo mv @ @-bad # Move current running root away
sudo mv @-<oldersnap> @ # Make older snap to current
sync
reboot
```
And this has ever worked for me :smile:

coderpop, 2016-01-14:
The commands I used are listed as the ones to use on the Xbain wiki. Maybe the wiki needs to change?

mkreisl, 2016-01-14:
I know it is there (I used this command one time for testing, and worked too)

coderpop, 2016-01-14:
OK, flash worked and I have sound again! - will see what upgrades it sees now.....

coderpop, 2016-01-14:
I guess I have drifted well off topic here, but I just applied all update/upgrade (except todays 2) and all is well again. 

But, after an apt-get update I get the following.....

W: There is no public key available for the following key IDs:
8B48AD6246925553
 NO_PUBKEY CBF8D6FD518E17E1

Do you think this is Xbian related?

coderpop, 2016-01-14:
Thanks for your time tonight, I must go now, but appreciate the help!

Cheers!

mkreisl, 2016-01-14:
NP, I'm still struggling to get lirc work with remapped lirc pins :rage:

coderpop, 2016-01-14:
Just saw your mail and couldn't leave it at that! - 

Will just remapping the pins do it? I don't use lirc so have no experience but maybe you need a python script to 'bit bang' the pin to give a PWM type output. The pin used is the only hardware controlled PWM on the pi, so swapping to a non-hardware PWM pin probably won't work without software to support it.

Of the two, lirc is only working a low frequency in the KHz range while the DAC will need the hardware to operate in the MHz range.

These might help.....

http://raspberrypi.stackexchange.com/questions/28510/lirc-raspberry-pi-2-no-input
http://www.raspberrypi.org/forums/viewtopic.php?f=91&t=86609
http://www.raspberrypi.org/documentation/configuration/pin-configuration.md

Maybe you already know all this, but I thought it might help!

mkreisl, 2016-01-14:
Thanks, just having success :smile:
My RPi1 which I have configured and used for LIRC did not work with GPIO17. Now  changed to GPIO22 and is now working. Maybe an other device occupies this GPIO

coderpop, 2016-01-14:
Good! :)

coderpop, 2016-01-15:
Here is the solution to the key issue above for anyone else who encounters it....

$ sudo apt-get install debian-keyring debian-archive-keyring
$ sudo apt-key update

coderpop, 2016-01-18:
Seems that Xbian-update is the culprit.

I just tried a working pi2 and only upgraded this module. Boot config was OK and aplay-l was showing the DAC, but no audio output.

mkreisl, 2016-01-18:
Hmm, interesting. 

So, what was the latest version of xbian-update which works on your system?
```
dpkg-query -W xbian-update
```
(I would assume xbian-update 20151221-0)

coderpop, 2016-01-18:
Your assumption is correct. 20151221-0 works OK but as soon as the upgrade to the one on the apt repo is installed, I2S stops working. I haven't tried shairplay on it's own yet as I don't need it.

If I may say so I have noticed another thing. That is that since the last kernel update to 4.1.13 the pi no longer recognises 64Gb memory cards that previously worked (mounted as storage devices, not for boot). Not had time to roll back and see if it is the kernel, but it's my guess that that's the most likely culpret.

coderpop, 2016-01-18:
To fully complete the test, shairplay has no problem upgrading, so it really is xbian-update that is causing this issue.

mkreisl, 2016-01-18:
Hmm, I have really no idea where the problem could be :disappointed: 

I'm using RPi2 with all updates (stable and staging) and I have sound with my Hifiberry Amp+ (using local built Kodi Jarvis RC which is already available on devel repo for everybody)

coderpop, 2016-01-22:
I notice a new xbian-update with yesterdays date.... Does that have the same problem or is it the replacement to cure the issue?

mkreisl, 2016-01-22:
I would assume that your issue still exists with the new xbian-update

coderpop, 2016-01-23:
I don't know, I was merely asking the question before I tried it!

I will have to make a backup tomorrow and then try again.

coderpop, 2016-01-24:
Today saw 88 updates for Rpi2. One of them was xbian-update...  I updated all except xbian-update and all was well. I then tried xbian-update and rebooted. I2S sound was gone. So flashed back to previsous working version and re-upgraded the remaining 87 updates.

coderpop, 2016-01-25:
As the CEC update issue is solved I will close this and open a new issue about xbian-update which is where we have got to.....

closed 2016-01-25 by coderpop