xbianonpi/xbian #802: RSS-Feed not readable on pi / works on windows

r-oldenburg, 2016-01-14:
Hi all,

i tried adding a new RSS source to my kodi on xbianonpi which was already working in my windows kodi. In the end I directly copied the part from sources.xml to make sure there are no typos or similar.

Nevertheless, kodi (on pi) always tells me that it could not load the directory content.
Seems to be the combination of this particular server and libCURL which seems to always send a "RANGE" header.

The relevant log part is this:
`21:21:35  84.798676 T:1644319664  NOTICE: Thread JobWorker start, auto delete: true`
`21:21:36  85.509460 T:1644319664   ERROR: CCurlFile::FillBuffer - Failed: HTTP returned error 416`
`21:21:36  85.509712 T:1644319664  NOTICE: CCurlFile::FillBuffer - Reconnect, (re)try 1`
`21:21:36  85.600594 T:1720079280  NOTICE: Thread FileCache start, auto delete: false`
`21:21:36  85.726318 T:1644319664   ERROR: failed to load xml from <rss://www.zdf.de:80/ZDFmediathek/podcast/222432>. error: <6>`
`21:21:36  85.731964 T:1954230272   ERROR: GetDirectory - Error getting rss://www.zdf.de:80/ZDFmediathek/podcast/222432/`
`21:21:36  85.736542 T:1954230272   ERROR: CGUIMediaWindow::GetDirectory(rss://www.zdf.de:80/ZDFmediathek/podcast/222432/) failed`
`21:21:36  85.747833 T:1720079280  NOTICE: Thread BackgroundLoader start, auto delete: false`

HTTP error 416 says "Requested Range not satisfiable".

BTW: using wget or curl on cmd line works perfectly fine **even on the pi**: 
`wget -O- http://www.zdf.de:80/ZDFmediathek/podcast/222432`

mkreisl, 2016-01-14:
@r-oldenburg 
I added above link (`http://www.zdf.de:80/ZDFmediathek/podcast/222432`) to my RssFeeds.xml.
Works perfectly (using Kodi Jarvis Beta 5)

r-oldenburg, 2016-01-14:
Right. But I was not talking about a regular RSS feed. 
My problem arises when trying to add a rss feed (podcast) as a **video source**.
Works fine on win, but not on xbian. :-(

r-oldenburg, 2016-01-14:
so it is about sources.xml

r-oldenburg, 2016-01-14:
And to be even more concise, it is about this feature: http://kodi.wiki/view/RSS_media_source

mkreisl, 2016-01-14:
So, WHY did you not show your `sources.xml` so we can see what you did? Would make thinks easier

r-oldenburg, 2016-01-14:
Here you are. My current sources.xml:
[sources.xml.txt](https://github.com/xbianonpi/xbian/files/90750/sources.xml.txt)

r-oldenburg, 2016-01-14:
And here the failing call debug-logged.
[debug-log-rss-zdf.txt](https://github.com/xbianonpi/xbian/files/90760/debug-log-rss-zdf.txt)

mkreisl, 2016-01-14:
MUCH better :smile: 
Added the relevant part into my sources.xml, again: works perfectly
(RPi2, Jarvis Beta 5, all repos (stable, staging, devel) enabled, System up2date)

r-oldenburg, 2016-01-14:
Ok. That's weird. 
Mine is also RPi2, don't know the specific xbian version.
Where can I find the version info within the system?

Downloaded the image "XBian_2016.01.03_rpi2" 
System is up2date (yesterday apt-get updated/upgraded)...

mkreisl, 2016-01-14:
Yes, indeed. 

This is what you get (with enabled stable, staging AND devel repo)
```
xbian-package-xbmc:
  Installiert:           16.0~BETA5-1452789651
  Installationskandidat: 16.0~BETA5-1452789651
  Versionstabelle:
 *** 16.0~BETA5-1452789651 0
        100 /var/lib/dpkg/status
     16.0~BETA5-1452597028 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi2-jessie armhf Packages
     16.0~BETA4-1449806132 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi2-jessie armhf Packages
     16.0~BETA3-1449348881 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi2-jessie armhf Packages
     15.2-1449275614 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2-1449200414 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2-1448685125 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2-1448575360 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2-1446623926 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2-1446508466 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2-1445988736 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2~RC3-1445290460 0
        500 http://apt.xbian.org/ stable/rpi2-jessie armhf Packages
     15.2~RC3-1443593390 0
        500 http://apt.xbian.org/ staging/rpi2-jessie armhf Packages
     15.2~RC3-1443548436 0
        500 http://apt.xbian.org/ staging/rpi2-jessie armhf Packages
     15.2~RC2-1442072724 0
        500 http://apt.xbian.org/ staging/rpi2-jessie armhf Packages
```
when issuing command `apt-cache policy xbian-package-xbmc`

r-oldenburg, 2016-01-14:
Ok, here is mine:
```
xbian-package-xbmc:
  Installed: 15.2-1449275614
  Candidate: 15.2-1449275614
  Version table:
 *** 15.2-1449275614 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
        100 /var/lib/dpkg/status
     15.2-1449200414 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
     15.2-1448685125 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
     15.2-1448575360 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
     15.2-1446623926 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
     15.2-1446508466 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
     15.2-1445988736 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
     15.2~RC3-1445290460 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi2-jessie armhf Packages
```

mkreisl, 2016-01-14:
You're using 15.2 which is still standard for now.

Sorry,  I have no further ideas how I can help you

r-oldenburg, 2016-01-14:
Any idea what this *FileCache* is about?
```
17:43:29 16006.757812 T:1794372528  NOTICE: Thread FileCache start, auto delete: false
17:43:29 16006.798828 T:1794372528    INFO: CFileCache::Process - Hit eof.
17:43:29 16006.849609 T:1794372528   DEBUG: Thread FileCache 1794372528 terminating
17:43:29 16006.850586 T:1582298032   ERROR: failed to load xml from <rss://www.zdf.de:80/ZDFmediathek/podcast/222432>. error: <12>
```

mkreisl, 2016-01-14:
No, I have same messages in my log

mkreisl, 2016-01-14:
I would assume that already downloaded content will be cached

r-oldenburg, 2016-01-14:
well. thanks for tryin... Then it must be caused by this sudden eof... will investigate further on this. Could you post a "successful" log?

mkreisl, 2016-01-14:
http://paste.ubuntu.com/14497771/

r-oldenburg, 2016-01-14:
are u sure? Shows only this:
1 /media/system/log/kodi.log

mkreisl, 2016-01-14:
Öhm, sorry (used echo instead of cat :sleepy:
http://paste.ubuntu.com/14497842/

r-oldenburg, 2016-01-14:
great. Thanks!

r-oldenburg, 2016-01-15:
Hey there again.
It seems that parsing (using tinyxml2) fails directly within the content (parsing a comment).
This indicates that the file is cropped somehow... Will try it again later with more degug flags turned on. Maybe a charset issue. 
I remember that I changed system keyboard, etc. to german settings.

r-oldenburg, 2016-01-16:
OK. I got it!!!
As so often, it is sooo simple.
Problem only occurs when trying to add this particular RSS feed (zdf.de) as they apparently try to recognize the User-Agent header info. And if it is not known it seems that they always return the regular html page instead of the RSS xml. That is why the xml parsing (consequently) fails.

I could solve the problem (temporarily) by putting fiddler in the middle, replacing the User-Agent header with something regular like "Mozilla/5.0 ...".
Suddenly the RSS feed can easily be added as a video source.

**So the current User-Agent "Kodi/15.2 (X11; Linux armv7l) XBian/1.0 HW_RaspberryPi/1.0 App_Bitness/32 Version/15.2-Git:2015-12-04-858ee60" seems to be problematic.**
Is this somewhere configurable? Could it be made configurable?
What is your "Jarvis Beta 5" User-Agent header value?

r-oldenburg, 2016-01-16:
And by the way: The Add-On "Mediathek" for german content also fails for the same reason. (And the same workaround helps)
Possibly some others also fail for the same reason...

mkreisl, 2016-01-16:
No idea. Why don't you test Jarvis? You only have to enable devel repo in `/etc/apt/sources.list.d/xbian.list`