xbianonpi/xbian #804: Rpi 2 will not boot without ntework connected.

coderpop, 2016-01-22:
As per the title... Boots fine if lan connected, but try to boot as 'stand alone' and it freezes at the network connecting screen and won't go any further.

mk01, 2016-01-23:
@coderpop 

there should failsafe mechanism kick in after maybe 100s.

try this ? https://github.com/xbianonpi/xbian-update/commit/9a05d6a1546920a68734bbce210c82530a7497e6

just btw - in case running without LAN (unrelated to WLAN usage) is standard use case, eth0(X) interfaces should be untagged as "auto" adapters. otherwise core scripts expects them to come up during boot.
(if you tag eth0 for instance allow-hotplug, it will be configured if cable connected, but won't delay booting the system in case it isn't connected).

coderpop, 2016-01-23:
-qn is already in place.

I just want it to boot with or without a lan cable attached.

I will try leaving it for 100s, but I think it has had a lot longer than that previously without booting (only solution is to pull the power and re-connect the power).

mk01, 2016-01-23:
@coderpop

if I understand you correctly - you mean it is doing thing WITH -qn in the command ???

coderpop, 2016-01-23:
Sorry for any confusion.

The '-qn' command was already present in the config file when the issue occured. It is not the solution...

coderpop, 2016-02-21:
After the recent updates this seems to be cured.

Thanks.

closed 2016-03-03 by mkreisl